johnfanv2/LenovoLegionLinux #96: [BUG] Fans not spinning, requesting help ASAP.

No-Biggie805, 2023-08-21:
**Problem Descripition**
Fans are not spinning even in the bios, i am even afraid to try out opening a game, cpu and gpu seem to be cool but TCTL and and temp1 sensor show real high values, i'd like to come back from zero.
How this was originated? well i executed setmyfancurve.sh and it hanged at the end i canceled but now fans are like this and i am afraid it starts getting too hot :(. PLS help did i ruin the fan curve on my PC?? 

Thank you.

**Model and Debug Info**

Distribution: Ubuntu 23.04
Model name: 15ACH6H
CPU model: AMD 5600H
GPU model: RTX 3060
Keyboard backlight: single colour
Light in lid or logo: no
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JU
        Version: Legion 5 15ACH6H
        Serial Number: PF3TY1R0
        UUID: d6014a52-e7f2-11ec-80f2-6c2408afb0e6
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JU_BU_idea_FM_Legion 5 15ACH6H
        Family: Legion 5 15ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN50WW
        Release Date: 11/24/2021
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.50
        Firmware Revision: 1.50

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 38
CPU temperature EC error: 0
CPU temperature EC: 38
CPU temperature ACPI error: 0
CPU temperature ACPI: 61
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 61
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 35
GPU temperature ACPI error: 0
GPU temperature ACPI: 60
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 60
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 2
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 8
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       4       0       49      0       59      0       59
1500     1600    5       4       46      55      56      60      56      60
1900     2000    5       4       52      59      57      61      57      61
2100     2200    4       3       56      63      58      62      58      62
2200     2300    2       2       58      72      59      63      59      63
2500     2500    2       2       67      77      60      64      60      64
2900     3000    2       2       72      81      60      68      127     127
3500     3500    2       2       78      127     65      127     127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

johnfanv2, 2023-08-22:
I guess you accidentially looked the fan controller, since I see that "EC lockfancontroller: true". This completely disables fan control and keeps the current fan speed. Please open the GUI and untick the box `Lock fan controller`.

Do you think that this was cause by the the script `setmyfancurve.sh` or have you used the GUI? If you run the script again will it lock the fan controller again.

No-Biggie805, 2023-08-22:
ok so apparently a bios reset did the trick, thank god i though it was going to be needed to do an RMA repair.. 
I'd just like to know something now, how will a simple script affect throrought the whole firmware? should it just be set at just the current OS? is it possible to deal with it any other way in case this happens?
I just need to know cause my panic was quite high, though i am aware of the risks hence why i came asking.
thank you.

johnfanv2, 2023-08-22:
If the fan controller is locked you can do one of the following (in order of effort):
1. unlock it by unticking `Lock fan controller` and press 'Apply to HW' in the python GUI (see README)
2. reset the embedded controller by pressing the power button for a long time (see README FAQ for details)
3. reset the BIOS
4. update, downgrade,or re-flash the BIOS software

Option 1 should work. I have tested it on the same BIOS software (but higher version). Options 2 or 3 will work.  Option 4 will work in any case.

However, this should not happen by the script. Do you think that this was caused by the the script setmyfancurve.sh or have you used the GUI? If you run the script again will it lock the fan controller again?

No-Biggie805, 2023-08-22:
> However, this should not happen by the script. Do you think that this was caused by the the script setmyfancurve.sh or have you used the GUI? If you run the script again will it lock the fan controller again?

Sorry for the late reply i went to sleep.
When i first got my PC either was user error or the repo had bugs, `sudo cat /sys/kernel/debug/legion/fancurve
` would hang and then with `setmyfancurve.sh` i had this happen, i did try the gui, but from i think everything set was disabled since i use FN+Q alot to begin with and experimented around locking sensors but then disabled it again.
When i then executed `setmyfancurve.sh` as stated it hanged and fans completely stopped spinning and at least on linux cpu+gpu were fine but already it did not feel normal when i got to windows its were it got real bad and so it throtled.
I was desperate, but right after that i found your workaround in the mainpage and reset EC.
Could you provide me the correct links of 3 and 4 for me to check out? Preferably in cases like this windows should not be even an option lmao.

Many thanks man, your work is quite astounishing btw.
Another thing when opening the gui to tray i cannot find it in KDE, is the anywere to have that set up or is it just better to do `setmyfancurve.sh` again, i already tested it and when i created the post it did seem to work, but i am a bit afraid now that is why.

No-Biggie805, 2023-08-24:
Closing out since the original point of the thread i solved out.

closed 2023-08-24 by No-Biggie805