johnfanv2/LenovoLegionLinux #97: [BUG] Desktop Session restoring spawns another `legion_gui` process

rozwell, 2023-08-22:
**Problem Descripition**

Firstly, `Autostart` itself works fine.

Unfortunately problem is with system Desktop Session restoring which spawns another `legion_gui` process, additionally without arguments, resulting in 2 instances where one has `Program must be run as root user!` warning message.

To reproduce: with session restoring enabled, just enable *Autostart*, then logout without closing `legion_gui` and login back.

Currently fast & easy workaround is to put `legion_gui` to excluded programs in system session settings.
 
 
**Model and Debug Info**
Distribution: Kubuntu 22.04.3 LTS
Model name: Legion 5 17ACH6
CPU model: AMD Ryzen 7 5800H

rozwell, 2023-08-25:
Here is the culprit:
```
$ ack legion_gui ~/.config/ksmserverrc 
program4=/usr/bin/legion_gui
restartCommand4=/usr/bin/legion_gui,-session,10dbd36769000169268858000000023600010_1692688876_114092,-name,legion_gui
```

MrDuartePT, 2024-04-25:
> Here is the culprit:
> 
> ```
> $ ack legion_gui ~/.config/ksmserverrc 
> program4=/usr/bin/legion_gui
> restartCommand4=/usr/bin/legion_gui,-session,10dbd36769000169268858000000023600010_1692688876_114092,-name,legion_gui
> ```

I running plasma 6 and I can't reproduce this error, also you are running 22.04 LTS I would recommend you to update for 23.04 or 24.04

closed 2024-04-25 by MrDuartePT