johnfanv2/LenovoLegionLinux #100: [Lenovo Legion Pro 5i 16'' (Gen 8, June 2023) KWCN36WW] Module Loads but FanControl doesn't work

octave-ati, 2023-08-25:
Hello,

Thanks a lot for this great module ! I bought the newest Gen8 Lenovo Legion Pro and sadly I am unable to use the fan control features with my setup. 
Hopefully it can easily be fixed since it's a new model you have probably not encountered yet !

**Problem Descripition**
Unable to use the fan control feature with the newest Legion 5i Pro.

The fancurve displays only zeros and I am unable to change it.

The computer also crashed when I tried to change the power mode.

**Model and Debug Info**

Distribution: Pop OS 22.04
Model name: Lenovo Legion Pro 5i 16'' (Gen 8)
CPU model: 13th Gen Intel(R) Core(TM) i7-13700HX
GPU model: NVIDIA RTX 4700
Keyboard backlight: RGB
Light in lid or logo: Yes
Light at IO-Ports at back: Don't know, I have disabled lightning

Output 

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WK
	Version: Legion Pro 5 16IRX8
	Serial Number: XXXX
	UUID: XXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WK_BU_idea_FM_Legion Pro 5 16IRX8
	Family: Legion Pro 5 16IRX8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN36WW
	Release Date: 04/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.36
	Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 41
CPU temperature EC error: 0
CPU temperature EC: 55
CPU temperature ACPI error: 0
CPU temperature ACPI: 41
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 41
GPU temperature error: 0
GPU temperature: 40
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 40
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 40
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1700
1 fanspeed EC error: 0
1 fanspeed EC: 18020
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1700
2 fanspeed error: 0
2 fanspeed: 1700
2 fanspeed EC error: 0
2 fanspeed EC: 14080
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 255
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 255
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100	 0	 0	 0	 0	 0	 0	 0	 0	 0
200	 0	 0	 0	 0	 0	 0	 0	 0	 0
300	 0	 0	 0	 0	 0	 0	 0	 0	 0
400	 0	 0	 0	 0	 0	 0	 0	 0	 0
500	 0	 0	 0	 0	 0	 0	 0	 0	 0
600	 0	 0	 0	 0	 0	 0	 0	 0	 0
700	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100	 0	 0	 0	 0	 0	 0	 0	 0	 0
200	 0	 0	 0	 0	 0	 0	 0	 0	 0
300	 0	 0	 0	 0	 0	 0	 0	 0	 0
400	 0	 0	 0	 0	 0	 0	 0	 0	 0
500	 0	 0	 0	 0	 0	 0	 0	 0	 0
600	 0	 0	 0	 0	 0	 0	 0	 0	 0
700	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
800	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
```

Thanks a lot in advance for any help that you can provide !

I have a very high sensitivity to noise so it is quite disturbing when the PC goes in "jet airplane mode" whenever it has to do some heavy computation, I have bought soundproof panels to try to insulate the noise but this is not perfect and I would love to be able to tweak my fan curve !

Don't hesitate to reach out if you want additional debug info !

johnfanv2, 2023-08-25:
It seems like the fancurve can  be read (100, 200, ... ) in first column. Unfortunately, in 2023 we can for now only set the custom fan curve in custom mode. This fan curve is only enabled in custom mode. Moreover, the units are fixed yet. So for now 1000 = 100% = max speed.

Can you get describe the exact steps you took to crash it? Did it freeze your whole laptop? Can you get  the  kernel logs for it. You get them with
```bash
# Logs of current session
sudo dmesg
# Logs of last session (before restart/crash)
journalctl -o short-precise -k -b -1
```

octave-ati, 2023-08-25:
So it is weird because I am currently on custom mode and the modifications I do do not register on the GUI. I am also unable to load and visualize the "premade" profiles. Currently working on getting you the logs !

octave-ati, 2023-08-25:
Please find the requested log of the crash : [log_clean.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/12438684/log_clean.txt)

I don't remember exactly what I was doing, I was mainly fiddling with the options to see if anything produced a meaningful change.

octave-ati, 2023-09-02:
Any updates? Still the same for me with the newest release

octave-ati, 2023-09-29:
Bumping for updates

octave-ati, 2023-09-29:
Sorry did not mean to close

bravosixgoingdark, 2023-10-07:
> It seems like the fancurve can be read (100, 200, ... ) in first column. Unfortunately, in 2023 we can for now only set the custom fan curve in custom mode. This fan curve is only enabled in custom mode. Moreover, the units are fixed yet. So for now 1000 = 100% = max speed.
> 
> Can you get describe the exact steps you took to crash it? Did it freeze your whole laptop? Can you get the kernel logs for it. You get them with
> 
> ```shell
> # Logs of current session
> sudo dmesg
> # Logs of last session (before restart/crash)
> journalctl -o short-precise -k -b -1
> ```

Custom mode for some reason likes to crash on me after a while. Same goes for Windows, the first time it crashes, then after it wakes back from crash it works like fine. But on Linux it just crash and crash for me.

A-Fax-Machine, 2026-09-03:
Bumping. While I am only testing LLL and have not installed it yet, I have the same Model, Product Name, and CPU/GPU configuration (but OS is CachyOS). My fan curve appears to be in a similar state as yours, though the output of `cat /sys/kernel/debug/legion/fancurve` looks somewhat different than yours since three years ago. My firmware version is newer I think: `KWCN54WW`

st0nie, 2026-09-14:
KWCN fan control was reworked in #567: on this firmware the WMI fan table takes level indices 0-10 (not percent), the Other Method paths are used, and the Python tools convert RPM to levels (and #571 reads the RPM ladders from the firmware at runtime). Please retest with the current release.

closed 2026-09-14 by st0nie