johnfanv2/LenovoLegionLinux #101: Support for Yoga Slim 7 Pro 14ARH7 82UU

florentineprinzessinzusachsen, 2023-08-26:
I'd be very thankful for support for the 2022 Yoga Slim 7 Pro.

Currently `sudo make forcereloadmodule` errors with `legion PNP0C09:00: Creating platform profile failed: -17`.

**Model and Debug Info**

Distribution: EndeavourOS 6.4.11-arch2-1 (64-bit)
Model name: Yoga Slim 7 Pro 14ARH7 82UU
CPU model: Ryzen 9 6900HS
GPU model: integrated
Keyboard backlight: single color with off/auto/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`:
```text
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82UU
        Version: Yoga Slim 7 Pro 14ARH7
        Serial Number: -
        UUID: -
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82UU_BU_idea_FM_Yoga Slim 7 Pro 14ARH7
        Family: Yoga Slim 7 Pro 14ARH7

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002F, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: KHCN29WW
        Release Date: 03/09/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.29
        Firmware Revision: 1.29

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Bios here: https://download.lenovo.com/consumer/mobiles/khcn29ww.exe

Thank you very much!

johnfanv2, 2023-08-26:
It seems like the kernel module was not loaded. I updated the code and added your model for testing. Please get the newest code. What is the output after you have run these commands
```bash
# Go to folder kernel_module in the directory LenovoLegionLinux
cd LenovoLegionLinux/kernel_module
# Compile
make
# Load but disable platform_profile since there might be conflict between ideapad driver and legion driver for now (will be fixed in future)
# This is the only difference compared to the instructions in the README
# This output is particularly interesting
sudo make forcereloadmodule_disableplatformprofile
sudo cat /sys/kernel/debug/legion/fancurve
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
```

Please also get the output of the following

## Disassembling ACPI tables

```bash
# Install requiered tools
sudo apt install acpica-tools
```

```bash
# Create folder for all the new files
mkdir acpi_re
cd acpi_re

# List ACPI tables and copy them
ls /sys/firmware/acpi/tables/
sudo cp --no-preserve=mode /sys/firmware/acpi/tables/*SDT* .

# Disassemble tables to output DSDT.dsl
iasl -e SSDT* -d DSDT
```

florentineprinzessinzusachsen, 2023-08-27:
Thank you!

**sudo make forcereloadmodule_disableplatformprofile**
```text
rmmod legion-laptop.ko || true
insmod legion-laptop.ko force=1 enable_platformprofile=0
dmesg --ctime
[So, 27. Aug 2023, 12:20:35] Linux version 6.4.11-arch2-1 (linux@archlinux) (gcc (GCC) 13.2.1 20230801, GNU ld (GNU Binutils) 2.41.0) #1 SMP PREEMPT_DYNAMIC Sat, 19 Aug 2023 15:38:34 +0000
[So, 27. Aug 2023, 12:20:35] Command line: initrd=\faf7a05819ea47d6aea93af797a51a62\6.4.11-arch2-1\initrd nvme_load=YES nowatchdog rw root=UUID=db9ce0f8-fb28-4d63-9575-8236208a5216 resume=UUID=e911a3d2-b099-4413-8262-3eefeac2089a systemd.machine_id=faf7a05819ea47d6aea93af797a51a62 quiet splash amdgpu.freesync_video=1 amdgpu.dc=1 amdgpu.dpm=1 amdgpu.si_support=1 amdgpu.cik_support=1
[So, 27. Aug 2023, 12:20:35] BIOS-provided physical RAM map:
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000000009f000-0x00000000000bffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000000100000-0x0000000009afffff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000009e00000-0x0000000009efffff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000009f00000-0x0000000009f27fff] ACPI NVS
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000009f28000-0x000000005fffefff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000005ffff000-0x000000005fffffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000060000000-0x0000000062363fff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000062364000-0x0000000064563fff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000064564000-0x000000006456cfff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000006456d000-0x000000006456ffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000064570000-0x000000007077efff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000007077f000-0x0000000072f7efff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000072f7f000-0x000000007af7efff] ACPI NVS
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000007af7f000-0x000000007affefff] ACPI data
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000007b000000-0x000000007bffffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x000000007ce00000-0x000000007fffffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x00000000c0200000-0x00000000c02fffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed80fff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x0000000100000000-0x00000003fe2fffff] usable
[So, 27. Aug 2023, 12:20:35] BIOS-e820: [mem 0x00000003ff340000-0x00000004a01fffff] reserved
[So, 27. Aug 2023, 12:20:35] NX (Execute Disable) protection: active
[So, 27. Aug 2023, 12:20:35] e820: update [mem 0x60457018-0x60461e57] usable ==> usable
[So, 27. Aug 2023, 12:20:35] e820: update [mem 0x60457018-0x60461e57] usable ==> usable
[So, 27. Aug 2023, 12:20:35] extended physical RAM map:
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000000000000-0x000000000009efff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000000009f000-0x00000000000bffff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000000100000-0x0000000009afffff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000009b00000-0x0000000009dfffff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000009e00000-0x0000000009efffff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000009f00000-0x0000000009f27fff] ACPI NVS
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000009f28000-0x000000005fffefff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000005ffff000-0x000000005fffffff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000060000000-0x0000000060457017] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000060457018-0x0000000060461e57] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000060461e58-0x0000000062363fff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000062364000-0x0000000064563fff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000064564000-0x000000006456cfff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000006456d000-0x000000006456ffff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000064570000-0x000000007077efff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000007077f000-0x0000000072f7efff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x0000000072f7f000-0x000000007af7efff] ACPI NVS
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000007af7f000-0x000000007affefff] ACPI data
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000007afff000-0x000000007affffff] usable
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000007b000000-0x000000007bffffff] reserved
[So, 27. Aug 2023, 12:20:35] reserve setup_data: [mem 0x000000007ce00000-0x000000007fffffff] reserved
[... cut ...]
External object resolution file            SSDT1
Input file SSDT1, Length 0x906 (2310) bytes
ACPI: SSDT 0x0000000000000000 000906 (v01 LENOVO CB-01    00000001 ACPI 00040000)
Pass 1 parse of [SSDT]
Pass 2 parse of [SSDT]
Pass 1 parse of [DSDT]
Firmware Error (ACPI): Failure creating named object [\_TZ.TZ01], AE_ALREADY_EXISTS (20221020/dswload-495)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20221020/psobject-372)
Could not parse ACPI tables, AE_ALREADY_EXISTS
```

Note, I did `sudo pacman -S acpica` since I am on Arch.

johnfanv2, 2023-08-27:
Ok, there is still some stuff to fix later.

Let try the most important one. Please use the python GUI to adapt the fan speed as described Open the GUI, click "Read from HW", then change the fan speeds, click "Apply from HW", and click "Read from HW". Check if your values are accepted and if you can hear a difference in fan speed. Try if small changes are accepted and larger changes are accepted and make a difference in reality.

florentineprinzessinzusachsen, 2023-08-28:
Thanks again for the quick reply and your support!

Here's the files zipped. I hope it contains all you need, otherwise please let me know: https://drive.google.com/file/d/1E7n8rh2-FDllEfpVnmnuU7vm-51y5ulf/view

Yes, reading, writing and re-reading the values works. I can hear the effect and smaller and incremental changes seem to work as well.

johnfanv2, 2023-08-29:
Ok, next we can try testing the power mode. I updated the code. Please get the newest code, compile again, and load the module again. Then change the power mode with Fn + Q and output `sudo cat /sys/kernel/debug/legion/fancurve` in each mode. Does you laptop has a LED, e.g. color of LED, to show the current power mode? In the output you should see that `powermode: 0`  changes to some other numerical value in each mode. 

You can also try to change the power mode, e.g. with the Python GUI. In you distro you might also have a power mode option to set the mode. In Ubuntu/Gnome it looks like this https://fostips.com/changing-power-mode-easy-ubuntu-21-04/. Maybe there is a similar setting in EndeavourOS.

closed 2026-09-12 by github-actions[bot]