johnfanv2/LenovoLegionLinux #102: Change default profiles for fan control and questions regarding automation

No-Biggie805, 2023-08-26:
**Is your feature request related to a problem? Please describe.**
So basically i've fiddling around your script and in terms of really having control of fan it does a nice job, but i think i would like just it to become a little bit more accessible or either its me that is missing some sort of configuration here, fans just are annoying and even with your script i hear them when programming with music on my speaker, ik there are headphones but i prefer speakers, the gui does not seem to work when closing to tray for me, and auto-start seems to be grayed out from the gui too.

If anything i mention below is already implemented below or otherwise still WIP feel free to mention about it i up for some troubleshooting or open for suggestion from my side too, i am not a pro developer to do much but i can do things myself too.  

**Describe the solution you'd like**
I'd basically just like to get easy to set the fans to quiet around 1300rpm baseline and then 1600rpm, ./setmyfancurve.sh works but if posssible and even more automated or accessible way through tray would have been awesome, if you know what iam talking about a similiar functionality to tuxedo control is good enough really.

**Describe alternatives you've considered**
An alternative way to do this would be for you to set the default rpm from the repo to the ones i pointed out, but yeah, i am not really an advanced person to talk what should be done, let me know if you have any suggestion regarding that, i can ofc save the script myself too in the meantime and use kde autostart but yeah then i need to use it manually, sorry i don really have any good alternative solution that i can think of regarding this..

**Model and Debug Info**
Distribution: kubuntu 23.04
Model name: Lenovo legion 5 15ACH6H
CPU model: AMD ryzen 5 5600h
GPU model: NVIDIA RTX 3060
Keyboard backlight: single colour 
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JU
        Version: Legion 5 15ACH6H
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JU_BU_idea_FM_Legion 5 15ACH6H
        Family: Legion 5 15ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN50WW
        Release Date: 11/24/2021
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.50
        Firmware Revision: 1.50

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Fancurve is just an example
Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 47
CPU temperature EC error: 0
CPU temperature EC: 47
CPU temperature ACPI error: 0
CPU temperature ACPI: 47
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 47
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 2
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 9
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       5       4       0       49      0       59      0       59
1300     1300    5       4       46      55      56      60      56      60
1600     1600    5       4       52      59      57      61      57      61
2100     2200    4       3       56      63      58      62      58      62
2200     2300    2       2       58      72      59      63      59      63
2500     2500    2       2       67      77      60      64      60      64
2900     2900    2       2       70      70      59      68      127     127
3500     3500    2       2       127     127     64      76      127     127
4200     4200    2       2       127     127     70      127     127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

johnfanv2, 2023-08-26:
If you do not install the GUI and the CLI it, you can only run the GUI as root user. But in this case, you cannot minimize to the tray or use autostart.

You can install it by using the .deb package for the python part which is published here: https://github.com/johnfanv2/LenovoLegionLinux/releases . You can also use the linked PPA if @MrDuartePT  has finished it. It will also allow to start a background service that always sets you configured profile when you change power mode. You can change these profiles in the GUI.

Let my know if you have any problems using this method. 

You also mentioned that it an "accessible way through tray would have been awesome". Do you want an entry in the tray to apply the profile, e.g. when you click on "Apply quiet profile" it should apply the fan curve in the quiet profile?

No-Biggie805, 2023-08-26:
> If you do not install the GUI and the CLI it, you can only run the GUI as root user. But in this case, you cannot minimize to the tray or use autostart.

Yes indeed i did not have the script installed i did kinda know the kernel modules and the python frontend were not really set together as since the main page explains building but from python it has the source code but not build for there.  

> You can install it by using the .deb package for the python part which is published here: https://github.com/johnfanv2/LenovoLegionLinux/releases . You can also use the linked PPA if @MrDuartePT has [...]

Yeah i sorted with .deb file + the modules built with just **make** thank you so much, it would have been nice to see it in the main page even if its just experimental, because i myself really appreciate this info because i was not getting there myself xp.
 
> Let my know if you have any problems using this method.

I did have one with the PPAs basically lenovolegionlinux-dkms errors out, can't really tell what its happening but i vote that its some conflit in my end given that i have the kernel modules built another way? I did do a `sudo make unninstall` however before doing that.

Here what the terminal spits out:
`ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/lenovolegionlinux-dkms.0.crash'
Error! Bad return status for module build on kernel: 6.2.0-27-generic (x86_64)
Consult /var/lib/dkms/lenovolegionlinux/1.0.0/build/make.log for more information.
dpkg: error processing package lenovolegionlinux-dkms (--configure):
 installed lenovolegionlinux-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 lenovolegionlinux-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

> You also mentioned that it an "accessible way through tray would have been awesome". Do you want an entry in the tray to apply the profile, e.g. when you click on "Apply quiet profile" it should apply the fan curve in the quiet profile?

Yes exacly that, if possible that is, my only wish here is to have a quiet profile so far, for performance mode i can just press the FN+Q, i don't have custom tho surprisingly for an instant i saw the power button turn pink  before red when fiddling around this evening also helps anything, its not a mode that i can change with FN+Q however.. 

Another request i would also have is like having a quiet mode but with the fan set to base of 1000rpm instead of 0rpm, i myself couldn't get to reach for that, but that now if its not really possible then its quite ok, because i now force 1300rpm and stays there for a bit instead of 1500/1600rpm which is where i here the fan whistle.
 
Many thanks so far for the amazing work out there^^.

johnfanv2, 2023-08-27:
I added the feature to set fan curve presets from the tray. Note that you can only set it from the tray if you have saved it before (currently a restart is needed after creating it).

On Ubuntu you can run the GUI as root and it will create a tray icon. But you have to do a right click to open the tray icon menu. You could try this first before dealing with the packages.

@MrDuartePT There seems to be a problem with the package.

MrDuartePT, 2023-08-27:
The error is because the kernel module is already install you need to delete and then install the deb.

MrDuartePT, 2023-08-27:
@johnfanv2 @No-Biggie805 dkms package now work you just need to ```apt-get update``` and ```sudo apt-get install lenovolegionlinux-dkms``` if dosent just readd the PPA.
[image]

No-Biggie805, 2023-08-27:
> I added the feature to set fan curve presets from the tray. Note that you can only set it from the tray if you have saved it before (currently a restart is needed after creating it).

Ok Tyvm, really appreciate on my end i have the options greyed out, for me no matter really like this is much appreciated.

@MrDuartePT much obliged, the daemon together with the modules installed successfully now. 

i just tested things out and fan curve works accordingly so really great.

I hope this will work well at least quiet mode is now good, like using it like this and press FN+Q for performance, i for the first time have been using this laptop without all the time worrying about the fans while doing the simple stuff.
Another suggestion i can make is like the daemon changing right through the firmware and like when setting the profile the firmware change colours according to what profile i am on, but with this i am happy with what you achieved, for me this is very nice many thanks.

Closing.

closed 2023-08-27 by No-Biggie805