johnfanv2/LenovoLegionLinux #109: [BUG]

miguelssm, 2023-09-05:
**Problem Descripition**
Lenovo Ideapad Slim 3 15AMN8 don't work keyboard and touchpad in any distribution.

**Model and Debug Info**

Distribution: Ubuntu 22.04, Ubunut 23.04, Linux Mint 21 (including version, e.g. Ubuntu 24.04)
Model name: ???
CPU model: AMD Ryzen 3 7320U
GPU model: NVIDIA Radeon
Keyboard backlight: no (
Light in lid or logo: no
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
MBIOS 3.3.0 present.

Handle 0x002F, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x000F, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82XQ
	Version: IdeaPad Slim 3 15AMN8
	Serial Number: XXXXXXXX
	UUID: XXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82XQ_BU_idea_FM_IdeaPad Slim 3 15AMN8
	Family: IdeaPad Slim 3 15AMN8

Handle 0x0034, DMI type 15, 31 bytes
System Event Log
	Area Length: 98 bytes
	Header Start Offset: 0x0000
	Header Length: 16 bytes
	Data Start Offset: 0x0010
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x00F0
	Status: Valid, Not Full
	Change Token: 0x00000005
	Header Format: Type 1
	Supported Log Type Descriptors: 4
	Descriptor 1: POST error
	Data Format 1: POST results bitmap
	Descriptor 2: PCI system error
	Data Format 2: None
	Descriptor 3: System reconfigured
	Data Format 3: None
	Descriptor 4: Log area reset/cleared
	Data Format 4: None
```

Output of `sudo dmidecode -t bios`:
`Handle 0x002D, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Abbreviated
	Installable Languages: 6
		en-US
		fr-FR
		ja-JP
		ko-KR
		zh-CHT
		zh-CHS
	Currently Installed Language: en-US

Handle 0x000E, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: L1CN32WW
	Release Date: 06/14/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		PNP is supported
		APM is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.32
	Firmware Revision: 1.32
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:

cat: /sys/kernel/debug/legion/fancurve: No existe el archivo o el directorio
```
/var/log/kern.log

miguelssm, 2023-09-05:
[kern.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12528742/kern.log)

miguelssm, 2023-09-05:
[kern.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12528945/kern.log)

kern.log with Ubuntu 23.04, same issue.

MrDuartePT, 2023-10-20:
This kernel driver dont aims to solve hardware problems, only aims to give support for LLT and Lenovo Vantage feature on Linux.

Pls use the bugzilla forum or Lenovo comunity forum to discuss this topic.

closed 2023-10-20 by MrDuartePT