johnfanv2/LenovoLegionLinux #112: Support for ideapad gaming 3 15IAH7

ArashkInGithub, 2023-09-06:
**Is your feature request related to a problem? Please describe.**
Almost every option in the application is grayed out.

**Describe the solution you'd like**
Add support for this laptop

**Model and Debug Info**

Distribution: Arch (including version, e.g. Ubuntu 24.04)
Model name: Ideapad gaming 3 15IAH7
CPU model: Intel core i5 12450H
GPU model: NVIDIA RTX 3050
Keyboard backlight: Single colour with medium and high brightness levels (Though, there are models with RGB)
Light in lid or logo: No
Light at IO-Ports at back: No

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82S9
	Version: IdeaPad Gaming 3 15IAH7
	Serial Number: Nice try
	UUID: No for here
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82S9_BU_idea_FM_IdeaPad Gaming 3 15IAH7
	Family: IdeaPad Gaming 3 15IAH7

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JMCN40WW(V3.03)
	Release Date: 08/25/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 3.40
	Firmware Revision: 3.40

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

No-Biggie805, 2023-12-14:
Could you try it this way, since you are using ubuntu? I am leaving the link here.
[https://mrduartept.github.io/LLL-pkg-repo/](url)

This will install the modules as well the frontend and configure all.
Currently i cannot get it to work automated but selecting profiles, (you have to set those up before hand), works pretty well and quite quick.

ArashkInGithub, 2023-12-14:
I'm afraid I don't use ubuntu. I forgot to delete the template portion but I use arch.

closed 2026-09-12 by github-actions[bot]