johnfanv2/LenovoLegionLinux #114: [BUG] Lenovo Legion 5 15ARH7H, Bios ver. JUCN59WW: no shutting down

Robert-ML, 2023-09-07:
**Problem Description**
After trying out the kernel module temporarily by following the steps in the README, I shutdown the system to reset it as I did not perform the permanent installation. After all the lights went out, the system started again and booted into GRUB. Not even the command `halt` in the command line of GRUB did not make the laptop to stay powered down.

I suppose my trial of the kernel modules is the cause of this behavior as it did not exist previously and I did not perform any system updates or changes in the BIOS. I have windows 10 on dual boot, but it also did not perform any system updates as I did not boot into it before noticing the behavior. Even when shutting down from Windows, the system turns back on.

I tried resetting the BIOS to default from within itself but the problem persists. I understood that there is an option to hard reset the bios holding the power button 20 seconds but I did not try it yet. If this happened to me, I think at least to make the most out of it and use my computer as a testing platform for this BUG :)) 

**Model and Debug Info**

Distribution: Ubuntu 22.04.3 LTS
Model name: Lenovo Legion 5 15ARH7H, Bios ver. JUCN59WW
CPU model: AMD® Ryzen 7 6800h with radeon graphics × 16
GPU model: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] / REMBRANDT (rembrandt, LLVM 15.0.7, DRM 3.49, 6.2.0-31-generic), note: in the BIOS the dynamic GPU option is selected
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RD
	Version: Legion 5 15ARH7H
	Serial Number: [censored]
	UUID: [censored]
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RD_BU_idea_FM_Legion 5 15ARH7H
	Family: Legion 5 15ARH7H

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JUCN59WW
	Release Date: 03/15/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.59
	Firmware Revision: 1.59

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

Robert-ML, 2023-09-18:
## Solution 

Holding the power button 20 seconds to shut down and power boot the laptop worked. It reset everything. The BIOS reset button in the software did not solve the problem, it needed this 20 second press.

Solution [source](https://forums.lenovo.com/t5/Gaming-Laptops/Legion-5-15ITH6H-Laptop-Lenovo-shutdown-problem/m-p/5178386?page=1#5781554).

MrDuartePT, 2023-10-20:
Since is solve i gonna close the issue.
And what you did is called a ec reset sometimes can help when you facing that kind of problems.

closed 2023-10-20 by MrDuartePT