johnfanv2/LenovoLegionLinux #117: Support for Lenovo Legion Y520-15IKBN

ehmad11, 2023-09-08:
**Is your feature request related to a problem? Please describe.**
Almost every option in the application is grayed out.

**Describe the solution you'd like**
Please add support for this laptop, my laptop is overheating

**Describe alternatives you've considered**
I tried windows Lenovo_NerveCenter with WINE and obviously it didn't work.

**Model and Debug Info**

Distribution: Fedora Linux 38 (including version, e.g. Ubuntu 24.04)
Model name: Lenovo Y520-15IKBN
CPU model: Intel i7-7700HQ (8) @ 3.800GHz (e.g. AMD Ryzen 7 5800H)
GPU model: NVIDIA GeForce GTX 1050 Mobile (e.g NVIDIA RTX 3700)
Keyboard backlight: single color  with  off/medium/bright (RGB, single color with off/on, single color with off/medium/bright)
Light in lid or logo: no (yes/no)
Light at IO-Ports at back: no (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 80WK
	Version: Lenovo Y520-15IKBN
	Serial Number: ###
	UUID: ####
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_80WK_BU_idea_FM_
	Family: IDEAPAD

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: 4KCN45WW
	Release Date: 01/11/2019
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 6 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.45
	Firmware Revision: 1.45

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
sudo cat /sys/kernel/debug/legion/fancurve
```

ulville, 2023-11-12:
Hey! I have the same laptop and I also would like to see this project supporting it someday even ideally gets merged into upstream kernel. I didn't have the time to test and help the development unfortunately...

But until then there is this project called [ExtremeCooling4Linux](https://gitlab.com/OdinTdh/extremecooling4linux) which is a python app that lets you enable and disable **extreme cooling mode** on our laptop model Y520. Just like the Nerve Center does. You need to disable Secure Boot though. It also has a system tray applet which you need to install separately.

I had helped the maintainer to test things on different bios versions a long time ago and it should probably work for you. It hasn't got updates for a while but it still works for me on Fedora 39.

PS: On Fedora I needed to install portio _one of the python modules that project requires_ through pip as a system module with sudo because that module is not in the Fedora repos.

ehmad11, 2023-11-12:
> Hey! I have the same laptop and I also would like to see this project supporting it someday even ideally gets merged into upstream kernel. I didn't have the time to test and help the development [...]

thank you i will try :+1:

ehmad11, 2023-11-12:
> thank you i will try 👍

@ulville it is working, you just made my day  :D thank you

ulville, 2023-11-12:
@ehmad11 you're welcome :)

Also a note for the maintainers of LLL. That project I suggested ( [ExtremeCooling4Linux](https://gitlab.com/OdinTdh/extremecooling4linux) ) probably includes most of the info needed to implement the same functionality within this project. And doing it with a kernel module would be a much better solution then a hacky python solution with unmaintained, potentially unsecure dependencies like [portio](https://pypi.org/project/portio/)

Semmu, 2026-09-08:
Yeah, this is still relevant. I tried to tackle it, but couldn't figure out the exact registers to manipulate.

st0nie, 2026-09-14:
Not planned: the Y520 (2017) uses a different, older EC/firmware generation than this driver supports (2020+ layouts; the allowlist starts at that generation). There is no realistic path to support it here — closing to keep the tracker honest.

closed 2026-09-14 by st0nie