johnfanv2/LenovoLegionLinux #119: [BUG] The GUI window is too high and some settings cannot be displayed

nukemiko, 2023-09-10:
**Problem Descripition**
The minimum window height is too large for the screen height (1600) so that settings below the window are outside the screen and cannot be displayed.
In addition, the elements within the window seem to have a fixed height, and there is a lack of side scroll bars. Even if the window is maximized, some settings below the window still cannot be displayed.

Screenshots:

The contents of "Fancurve Preset" are lost when the window is not maximized. (They can be displayed when the window is maximized, but still lack the red warning text at the bottom, and the three global buttons (Quit/Save/Save and Quit).)
[image]

And "Other Options" tab. (Cannot "Read from HW" and "Apply to HW")
[image]
[image]

**Model and Debug Info**

Distribution: Arch Linux (including version, e.g. Ubuntu 24.04)
Model name: 82JQ
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3060 Mobile
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: yes

sysms, 2023-09-17:
Same here, so the GUI is currently not usable for me.

Model and Debug Info

Distribution: Kali Linux 2023.3
Model name: Lenovo Legion 5 15ARH7H
CPU model: AMD Ryzen 7 6800H
GPU model: NVIDIA RTX 3060 Mobile
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no

luni3359, 2023-10-07:
@nukemiko as a temporary workaround you can press and hold Meta and drag the window around to access the rest of the settings.

MrDuartePT, 2023-10-20:
Well seems we need to fix the windows by adding a way to resize itself or add scroll bar.
I will try to give a look later since I not have time in the next 2 weak.

MrDuartePT, 2024-03-29:
Sorry for the delay I kinda forget about the issue :cry:

nukemiko, 2024-03-30:
This problem appers to still not be fixed. The upper and lower ends of the window still overflow the screen, and the side scroll bar doesn't work at all.

Screenshot in KDE Plasma (X11):
[image]

Screenshot in KDE Plasma (Wayland):
[image]

MrDuartePT, 2024-03-30:
Ok I gonna see what I can do. Since for me is not a problem since the window fit on the screen.

sysms, 2024-03-30:
For me, the new scrollbar kind of works, but does not scroll until the very end of the window:

[image]

Also when clicking on the "Read from HW" button on the "Other Options" tab, I get some weird errors in the terminal:

`update_view_from_feature 0
INFO:legion_linux.legion:Feature CPUCrossLoadingPowerLimit reading
INFO:legion_linux.legion:Feature CPUCrossLoadingPowerLimit reading: 0
update_view_from_feature 0
INFO:legion_linux.legion:Feature CPUAPUSPPTPowerLimit reading
INFO:legion_linux.legion:Feature CPUAPUSPPTPowerLimit reading: 0
update_view_from_feature 0
INFO:legion_linux.legion:Feature GPUCTGPPowerLimit reading
INFO:legion_linux.legion:Feature GPUCTGPPowerLimit reading: 35
update_view_from_feature 0
INFO:legion_linux.legion:Feature GPUPPABPowerLimit reading
INFO:legion_linux.legion:Feature GPUPPABPowerLimit reading: 15
update_view_from_feature 0
INFO:legion_linux.legion:Feature GPUTemperatureLimit reading
ERROR:legion_linux.legion:Feature GPUTemperatureLimit reading error [Errno 22] Invalid argument
ERROR:legion_linux.legion:[49265.834137] legion_laptop: get_simple_wmi_attributewith raw value: 0
[49265.834370] legion_laptop: ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:5: ACPI buffer length: 16
[49265.834580] legion_laptop: ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:3: ACPI buffer length: 16
[49265.834782] legion_laptop: Unexpected ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:8: expected type 1 but got 3
[49338.648804] legion_laptop: ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: ACPI buffer length: 16
[49338.649029] legion_laptop: ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: ACPI buffer length: 16
[49338.649241] legion_laptop: Unexpected ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:8: expected type 1 but got 3
[49338.657870] legion_laptop: get_simple_wmi_attributewith raw value: 0
[49338.658048] legion_laptop: get_simple_wmi_attributewith raw value: 0
[49338.658285] legion_laptop: ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:5: ACPI buffer length: 16
[49338.658498] legion_laptop: ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:3: ACPI buffer length: 16
[49338.658710] legion_laptop: Unexpected ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:8: expected type 1 but got 3
[49345.250265] legion_laptop: ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: ACPI buffer length: 16
[49345.250495] legion_laptop: ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: ACPI buffer length: 16
[49345.250704] legion_laptop: Unexpected ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:8: expected type 1 but got 3
[49345.259274] legion_laptop: get_simple_wmi_attributewith raw value: 0
[49345.259468] legion_laptop: get_simple_wmi_attributewith raw value: 0
[49345.259710] legion_laptop: ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:5: ACPI buffer length: 16
[49345.259943] legion_laptop: ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:3: ACPI buffer length: 16
[49345.260154] legion_laptop: Unexpected ACPI result for da7547f1-824d-405f-be79-d9903e29ced7:8: expected type 1 but got 3

Error occured [Errno 22] Invalid argument
Traceback (most recent call last):
  File "/opt/LenovoLegionLinux/python/legion_linux/legion_linux/./legion_gui.py", line 482, in update_view_from_feature
    value = self.feature.get()
            ^^^^^^^^^^^^^^^^^^
  File "/opt/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 390, in get
    return self._read_file_int(self.filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 269, in _read_file_int
    return int(self._read_file_str(file_path))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 266, in _read_file_str
    raise err
  File "/opt/LenovoLegionLinux/python/legion_linux/legion_linux/./../legion_linux/legion.py", line 260, in _read_file_str
    out = str(filepointer.read()).strip()
              ^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument

`

If I can/should provide any more info, please let me know.

closed 2024-03-31 by MrDuartePT

MrDuartePT, 2024-03-31:
Should be ok now (screenshot take with 125% scale on 1080p):
[image]

sysms, 2024-04-02:
@MrDuartePT thanks for fixing the GUI, it now works as expected! :D

But the other error messages remain, when clicking on the "Read from HW" button on the "Other Options" tab (see https://github.com/johnfanv2/LenovoLegionLinux/issues/119#issuecomment-2028458008). 

Is this expected, or should I open another issue for this?

Thanks again!

commit c460f141 (2024-03-29): Add missing line and fix whitespace

Closes: #119