johnfanv2/LenovoLegionLinux #123: Legion_Slim_7 core i9 not working with your patch

Marwan-Dalaty, 2023-09-13:
**Problem Descripition**

**Model and Debug Info**
https://psref.lenovo.com/Product/Legion_slim_7_16IRH8

Distribution: Ubuntu 24.04.3 TLS
Model name: ???
CPU model: Intel Core i9
GPU model: NVIDIA® GeForce RTX™ 4070

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82Y3
	Version: Legion Slim 7 16IRH8
	Serial Number: MP2F6ZSC
	UUID: 5c4d1f57-dcdf-11ed-8c91-088fc3e9e191
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82Y3_BU_idea_FM_Legion Slim 7 16IRH8
	Family: Legion Slim 7 16IRH8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Output of `sudo dmidecode -t bios`:
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: M0CN27WW
	Release Date: 03/30/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.27
	Firmware Revision: 1.27

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
cat: /sys/kernel/debug/legion/fancurve: No such file or directory)

MrDuartePT, 2023-10-19:
> 
[Wed Sep 13 23:21:16 2023] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
<
The device is not added in the device list yet.
Pls dowload the source build the kernel module and use:
```sudo make reloadmodule```
and see if the model works

MrDuartePT, 2023-10-19:
add suport to your model with this revert back with testing (03a2ec7)

emilf, 2025-09-07:
@FrostByghte Same here. I have 16IRH8 and no fan curves to edit. 
`
# cat /sys/kernel/debug/legion/fancurve

EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve fancurve_pmw_speed fancurve_rpm_speed
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4245487636
temperature access method: 5
CPU temperature error: 0
CPU temperature: 45
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 45
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 45
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 35
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1900
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1900
2 fanspeed error: 0
2 fanspeed: 1900
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1900
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1900
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 230
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 1
WMI light IO port: 2
WMI light Y logo/lid: 2
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
Max speed for fancurve: 10000
Current fan curve in hardware:
Fan curve current point id: 0
Fan curve points size: 0
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
Fan curve current point id: 0
Fan curve points size: 0
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================`

closed 2026-09-12 by github-actions[bot]