johnfanv2/LenovoLegionLinux #124: [BUG]

18aKissD, 2023-09-21:
**Problem Descripition**
sudo make reloadmodule comes back with error: could not insert module legion-laptop.ko: Key was rejected by service

**Model and Debug Info**

Distribution: Ubuntu 23.04
Model name: 2021 Legion 5 15ACH6H
CPU model: AMD Ryzen 5 5600h
GPU model: Nvidia RTX 3060
Keyboard backlight: SIngle color
Light in lid or logo: ?no
Light at IO-Ports at back: Only for charging

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82JU
	Version: Legion 5 15ACH6H
	Serial Number: -
	UUID: -
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82JU_BU_idea_FM_Legion 5 15ACH6H
	Family: Legion 5 15ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: GKCN54WW
	Release Date: 05/05/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.54
	Firmware Revision: 1.54

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
```

MrDuartePT, 2023-10-20:
@18aKissD 
This is related to ubuntu it seems a secure boot error.
Pls disable secure boot and try to renable the module.
I gonna close this since you have the same module as me and this issue is not related to your module is a Ubuntu issue.

closed 2023-10-20 by MrDuartePT

18aKissD, 2023-10-20:
Yes you're right, I just forgot to update. Though, thanks for helping!

MrDuartePT, 2023-10-20:
No problem!