johnfanv2/LenovoLegionLinux #127: Please add support for Yoga Air 14S (Yoga Slim 7 gen 8 on EU market)

mark-herbert42, 2023-10-09:
**Is your feature request related to a problem? Please describe.**
The fan on this laptop almost always on for no reason - temps are 28-30 degrees and fans are running as mad. Whoever designed the EC fan curve for this machine is a complete idiot....  

**Describe the solution you'd like**
Change fan curve and have quiet laptop as this one should be, so it will not start until hottest component of the system will reach 50 degrees.

**Model and Debug Info**

Distribution: Gentoo 
Model name: Lenovo Yoga Air 14S (I have Chineese version, on EU market it sold as Yoga Slim 7 14 Gen8)
CPU model: AMD Ryzen 7 7840S (repackaged for Lenovo 7840U chip in fact)

Keyboard backlight: single color with off/medium/bright)
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0035, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x0015, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83AA
	Version: YogaAir 14s APU8
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83AA_BU_idea_FM_YogaAir 14s APU8
	Family: YogaAir 14s APU8

Handle 0x003A, DMI type 15, 31 bytes
System Event Log
	Area Length: 130 bytes
	Header Start Offset: 0x0000
	Header Length: 16 bytes
	Data Start Offset: 0x0010
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x00F0
	Status: Valid, Not Full
	Change Token: 0x00000007
	Header Format: Type 1
	Supported Log Type Descriptors: 4
	Descriptor 1: POST error
	Data Format 1: POST results bitmap
	Descriptor 2: PCI system error
	Data Format 2: None
	Descriptor 3: System reconfigured
	Data Format 3: None
	Descriptor 4: Log area reset/cleared
	Data Format 4: None
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0033, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Abbreviated
	Installable Languages: 6
		en-US
		fr-FR
		ja-JP
		ko-KR
		zh-CHT
		zh-CHS
	Currently Installed Language: en-US

Handle 0x0014, DMI type 0, 24 bytes
BIOS Information
	Vendor: LENOVO
	Version: M6CN34WW
	Release Date: 08/17/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PC Card (PCMCIA) is supported
		PNP is supported
		APM is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.34
	Firmware Revision: 1.34

```

MrDuartePT, 2023-10-20:
I added support with this commit pls test all the features dac5a7c

mark-herbert42, 2023-10-21:
Thank you.  Looks like lenovo made this laptop working differently - it has same 8227 but does not follow the Legion way
[    2.936380] legion_laptop: legion_laptop starts loading
[    2.936425] legion PNP0C09:00: legion_laptop platform driver probing
[    2.936427] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 83AA; DMI_BIOS_VERSION:M6CN34WW
[    2.936429] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 83AA; DMI_BIOS_VERSION:M6CN34WW
[    2.936432] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[    2.936434] legion PNP0C09:00: Using configuration for system: M6CN
[    2.936435] legion PNP0C09:00: Skipping ACPI _STA check
[    2.936436] legion_laptop: Creating RAM access to embedded controller
[    2.936445] legion_laptop: Succeffuly mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x00000000e7748810
[    2.936534] legion PNP0C09:00: Read embedded controller ID 0x8227
[    2.936535] legion PNP0C09:00: Creating debugfs inteface
[    2.936541] legion_laptop: Creating sysfs inteface
[    2.936547] legion_laptop: Creating hwmon interface
[    2.936577] legion_laptop: Creating platform profile support
[    2.936580] legion_laptop: Init WMI driver support
[    2.936593] legion_wmi 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294: Register after probing for WMI.
[    2.936614] legion_laptop: Init keyboard backlight LED driver
[    2.936616] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[    2.936618] legion_laptop: Error for WMI method call to get brightness
[    2.936619] legion_laptop: Error reading keyboard brighntess
[    2.936620] legion PNP0C09:00: Init keyboard backlight LED driver failed. Skipping ...
[    2.936621] legion_laptop: Init Y-Logo LED driver
[    2.936622] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[    2.936623] legion_laptop: Error for WMI method call to get brightness
[    2.936624] legion_laptop: Error reading brighntess for light: 3
[    2.936625] legion PNP0C09:00: Init Y-Logo LED driver failed. Skipping ...
[    2.936626] legion_laptop: Init IO-Port LED driver
[    2.936627] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[    2.936628] legion_laptop: Error for WMI method call to get brightness
[    2.936629] legion_laptop: Error reading brighntess for light: 5
[    2.936629] legion PNP0C09:00: Init IO-Port LED driver failed. Skipping ...
[    2.936630] legion PNP0C09:00: legion_laptop loaded for this device
[    4.011533] legion_laptop: Set powermode

MrDuartePT, 2023-10-21:
what the ouput of ```doas cat /sys/kernel/debug/legion/fancurve```

mark-herbert42, 2023-10-21:
EC Chip ID: 8227
EC Chip Version: 626
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4128047120
temperature access method: 5
CPU temperature error: -14
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 57
CPU temperature WMI error: -14
CPU temperature WMI: 57
CPU temperature WMI2 error: -14
CPU temperature WMI2: 57
CPU temperature WMI3 error: -14
CPU temperature WMI3: 57
GPU temperature error: -14
GPU temperature: 57
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: -14
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 5
1 fanspeed error: -14
1 fanspeed: 1454829568
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 3400
1 fanspeed WMI error: -14
1 fanspeed WMI: 3400
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 3400
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 3400
2 fanspeed error: -14
2 fanspeed: 3400
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2700
2 fanspeed WMI error: -14
2 fanspeed WMI: 2700
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 2700
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 2700
powermode access method: 3
powermode error: -14
powermode: 67305474
powermode EC error: 0
powermode EC: 10
powermode ACPI error: -5
powermode ACPI: 1397501440
powermode WMI error: -14
powermode WMI: 1397501440
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: -22
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

MrDuartePT, 2023-10-21:
yha it have the same problems that other gen8 (2023) laptop, lenovo change how the fancurve works we still not have a solution to that

w0utert, 2023-11-10:
Any progress on this? I have the exact same laptop (EU version, dmidecode output is almost byte-to-byte the same as posted by @mark-herbert42) and the terrible fan curve is driving me crazy. This laptop will spin up the fans as low as ~35C degrees so even the shortest burst of CPU activity will activate them, after which they keep runnning for minutes while the Tctl is below 35C. If you use this machine for light tasks the fan is continously pulsating between either on/low or low/medium, depsite the fact that it never even gets warm to the touch.

Is there anything I can do to help figure out how to get fan control working on this laptop model? Is there any kind of technical documentation anywhere on how the Lenovo Embedded Controller or any other parts involved with the cooling setup work?

MrDuartePT, 2023-11-10:
> Any progress on this? I have the exact same laptop (EU version, dmidecode output is almost byte-to-byte the same as posted by @mark-herbert42) and the terrible fan curve is driving me crazy. This laptop will spin up the fans as low as slightly above ~35C degrees so even the shortest burst of CPU activity will activate them, after which they keep runnning for minutes while the Tctl is below 35C. If you use this machine for light tasks the fan is continously pulsating between either off/low or low/medium, despite the fact that it never even gets warm to the touch. As I'm typing this the stupid fan is expelling air out of the case that actually feels cooler than ambient :-(
> 
> Is there anything I can do to help figure out how to get fan control working on this laptop model? Is there any kind of technical documentation anywhere on how the Lenovo Embedded Controller or any other parts involved with the cooling setup work?

That should happen on a balance or quiet but could happen in performance mode. Try use quiet mode by press fn+q.

mark-herbert42, 2023-11-10:
> That shouldnt happen on a balance or quiet but could happen in performance mode. Try use quiet mode by press fn+q.

This laptop is just terrible for this point. Here quiet mode is always on fan. Balanced in fact is more silent under light load. That was a real surprise to me because I used quiet but then found reddit comment about strange fan behavior, and switched to balanced. It is still not quet - fans are always on if I even touch the key. But at least when I close the lid at night it stops.

w0utert, 2023-11-10:
I've been running it quiet all the time, set in BIOS and/or from KDE, in both modes the low fan speed threshold is slightly above ~35C. I switched to Balanced now as @mark-herbert42 suggested, initial impression is now that the fans kick in even earlier but remain at lower speed slightly longer? It's hard to say because of the pulsating behavior (once it kicks in, it will keep running for at least a minute or two, but when it finally stops it will kick back in at the slightest load).

I've also been playing with Smokeless_UMAF but changing the fan settings there does not seem to make a difference, it sems the fan temperature/PWM table I can enter in manual mode is simply not used.

MrDuartePT, 2023-11-10:
Don’t touch fan stuff in there probably you set the level1 in the bios did makes the fan allways on

w0utert, 2023-11-10:
I have changed it back to auto already, behavior in quiet mode seems like it was before…

Do you think there is any change someone will figure out how to get working fan control from Linux, or is everything in these laptops completely hidden away in firmware to the point only the manufacturer can change anything about it? It’s very disappointing, I would even disable boost or limit TDP or clockspeeds, whatever it takes to make this thing silent under light load, but it seems none of the typical fan/cpu/power/acpi tools work for anything on this thing :-(

mark-herbert42, 2023-11-10:
This laptop seems to be the worst in linux compatibilty I ever had.  Terrible sound - more or less solved now but still need to download unofficial firmware and install rare almost unknown daemon developed 2 days ago for Legion and patched for Yoga today. 
Suspend to ram completely broken.
 Fans always on and quet mode is worse than performance.  there is still some hope that future firmware may fix it , but at least the yesterday firmware release (released only for Chinese versions so far) did not do any good except of some useless secutity fixups.

w0utert, 2023-11-10:
Well the power management flaws are the same in Linux and Windows so that should give some hope there will be improvements, I would just like to be able to control the fans directly from Linux instead of handing full control over to the Lenovo bios…

The other Linux issues will be resolved in time I’m sure, Phoenix platform is still relatively new so some rough edges are to be expected.

mark-herbert42, 2023-11-10:
The bad thing it is not issue with phoenix platform - other phoenix laptops suspend to ram ok. That's the issue of this particular model and buggy firmware  
https://gitlab.freedesktop.org/drm/amd/-/issues/2812

closed 2026-09-12 by github-actions[bot]