johnfanv2/LenovoLegionLinux #128: Legion slim 7 no audio from internal speakers

Legion2016, 2023-10-12:
I found this patch that might help some Lenovo users like me that have no audio from internal speakers.  I know nothing about patches and I have no idea how to add a patch and was hoping someone could look at it.
https://bugzilla.kernel.org/show_bug.cgi?id=216194#c66

lenovo-7i-gen7-sound-6.2.0-rc3-0.0.5b-002.patch

trivoldus28, 2023-10-14:
Thanks! I was just going to look up why the speakers don't work. Regarding the patch you'd need to patch the kernel and build it. There are some directions in that thread you linked.

MrDuartePT, 2023-10-19:
This issue is out of the scope of this kernel module.
The kernel module only try to provide the same features you will get with Lenovo Legion Toolkit on windows.
For hardware issues discuss in the bugzilla or lenovo community.

For applying patch you need to dowload kernel 6.2.0 source, and run this commands inside the folder
```
#Apply patch
patch -p1 < lenovo-7i-gen7-sound-6.2.0-rc3-0.0.5b-002.patch #be sure to have the patch inside the folder

#Copy distro kernel config
zcat /proc/config.gz >/usr/src/linux/.config

#Build the kernel
make --jobs=16
make modules_install
make install
```
The you just need to rebuild mkinitcpio, grub or other thing you use and reboot in the new kernel

closed 2023-10-19 by MrDuartePT