johnfanv2/LenovoLegionLinux #132: Adding support for Lenovo Legion Slim 7 16APH8

trivoldus28, 2023-10-19:
Model and Debug Info

Model name: Lenovo Legion Slim 7 16APH8
CPU model: AMD Ryzen 7 7840HS
GPU model: NVIDIA RTX 4600
Keyboard backlight: RGB per key with off/medium/bright

Output of sudo dmidecode -t system:
```
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82Y4
        Version: Legion Slim 7 16APH8
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82Y4_BU_idea_FM_Legion Slim 7 16APH8
        Family: Legion Slim 7 16APH8

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

`-t bios`

```
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: M1CN34WW
        Release Date: 08/07/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.34
        Firmware Revision: 1.34

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

`Fan curve`

cat: /sys/kernel/debug/legion/fancurve: No such file or directory

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/12889007/fwts_wmi.log)

[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/12889066/acpi_re.tar.gz)

Thanks for all the work!

Marwan-Dalaty, 2023-10-19:
Hi @trivoldus28 , it worked with you the new patch for legion 7?

trivoldus28, 2023-10-19:
Let me try later today!

Marwan-Dalaty, 2023-10-19:
@trivoldus28 he added new patch here https://github.com/johnfanv2/LenovoLegionLinux/commit/03a2ec7db61aed847780bc3fca9c40b61a992fab
please try it and let me know if it will work with you.
thanks

trivoldus28, 2023-10-20:
@Marwan-Dalaty I'm getting this error. Seems like my model's BIOS is `M1CN` vs the patch's `M0CN`
```
[Sun Oct 22 03:35:05 2023] legion_laptop: Loading legion_laptop
[Sun Oct 22 03:35:05 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sun Oct 22 03:35:05 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sun Oct 22 03:35:05 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sun Oct 22 03:35:05 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[Sun Oct 22 03:35:05 2023] legion PNP0C09:00: Module not usable for this laptop because it is not in allowlist. Notify the maintainer if you want to add your device or force load with param force.
[Sun Oct 22 03:35:05 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Sun Oct 22 03:35:05 2023] legion: probe of PNP0C09:00 failed with error -12
```

trivoldus28, 2023-10-20:
If I changed the code to use `M0CN`, it still results in error:
```
[Sun Oct 22 03:40:26 2023] legion_laptop: Loading legion_laptop
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sun Oct 22 03:40:26 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: Using configuration for system: M1CN
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: Skipping ACPI _STA check
[Sun Oct 22 03:40:26 2023] legion_laptop: Creating RAM access to embedded controller
[Sun Oct 22 03:40:26 2023] legion_laptop: Successfully mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x000000006077e8a9
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: Read embedded controller ID 0x5507
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: Expected EC chip id 0x8227 but read 0x5507
[Sun Oct 22 03:40:26 2023] legion_laptop: Unloading legion ecram
[Sun Oct 22 03:40:26 2023] legion_laptop: Unloading legion ecram done
[Sun Oct 22 03:40:26 2023] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x000000006077e8a9
[Sun Oct 22 03:40:26 2023] legion_laptop: Unloading legion shared
[Sun Oct 22 03:40:26 2023] legion_laptop: Unloading legion shared done
[Sun Oct 22 03:40:26 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Sun Oct 22 03:40:26 2023] legion: probe of PNP0C09:00 failed with error -12
```

trivoldus28, 2023-10-20:
And if I modified the EC ID of the same config to be `0x5507`:
```
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sun Oct 22 03:43:33 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Using configuration for system: M1CN
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Skipping ACPI _STA check
[Sun Oct 22 03:43:33 2023] legion_laptop: Creating RAM access to embedded controller
[Sun Oct 22 03:43:33 2023] legion_laptop: Successfully mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x00000000cbf7b8aa
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Read embedded controller ID 0x5507
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Creating debugfs interface
[Sun Oct 22 03:43:33 2023] legion_laptop: Creating sysfs interface
[Sun Oct 22 03:43:33 2023] legion_laptop: Creating hwmon interface
[Sun Oct 22 03:43:33 2023] legion_laptop: Creating platform profile support
[Sun Oct 22 03:43:33 2023] legion_laptop: Init WMI driver support
[Sun Oct 22 03:43:33 2023] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[Sun Oct 22 03:43:33 2023] legion_laptop: Init keyboard backlight LED driver
[Sun Oct 22 03:43:33 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Oct 22 03:43:33 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 3, but got 0
[Sun Oct 22 03:43:33 2023] legion_laptop: Error reading keyboard brightness
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Failed to init keyboard backlight LED driver. Skipping ...
[Sun Oct 22 03:43:33 2023] legion_laptop: Init Y-Logo LED driver
[Sun Oct 22 03:43:33 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Oct 22 03:43:33 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Oct 22 03:43:33 2023] legion_laptop: Init IO-Port LED driver
[Sun Oct 22 03:43:33 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Oct 22 03:43:33 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[Sun Oct 22 03:43:33 2023] legion_laptop: Error reading brightness for light: 5
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: Failed to init IO-Port LED driver. Skipping ...
[Sun Oct 22 03:43:33 2023] legion PNP0C09:00: legion_laptop loaded for this device
```

trivoldus28, 2023-10-20:
There are two configs with EC ID = 0x5507 (`model_kwcn` and `model_lpcn`) and neither of them worked. Error is something like:
```
[Thu Oct 19 21:20:15 2023] legion_laptop: Loading legion_laptop
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Thu Oct 19 21:20:15 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: Using configuration for system: M1CN
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: Could not evaluate ACPI _STA
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: Could not init ACPI access: -5
[Thu Oct 19 21:20:15 2023] legion_laptop: Unloading legion shared
[Thu Oct 19 21:20:15 2023] legion_laptop: Unloading legion shared done
[Thu Oct 19 21:20:15 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Thu Oct 19 21:20:15 2023] legion: probe of PNP0C09:00 failed with error -5
```

If I skip the ACPI check, the module loads but the fan curve is either empty or all zeros.

MrDuartePT, 2023-10-20:
thanks for pointing this out i gonna fix it

MrDuartePT, 2023-10-20:
it seems like multiple variation depending of the config choose, dint expect to the bios change between config (between cpu brand is normal).

MrDuartePT, 2023-10-20:
@trivoldus28 try see if works now
9006d7763736197a676bdeda86b696c96fea1142

MrDuartePT, 2023-10-20:
do me a favor test all the feature if works close the issue

trivoldus28, 2023-10-20:
Thanks for the prompt response! Will give it a try.

Marwan-Dalaty, 2023-10-20:
HI @trivoldus28 can you advise how you are installing the tool, as I'm always facing this error E: Unable to locate package python3-darkdetect
 and I cant find python3-darkdetect to install it

MrDuartePT, 2023-10-20:
> HI @trivoldus28 can you advise how you are installing the tool, as I'm always facing this error E: Unable to locate package python3-darkdetect and I cant find python3-darkdetect to install it trying to run the below command sudo apt-get install -y make gcc linux-headers-$(uname -r) build-essential git lm-sensors wget python3-pyqt5 python3-yaml python3-venv python3-pip python3-argcomplete python3-darkdetect

use the repo ppa, See the folder package_repo.

trivoldus28, 2023-10-21:
@MrDuartePT Pulled, and I'm getting some errors with reading backlight and keyboard brightness:
```
[Sat Oct 21 18:34:28 2023] legion_laptop: Loading legion_laptop
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sat Oct 21 18:34:28 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN34WW
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Using configuration for system: M1CN
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Skipping ACPI _STA check
[Sat Oct 21 18:34:28 2023] legion_laptop: Creating RAM access to embedded controller
[Sat Oct 21 18:34:28 2023] legion_laptop: Successfully mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x000000009973da7e
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Read embedded controller ID 0x5507
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Creating debugfs interface
[Sat Oct 21 18:34:28 2023] legion_laptop: Creating sysfs interface
[Sat Oct 21 18:34:28 2023] legion_laptop: Creating hwmon interface
[Sat Oct 21 18:34:28 2023] legion_laptop: Creating platform profile support
[Sat Oct 21 18:34:28 2023] legion_laptop: Init WMI driver support
[Sat Oct 21 18:34:28 2023] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[Sat Oct 21 18:34:28 2023] legion_laptop: Init keyboard backlight LED driver
[Sat Oct 21 18:34:28 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:34:28 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 3, but got 0
[Sat Oct 21 18:34:28 2023] legion_laptop: Error reading keyboard brightness
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Failed to init keyboard backlight LED driver. Skipping ...
[Sat Oct 21 18:34:28 2023] legion_laptop: Init Y-Logo LED driver
[Sat Oct 21 18:34:28 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:34:28 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:34:28 2023] legion_laptop: Init IO-Port LED driver
[Sat Oct 21 18:34:28 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:34:28 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[Sat Oct 21 18:34:28 2023] legion_laptop: Error reading brightness for light: 5
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: Failed to init IO-Port LED driver. Skipping ...
[Sat Oct 21 18:34:28 2023] legion PNP0C09:00: legion_laptop loaded for this device
```

The fan curve is also empty when I run `sudo cat /sys/kernel/debug/legion/fancurve`:
```
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 1074025472
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

Marwan-Dalaty, 2023-10-21:
HI @MrDuartePT , I can't find the PPA for python-darkdetect, and all what I can find online is to use it with pip.
Please can you help more to get it done and test the above.
@trivoldus28 sorry to bother you, but I appreciate your help too if you can help me to do it

MrDuartePT, 2023-10-21:
It here
https://github.com/johnfanv2/LenovoLegionLinux/tree/main/package_repo

MrDuartePT, 2023-10-21:
> @MrDuartePT Pulled, and I'm getting some errors with reading backlight and keyboard brightness:
>
> ```
> [Sat Oct 21 18:34:28 2023] legion_laptop: Loading legion_laptop
> [Sat Oct 21 18:34:28 2023] legion [...]

New to see this @johnfanv2 i think is some problem with the 2023 model.

Marwan-Dalaty, 2023-10-21:
@MrDuartePT i'm getting the below message after I run make reloadmodule
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.035:10036): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.login1.Manager" member="PrepareForSleep" name=":1.18" mask="receive" pid=4782 label="snap.slack.slack" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.035:10037): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.login1.Manager" member="PrepareForSleep" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.035:10038): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.039:10039): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.039:10040): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.043:10041): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.043:10042): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:10 2023] audit: type=1107 audit(1697840926.123:10043): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid=1064 peer_label="unconfined"
                            exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[Sat Oct 21 16:32:11 2023] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x1f
[Sat Oct 21 16:32:11 2023] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[Sat Oct 21 16:32:11 2023] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[Sat Oct 21 16:32:11 2023] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x0
[Sat Oct 21 16:32:12 2023] pcieport 0000:00:1c.0: pciehp: Slot(0): No link
[Sat Oct 21 16:32:14 2023] wlp0s20f3: authenticate with c4:41:1e:98:c8:51
[Sat Oct 21 16:32:14 2023] wlp0s20f3: 80 MHz not supported, disabling VHT
[Sat Oct 21 16:32:14 2023] wlp0s20f3: send auth to c4:41:1e:98:c8:51 (try 1/3)
[Sat Oct 21 16:32:14 2023] wlp0s20f3: authenticated
[Sat Oct 21 16:32:14 2023] wlp0s20f3: associate with c4:41:1e:98:c8:51 (try 1/3)
[Sat Oct 21 16:32:14 2023] wlp0s20f3: RX AssocResp from c4:41:1e:98:c8:51 (capab=0x1431 status=0 aid=3)
[Sat Oct 21 16:32:14 2023] wlp0s20f3: associated
[Sat Oct 21 16:32:14 2023] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[Sat Oct 21 16:32:18 2023] kauditd_printk_skb: 1 callbacks suppressed
[Sat Oct 21 16:32:18 2023] audit: type=1107 audit(1697840933.607:10045): pid=1008 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/login1" interface="org.freedesktop.DBus.Properties" member="PropertiesChanged" name=":1.18" mask="receive" pid=620168 label="snap.firefox.firefox" peer_pid [... cut ...]
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion ecram done
[Sat Oct 21 18:12:10 2023] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x00000000835c3b45
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion shared
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion shared done
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Sat Oct 21 18:12:10 2023] legion: probe of PNP0C09:00 failed with error -12
bash -c "./issue-warning.sh"

MrDuartePT, 2023-10-21:
Pls add the dmesg as a code block. It impossible to read

MrDuartePT, 2023-10-21:
@Marwan-Dalaty if you what me to solve your problems I need frist the output of the command like it was a new issue. And pls not post the whole dmesg output only the lenovo_laptop part.

MrDuartePT, 2023-10-21:
For what I see it just the ec id is wrong you can try to forcemodulereload

Marwan-Dalaty, 2023-10-21:
HI @MrDuartePT , running the below commands, 
```
git clone https://github.com/johnfanv2/LenovoLegionLinux.git
cd `LenovoLegionLinux/kernel_module`
make
sudo make reloadmodule
```
gave me the below output 
```
[Sat Oct 21 18:12:10 2023] legion_laptop: legion_laptop exit
[Sat Oct 21 18:12:10 2023] legion_laptop: Loading legion_laptop
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: legion_laptop platform driver probing
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y3; DMI_BIOS_VERSION:M0CN27WW
[Sat Oct 21 18:12:10 2023] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y3; DMI_BIOS_VERSION:M0CN27WW
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: Using configuration for system: M0CN
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: Skipping ACPI _STA check
[Sat Oct 21 18:12:10 2023] legion_laptop: Creating RAM access to embedded controller
[Sat Oct 21 18:12:10 2023] legion_laptop: Successfully mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x00000000835c3b45
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: Read embedded controller ID 0x5507
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: Expected EC chip id 0x8227 but read 0x5507
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion ecram
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion ecram done
[Sat Oct 21 18:12:10 2023] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x00000000835c3b45
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion shared
[Sat Oct 21 18:12:10 2023] legion_laptop: Unloading legion shared done
[Sat Oct 21 18:12:10 2023] legion PNP0C09:00: legion_laptop not loaded for this device
[Sat Oct 21 18:12:10 2023] legion: probe of PNP0C09:00 failed with error -12
bash -c "./issue-warning.sh"
```

how can I do forcemodulereload?

MrDuartePT, 2023-10-21:
use make forcereloadmodule

Marwan-Dalaty, 2023-10-21:
erro after writing the above command

`[Sat Oct 21 18:35:34 2023] legion_laptop: Init keyboard backlight LED driver
[Sat Oct 21 18:35:34 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:35:34 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 3, but got 0
[Sat Oct 21 18:35:34 2023] legion_laptop: Error reading keyboard brightness
[Sat Oct 21 18:35:34 2023] legion PNP0C09:00: Failed to init keyboard backlight LED driver. Skipping ...
[Sat Oct 21 18:35:34 2023] legion_laptop: Init Y-Logo LED driver
[Sat Oct 21 18:35:34 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:35:34 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:35:34 2023] legion_laptop: Init IO-Port LED driver
[Sat Oct 21 18:35:34 2023] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sat Oct 21 18:35:34 2023] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[Sat Oct 21 18:35:34 2023] legion_laptop: Error reading brightness for light: 5
[Sat Oct 21 18:35:34 2023] legion PNP0C09:00: Failed to init IO-Port LED driver. Skipping ...
[Sat Oct 21 18:35:34 2023] legion PNP0C09:00: legion_laptop loaded for this device
bash -c "./issue-warning.sh"
`

MrDuartePT, 2023-10-21:
it look alright.
what ```sudo cat /sys/kernel/debug/legion/fancurve``` give you

Marwan-Dalaty, 2023-10-21:
`EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4244963348
temperature access method: 5
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 52
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 52
GPU temperature error: 0
GPU temperature: 46
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 46
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 46
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2100
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2100
2 fanspeed error: 0
2 fanspeed: 2100
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2100
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2100
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 230
powermode ACPI error: -5
powermode ACPI: -1073458176
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

`

Marwan-Dalaty, 2023-10-21:
```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4244963348
temperature access method: 5
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 52
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 52
GPU temperature error: 0
GPU temperature: 46
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 46
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 46
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2100
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2100
2 fanspeed error: 0
2 fanspeed: 2100
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2100
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2100
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 230
powermode ACPI error: -5
powermode ACPI: -1073458176
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

MrDuartePT, 2023-10-21:
yup is the something that @trivoldus28 we need to give a look at this later.
In the 2023 model the fan control change

Marwan-Dalaty, 2023-10-21:
ok but keyboard light should work? as it is not working

MrDuartePT, 2023-10-21:
if you press fn+space the keyboard light should turn on.
For that feature the kernel module is not needed

Marwan-Dalaty, 2023-10-21:
thanks @MrDuartePT , keyboard light is working. Hope you can find a solution for the fan soon.

rogowskys, 2023-10-25:
This patch works with my 16APH8, however I'm having some big problems still, with or without the patch.  I have tested with Nobara 38 and Fedora 39 (beta) and both exhibit the same behaviour:

1. (On battery), plug in AC cord and system shuts off immediately.
2. (On battery or AC) Changing screen brightness shuts the laptop off.  This may happen after one attempt to dim, or after 4 or 5 keypresses. 

The led around the power button will flash rapidly for a few seconds.  It powers back on, as per a cold boot.

MrDuartePT, 2023-10-26:
If you have this problem without the kernel module loaded this is not a problem with the module. I would advise you to reset the ec (long press the power button without any the power adapter and other cable connected).
If still happen try update the kernel or use other distro.

MrDuartePT, 2023-10-26:
> ```
> EC Chip ID: 5507
> EC Chip Version: 2a4
> legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
> legion_laptop ec_readonly: 0
> ACPI CFG error: 0
> ACPI CFG: [...]

Can you give me the hexdumps for the 3 power modes: `hexdump -c /sys/kernel/debug/legion/ecmemory`

I gonna try give a look from 2 weak from now

trivoldus28, 2023-10-28:
@MrDuartePT This one is for Slim 7 16APH8.
```
sudo cat /sys/kernel/debug/legion/ecmemoryram | hexdump -C
00000000  00 00 00 8c 00 00 00 00  01 02 80 10 00 00 00 00  |................|
00000010  00 00 2e 2c 2d 28 2b 29  72 00 00 00 40 62 00 80  |...,-(+)r...@b..|
00000020  e6 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 08 00 6e 77  6f 64 61 00 32 30 32 32  |......nwoda.2022|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  00 00 04 c8 40 00 00 00  2b 00 00 08 00 2a 00 00  |....@...+....*..|
000000b0  00 29 00 00 2b 00 00 00  00 00 00 0f 00 7a 00 00  |.)..+........z..|
000000c0  00 02 d3 05 7d 09 66 3d  c8 3c 06 27 82 27 0f 1f  |....}.f=.<.'.'..|
000000d0  60 11 60 11 95 11 82 27  f5 03 59 11 00 00 00 00  |`.`....'..Y.....|
000000e0  00 00 00 58 0f 5c 0f 5a  0f 58 0f 00 00 00 00 00  |...X.\.Z.X......|
000000f0  00 00 00 00 00 00 88 45  07 05 00 87 00 00 00 00  |.......E........|
00000100  00 00 00 00 00 00 05 00  3e 00 38 40 13 3a 50 15  |........>.8@.:P.|
00000110  3c 50 24 44 50 24 44 50  24 44 50 24 44 50 24 44  |<P$DP$DP$DP$DP$D|
00000120  69 24 5e 6c 24 00 00 00  00 00 00 00 00 00 00 3e  |i$^l$..........>|
00000130  00 38 40 13 3a 50 15 3c  50 24 44 50 24 44 50 24  |.8@.:P.<P$DP$DP$|
00000140  44 50 24 44 50 24 44 69  24 5e 6c 24 00 00 00 00  |DP$DP$Di$^l$....|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 01 00  00 00 00 00 00 00 00 00  |................|
00000210  02 0c 0e 11 15 15 15 15  15 2a 00 00 00 00 00 00  |.........*......|
00000220  5a a5 00 e6 00 00 00 00  00 00 00 45 f6 00 00 00  |Z..........E....|
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000240  01 23 00 00 00 00 00 20  04 00 02 00 00 00 00 00  |.#..... ........|
00000250  00 00 00 40 00 00 00 00  00 00 00 00 00 00 00 00  |...@............|
00000260  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000270  e7 db 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000300  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
```

```
sudo cat /sys/kernel/debug/legion/ecmemory | hexdump -C
00000000  00 00 00 8c 00 00 00 00  01 02 80 10 00 00 00 00  |................|
00000010  00 00 2e 2c 2e 29 2c 2a  72 00 00 00 40 62 00 80  |...,.),*r...@b..|
00000020  e6 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 08 00 6e 77  6f 64 61 00 32 30 32 32  |......nwoda.2022|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  00 00 00 c8 40 00 00 00  23 00 00 08 00 2a 00 00  |....@...#....*..|
000000b0  00 2a 00 00 23 00 00 00  00 00 00 0f 00 7a 00 00  |.*..#........z..|
000000c0  00 02 53 06 7d 09 8c 3d  c8 3c 06 27 82 27 10 1f  |..S.}..=.<.'.'..|
000000d0  60 11 5f 11 95 11 82 27  f5 03 59 11 00 00 00 00  |`._....'..Y.....|
000000e0  00 00 00 62 0f 65 0f 63  0f 61 0f 00 00 00 00 00  |...b.e.c.a......|
000000f0  00 00 00 00 00 00 88 45  02 0d 00 85 00 00 00 00  |.......E........|
00000100  00 00 00 00 00 00 05 00  3e 00 38 40 13 3a 50 15  |........>.8@.:P.|
00000110  3c 50 24 44 50 24 44 50  24 44 50 24 44 50 24 44  |<P$DP$DP$DP$DP$D|
00000120  69 24 5e 6c 24 00 00 00  00 00 00 00 00 00 00 3e  |i$^l$..........>|
00000130  00 38 40 13 3a 50 15 3c  50 24 44 50 24 44 50 24  |.8@.:P.<P$DP$DP$|
00000140  44 50 24 44 50 24 44 69  24 5e 6c 24 00 00 00 00  |DP$DP$Di$^l$....|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000200  00 00 00 00 00 00 01 00  00 00 00 00 00 00 00 00  |................|
00000210  02 0c 0e 11 15 15 15 15  15 2a 00 00 00 00 00 00  |.........*......|
00000220  5a a5 00 e6 00 00 00 00  00 00 00 45 f6 00 00 00  |Z..........E....|
00000230  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000240  01 23 00 00 00 00 00 20  04 00 02 00 00 00 00 00  |.#..... ........|
00000250  00 00 00 40 00 00 00 00  00 00 00 00 00 00 00 00  |...@............|
00000260  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000270  e7 db 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000280  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000300
```

cy1der, 2023-12-07:
Any updates on this? My situation is a carbon copy of #99.

easymod0, 2023-12-19:
I have the same laptop but with a newer bios revision (36 vs 34) and the module will not load for me:

[    1.562681] legion_laptop: legion_laptop starts loading
[    1.562732] legion PNP0C09:00: legion_laptop platform driver probing
[    1.562734] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN36WW
[    1.562735] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82Y4; DMI_BIOS_VERSION:M1CN36WW
[    1.562739] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[    1.562741] legion PNP0C09:00: Module not useable for this laptop because it is not in allowlist. Notify maintainer if you want to add your device or force load with param force.
[    1.562742] legion PNP0C09:00: legion_laptop not loaded for this device
[    1.562790] legion: probe of PNP0C09:00 failed with error -12

biggusivanus, 2026-05-02:
### EC memory dumps for Legion Slim 7 16APH8 (82Y4) across all power profiles

Model: Legion Slim 7 16APH8 (82Y4)
BIOS: M1CN series
OS: CachyOS (Arch), kernel 7.0.2-2-cachyos
Note: Module doesn't load without allowlist entry. EC dumps obtained via ec_sys + /sys/kernel/debug/ec/ec0/io.

**Finding**: max-power in ACPI platform_profile maps to the custom mode EC slot (purple LED), not a variant of performance mode. This is confirmed by the LED color change and fan behavior matching the Windows Vantage custom profile I had previously configured (a silent profile copied from quiet mode with a conservative fan curve).

**Key register observations from diffs (all 4 modes):**

| Offset | low-power | balanced | performance | max-power | Notes |
|--------|-----------|----------|-------------|-----------|-------|
| 0x0B | 0x50 | 0x10 | 0x30 | 0xF0 | Possible TDP/power value |
| 0xA2 | 0x00 | 0x04 | 0x04 | 0x00 | Flag set in balanced+performance, clear in low-power+max-power |
| 0xA8 | 0x23 (35°C) | 0x2C (44°C) | 0x2D (45°C) | 0x23 (35°C) | Likely temperature threshold |
| 0xB1 | 0x2A (42°C) | 0x2B (43°C) | 0x2C (44°C) | 0x28 (40°C) | Likely temperature threshold |
| 0xB4 | 0x23 (35°C) | 0x2C (44°C) | 0x2D (45°C) | 0x23 (35°C) | Mirrors 0xA8 |
| 0xF8 | 0x04 | 0x1D | 0x07 | 0x08 | Likely dynamic/sensor value, not a profile index |

Note: 0xF8 initially looked like a profile index but balanced returns 0x1D (29), breaking the pattern -- likely a live counter or sensor reading.

**Working acpi_call paths for rapid charge:**

Query:   
```echo "\_SB.PCI0.LPC0.EC0.FCGM" | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call```

Enable:  
```echo "\_SB.PCI0.LPC0.EC0.VPC0.SBMC 0x07" | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call```

Disable: 
```echo "\_SB.PCI0.LPC0.EC0.VPC0.SBMC 0x08" | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call```

Below are the relevant hex dumps as well as the decompiled DSDT.

**Raw hex dumps:**
<details>
<summary>ec_low_power.hex</summary>

```
00000000  00 01 00 8c 00 00 00 08  01 02 90 50 00 00 00 00  |...........P....|
00000010  00 00 2a 2f 2d 2b 2e 2b  72 00 00 00 d0 62 80 00  |..*/-+.+r....b..|
00000020  e6 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 79 00 6e 77  6f 64 61 00 32 30 32 32  |....y.nwoda.2022|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  00 00 00 c8 40 00 00 00  23 00 00 08 00 2a 00 00  |....@...#....*..|
000000b0  00 2a 00 00 23 00 00 00  00 00 00 0d 00 7a 00 00  |.*..#........z..|
000000c0  00 00 f7 1b 53 03 22 41  c8 3c 06 27 23 23 50 20  |....S."A.<.'##P |
000000d0  00 00 00 00 95 0f 23 23  83 03 59 11 00 04 00 00  |......##..Y.....|
000000e0  00 00 00 48 10 48 10 48  10 49 10 00 00 00 00 00  |...H.H.H.I......|
000000f0  00 00 00 00 00 00 60 45  04 0d 00 ff ff 00 00 00  |......`E........|
00000100
```

</details>

<details>
<summary>ec_balanced.hex</summary>

```
00000000  00 01 00 8c 00 00 00 08  01 02 90 10 00 00 00 00  |................|
00000010  00 00 2c 30 2d 2b 2e 2c  72 00 00 00 d0 62 80 00  |..,0-+.,r....b..|
00000020  e6 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 79 00 6e 77  6f 64 61 00 32 30 32 32  |....y.nwoda.2022|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... 36 lines cut ...]
<summary>ec_max_power.hex</summary>

```
00000000  00 01 00 8c 00 00 00 08  01 02 90 f0 00 00 00 00  |................|
00000010  00 00 2d 2c 2d 2b 2d 2a  72 00 00 00 d0 62 80 00  |..-,-+-*r....b..|
00000020  e6 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000060  00 80 16 17 79 00 6e 77  6f 64 61 00 32 30 32 32  |....y.nwoda.2022|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  00 00 00 c8 40 00 00 00  23 00 00 08 00 2a 00 00  |....@...#....*..|
000000b0  00 28 00 00 23 00 00 00  00 00 00 0d 00 7a 00 00  |.(..#........z..|
000000c0  00 00 f7 1b 53 03 22 41  c8 3c 06 27 23 23 50 20  |....S."A.<.'##P |
000000d0  00 00 00 00 95 0f 23 23  83 03 59 11 00 04 00 00  |......##..Y.....|
000000e0  00 00 00 49 10 48 10 48  10 49 10 00 00 00 00 00  |...I.H.H.I......|
000000f0  00 00 00 00 00 00 60 45  08 0d 00 ff ff 00 00 00  |......`E........|
00000100
```

</details>

**Fully decompiled DSDT:** [dsdt.dsl](https://gist.github.com/biggusivanus/94a4a76aebff58eab32f58002f5cc402)

Additionally, I would like to disclose that I was quite heavily assisted by an LLM for my outputs, I am quite new to linux (this is my third or fourth time trying it each time with a different distro, but so far this is the happiest I have been). Though I have always been a poweruser, this is my first time contributing on github issues so I apologise in advance if this was the wrong place to provide this information.

closed 2026-09-12 by github-actions[bot]