johnfanv2/LenovoLegionLinux #133: Legion 7 pro AMD Support

lyckligtax, 2023-10-23:
The fans are driving me nuts. Will do anything to help. Also can we donate somewhere?

Module compilation and loading without issues  
Fancurve is all zero  
EC Chip ID differs

Distribution: Endeavor OS
Model name: Lenovo Legion 7 Pro 16ARX8H
CPU model:  AMD 7945HX
GPU model: NVIDIA 4080
Keyboard backlight: RGB
Light in lid or logo: no
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
System Information
	Manufacturer: LENOVO
	Product Name: 82WS
	Version: Legion Pro 7 16ARX8H
	Serial Number: 
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WS_BU_idea_FM_Legion Pro 7 16ARX8H
	Family: Legion Pro 7 16ARX8H
```

Output of `sudo dmidecode -t bios`:
```text
BIOS Information
	Vendor: LENOVO
	Version: LPCN45WW
	Release Date: 08/02/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.45
	Firmware Revision: 1.45
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 5
CPU temperature error: 0
CPU temperature: 71
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 71
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 71
GPU temperature error: 0
GPU temperature: 42
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 42
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 42
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2900
1 fanspeed EC error: 0
1 fanspeed EC: 48
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2900
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2900
2 fanspeed error: 0
2 fanspeed: 3000
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 3000
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 3000
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 16
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
2500	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
2500	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
```

MrDuartePT, 2023-10-26:
You did try change the fancurve in the gui?
The gen 8 behave I kinda strange and for some model we don’t get any point and other the point. 
But we still trying to found how to fix this. 

you can give me the hexdump bettwen the 3 power modes: `hexdump -c /sys/kernel/debug/legion/ecmemory`

lyckligtax, 2023-10-26:
> You did try change the fancurve in the gui?
Yes, nothing changed

> you can give me the hexdump bettwen the 3 power modes:
Sure, I switched the profiles via Gnome Switcher

ecmemoryram@balanced
```
00000000  00 01 00 02 00 00 12 00  0f 0c 00 00 00 00 18 28  |...............(|
00000010  00 00 80 00 05 00 10 06  00 01 06 00 00 00 00 04  |................|
00000020  10 00 01 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 40 02  |..............@.|
00000040  20 03 02 00 50 01 00 00  00 00 00 00 54 57 00 00  | ...P.......TW..|
00000050  4a 01 2c 01 00 00 c0 81  00 07 00 0a 00 00 00 00  |J.,.............|
00000060  00 80 02 05 00 00 00 08  00 00 00 00 01 00 20 01  |.............. .|
00000070  00 02 10 01 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 0b 00 00 00  e0 00 42 00 81 00 01 53  |..........B....S|
00000090  4d 50 00 32 30 32 32 00  4c 32 32 4d 34 50 43 31  |MP.2022.L22M4PC1|
000000a0  00 09 00 8e 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000b0  2c 2a 00 27 26 00 28 23  02 11 00 00 00 80 01 00  |,*.'&.(#........|
000000c0  70 00 a6 28 dc 00 27 45  a0 3c 06 27 a6 28 64 00  |p..(..'E.<.'.(d.|
000000d0  00 00 00 00 00 00 00 16  00 00 00 00 00 00 00 00  |................|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 01 02 00  |................|
000000f0  03 00 00 00 2c 57 00 00  00 00 00 00 00 00 11 00  |....,W..........|
00000100  80 00 00 04 10 00 02 02  72 1e 77 77 77 32 77 81  |........r.www2w.|
00000110  00 00 ff ff 00 00 00 00  00 00 3d 35 00 00 00 00  |..........=5....|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000130  00 00 0c 11 00 00 00 00  92 00 00 00 00 00 00 00  |................|
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  00 19 00 50 00 00 00 00  00 00 00 00 00 00 00 00  |...P............|
000001e0  30 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |0...............|
000001f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000200  00 00 00 00 00 00 00 00  00 00 00 00 00 04 00 00  |................|
00000210  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000220  00 00 00 00 00 00 00 38  00 e1 02 01 a7 28 00 00  |.......8.....(..|
00000230  01 00 00 00 00 00 00 19  19 00 00 00 00 00 00 02  |................|
00000240  00 00 00 00 00 00 00 1e  00 00 03 00 00 00 00 00  |................|
00000250  3c 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |<...............|
00000260  58 34 58 50 33 39 43 30  30 36 57 00 00 00 00 00  |X4XP39C006W.....|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002f0  00 00 00 00 00 00 2a 26  01 03 02 00 00 00 00 00  |......*&........|
00000300  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000320  01 20 1f 00 ff ff 20 10  00 00 1b 20 10 00 00 10  |. .... .... ....|
00000330  02 20 10 00 00 10 20 1f  00 ff ff 20 10 00 00 1b  |. .... .... ....|
00000340  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000380  ff 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000390  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000003e0  ff 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000003f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000400  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000480  40 eb bf 4c 1e 9b 11 b2  a8 5c a6 b4 64 45 db 42  |@..L.....\..dE.B|
00000490  ff fb ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000004a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000500  00 01 00 00 00 00 00 08  00 00 00 00 00 00 00 00  |................|
00000510  00 40 00 20 00 00 00 00  01 00 20 01 00 02 10 01  |.@. ...... .....|
00000520  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000540  00 04 05 db 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000550  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000560  00 00 00 00 00 00 00 00  00 00 00 00 00 40 00 00  |.............@..|
00000570  80 00 1e 00 00 00 00 00  00 00 00 00 10 00 00 00  |................|
00000580  00 00 00 00 00 00 00 00  80 00 00 00 00 00 ff 00  |................|
00000590  00 00 00 00 00 00 00 00  00 00 00 00 28 00 00 00  |............(...|
[... 242 lines cut ...]
000000c0  70 00 a6 28 dc 00 26 45  a0 3c 06 27 a6 28 64 00  |p..(..&E.<.'.(d.|
000000d0  00 00 00 00 00 00 00 17  00 00 00 00 00 00 00 00  |................|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 01 02 00  |................|
000000f0  03 00 00 00 2c 57 00 00  00 00 00 00 00 00 11 00  |....,W..........|
00000100  80 00 00 04 10 00 02 02  72 1e 77 77 77 32 77 81  |........r.www2w.|
00000110  00 00 ff ff 00 00 00 00  00 00 3d 35 00 00 00 00  |..........=5....|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000130  00 00 05 38 00 00 00 00  92 00 00 00 00 00 00 00  |...8............|
00000140  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001d0  00 19 00 50 00 00 00 00  00 00 00 00 00 00 00 00  |...P............|
000001e0  30 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |0...............|
000001f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
[... cut ...]
00000220  00 00 00 00 00 00 00 38  00 67 02 01 a7 28 00 00  |.......8.g...(..|
00000230  01 00 00 00 00 00 00 19  19 00 00 00 00 04 00 01  |................|
00000240  00 00 00 00 00 00 00 1e  00 00 03 00 00 00 00 00  |................|
00000250  3c 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |<...............|
00000260  58 34 58 50 33 39 43 30  30 36 57 00 00 00 00 00  |X4XP39C006W.....|
00000270  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000002f0  00 00 00 00 00 00 2d 28  01 03 02 00 00 00 00 00  |......-(........|
00000300
```

MrDuartePT, 2023-10-26:
I will need to do dump on mine and compare because it doesn't give me an info what is the temperature values.

88Dayone88, 2023-11-08:
Same Problem for me fancurve = 0 and you cant read out/change the power options. Do you need any other information?

MrDuartePT, 2023-11-08:
@88Dayone88 I was thinking the power option was working? Pls show what power option didn’t work for you

88Dayone88, 2023-11-08:
[image]
Read from HW doesnt work at all in any mode. Switching power modes works.

MrDuartePT, 2023-11-08:
You force load the module @88Dayone88?
If not, well it good to know that power mode option are also broken.
Unfortunately I will not have time to look at how to solve it before the end of the year.

vihu, 2023-11-22:
@lyckligtax Hello, I couldn't really find any other information about someone using Linux on Legion 7 Pro AMD other than this issue thread. Other than the fan noise, do you notice anything else not working? (Wifi, sound, external display etc all work?)

88Dayone88, 2023-11-24:
> You force load the module @88Dayone88? If not, well it good to know that power mode option are also broken. Unfortunately I will not have time to look at how to solve it before the end of the year.
> 
> unfortunately the hex dump not giving any information what values correspond to temperatures

Sorry for the late reply, im not force loading it.

> @lyckligtax Hello, I couldn't really find any other information about someone using Linux on Legion 7 Pro AMD other than this issue thread. Other than the fan noise, do you notice anything else not working? (Wifi, sound, external display etc all work?)

The only things not working are sound on the internal speakers (only sound from 1 side) and the stuff in this thread. Everything else works fine.

lyckligtax, 2023-11-25:
> @lyckligtax Hello, I couldn't really find any other information about someone using Linux on Legion 7 Pro AMD other than this issue thread. Other than the fan noise, do you notice anything else not working? (Wifi, sound, external display etc all work?)

Everything works but:

Sound: 
Using Kernel 6.6.2 I finally got working sound on one of the two speakers working. Though this only sometimes work

Fans: 
Cannot control them at all manually but sometimes they are completely silent despite bing under heavy load and temperatures rising. Sometimes they go full blast when nothing is happening.

Touchpad: 
Some gestures do no seam to work anymore. My Thinkpad eg can use 3 finger gestures, the Legion cannot.

Graphics Card:
The 4080 often gives me headaches with screen tearing but I think that might be a bad driver issue

88Dayone88, 2024-01-18:
Anything new about this issue? Can I support somehow? Do you need additional logs?

Syderitic, 2024-06-24:
[image]
I also suffer from this.

I can read the fan speed, but I can't set to anything else.

All the hardware works on my laptop (sound,graphics,etc) 

Distribution: Gentoo
Model name: Lenovo Legion 7 Pro 16ARX8H
CPU model: AMD 7945HX
GPU model: NVIDIA 4090
Keyboard backlight: RGB
Light in lid or logo: no
Light at IO-Ports at back: yes

System Information
        Manufacturer: LENOVO
        Product Name: 82WS
        Version: Legion Pro 7 16ARX8H
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WS_BU_idea_FM_Legion Pro 7 16ARX8H
        Family: Legion Pro 7 16ARX8H

dpn982, 2024-08-05:
I have this model as well with the RTX 4080 instead.  Everything except for fan curve works fine (including conservation mode, rapid charging, power profiles) under Fedora 40.  I am unable read or write the fan curve data.  Just curious if anyone has looked into this more, no rush I understand that this is a side project and not supported by Lenovo (although I wish they would come to their senses and give us at least some drivers).  Let me know if you want me to collect some data.

Ryvix, 2024-09-02:
Hello,

Thank you for working on this!

I'm using this model as well but with a 4090 and have similar issues. No internal sound, no fan curve. Other than some potential errors in the dmesg that I've added at the bottom below I think other things are working ok, at least so far that I've noticed.

I was able to get the sound working following steps I found online by running a script but it still goes out randomly and I have to run the script again, but I think this might be a kernel or firmware issue which might be out of the scope of this project.

I've also noticed sometimes when I first boot, the fans show as 0 rpm which doesn't seem good. Just now I booted up and they were at 0, but after a reboot they seem to work agian. However, I've noticed that it goes back to 0 if I press Fn + Q to cycle the power modes. Each time I press that it logs this in dmesg:
`legion_laptop: Fan event: legion type: 2;  acpi type: 3 (1=integer)`

**Model and Debug Info**

Distribution: CachyOS Linux 240818
Model name: Legion Pro 7 16ARX8H
CPU model: AMD Ryzen 9 7945HX
GPU model: NVIDIA GeForce RTX 4090
Keyboard backlight: RGB
Light in lid or logo: no
Light at IO-Ports at back: yes

`sudo dmidecode -t system`
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82WS
        Version: Legion Pro 7 16ARX8H
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WS_BU_idea_FM_Legion Pro 7 16ARX8H
        Family: Legion Pro 7 16ARX8H

Handle 0x0017, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

`sudo dmidecode -t bios`
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: LPCN52WW
        Release Date: 05/21/2024
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
[... 126 lines cut ...]
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
=====================
```

I noticed some red notices in dmesg on boot that I'm not sure are useful but may as well include them just in case.

`journalctl -p 3 -xb`
```text
Sep 02 19:14:50 user kernel: ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.PB2], AE_NOT_FOUND (20240322/dswload2-163)
Sep 02 19:14:50 user kernel: ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20240322/psobject-220)
Sep 02 19:14:50 user kernel: hub 6-0:1.0: config failed, hub doesn't have any ports! (err -19)
Sep 02 19:14:50 user kernel: 
Sep 02 19:14:51 user kernel: NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
Sep 02 19:14:52 user kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: Invalid data passed [NV_ERR_INVALID_DATA] (0x00000025) returned from PlatformRequestHandler failed to get target temp from SBIOS @ platform_request_handler_ctrl>
Sep 02 19:14:52 user kernel: NVRM: nvAssertOkFailedNoLog: Assertion failed: Invalid data passed [NV_ERR_INVALID_DATA] (0x00000025) returned from PlatformRequestHandler failed to get platform power mode from SBIOS @ platform_request_hand>
Sep 02 19:14:52 user kernel: NVRM: testIfDsmSubFunctionEnabled: GPS ACPI DSM called before _acpiDsmSupportedFuncCacheInit subfunction = 11.
Sep 02 19:14:53 user kernel: amdgpu 0000:06:00.0: [drm] *ERROR* dc_dmub_srv_log_diagnostic_data: DMCUB error - collecting diagnostic data
Sep 02 19:14:59 user kwin_wayland[1059]: kwin_scene_opengl: Invalid framebuffer status:  "GL_FRAMEBUFFER_INCOMPLETE_MISSING_ATTACHMENT"
Sep 02 19:14:59 user kwin_wayland[1059]: kwin_scene_opengl: Invalid framebuffer status:  "GL_FRAMEBUFFER_INCOMPLETE_ATTACHMENT"
Sep 02 19:16:56 user kwin_wayland[1059]: kwin_scene_opengl: Invalid framebuffer status:  "GL_FRAMEBUFFER_INCOMPLETE_ATTACHMENT"
```

If you want any other info just let me know.

artworkad, 2025-02-12:
Same issue for me on Lenovo Legion 7 Pro Ryzen 7945HX and RTX 4080. Installation works fine but fan curve shows just zeros. Fan speed is fixed at 2000rpm in power saver mode.

Gert-dev, 2025-06-11:
Just FYI and without wanting to cross-post my own comment, but some models such as the 16ARX8H currently appear to only show a _custom_ fancurve in the fancurve table and it only gets populated when activating the fourth special cooling mode. This may also be the case for other newer models.

There are some more details in https://github.com/johnfanv2/LenovoLegionLinux/issues/234#issuecomment-2963760187.

xdlolrgb, 2025-06-14:
honestly he should try to get this merged into the kernel. this would get better support if it gets there probably

closed 2026-09-12 by github-actions[bot]