johnfanv2/LenovoLegionLinux #137: Fedora 39 update breaks rpm package dependencies

class1c-j, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
With the update to Fedora 39, I now have python 3.12 installed by default, meaning that trying to install lll with the instructions from https://mrduartept.github.io/LLL-pkg-repo/ yields in 

```
Error:
 Problem 1: conflicting requests
  - nothing provides python(abi) = 3.11 needed by python3-darkdetect-0.8.0-1.noarch from LLL-pkg-repo
 Problem 2: conflicting requests
  - nothing provides python(abi) = 3.11 needed by python3-lenovolegionlinux-0.0.6-1.noarch from LLL-pkg-repo
  - nothing provides python(abi) = 3.11 needed by python3-lenovolegionlinux-0.0.8-1.noarch from LLL-pkg-repo
  - nothing provides python(abi) = 3.11 needed by python3-lenovolegionlinux-0.0.9-1.noarch from LLL-pkg-repo
``` 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Update the dependencies to include python3.12 or if needed update the codebase to the new version

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Installing python 3.11 or using the driver without the python package for the GUI

**Model and Debug Info**

Distribution: Fedora 39 (including version, e.g. Ubuntu 24.04)
Model name: Legion Slim 5
CPU model: Intel i7 13700h (e.g. AMD Ryzen 7 5800H)
GPU model: RTX 4070 (e.g NVIDIA RTX 3700)
Keyboard backlight: Single color with off/medium/bright (RGB, single color with off/on, single color with off/medium/bright)
Light in lid or logo: no (yes/no)
Light at IO-Ports at back: no (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82YA
	Version: Legion Slim 5 16IRH8
	Serial Number:
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82YA_BU_idea_FM_Legion Slim 5 16IRH8
	Family: Legion Slim 5 16IRH8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: M2CN25WW
	Release Date: 05/16/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.25
	Firmware Revision: 2.25

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4245487636
temperature access method: 5
CPU temperature error: 0
CPU temperature: 49
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 49
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 49
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 35
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 0
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 230
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 1
WMI light y logo/lid: 1
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

Manuka-Rathnayake, 2023-11-13:
same problem did you solve it?

arzkar, 2023-11-22:
Same problem. Cant upgrade to fedora 39 because of this. Any updates?

Fenimoure, 2023-11-29:
Same problem for me

MrDuartePT, 2023-12-04:
The update only was to make the rpm work on Fedora 39 but the devices support didn’t change.
In other work the problem is not the rpm package itself but the devices supported by LLL, basically for 2023 model (gen 8) we still not found out how to make the fancurve work.

closed 2023-12-04 by MrDuartePT

Fenimoure, 2023-12-04:
Hm, following [instructions for Fedora 39 on github](https://github.com/johnfanv2/LenovoLegionLinux/tree/main/package_repo) and then getting a message that python3-darkdetect and python3-lenovolegionlinux packages were not found.

Fenimoure, 2023-12-04:
I am not sure anymore. Are there going to be any other updates to instructions or is this error unintended:

  - Curl error (3): URL using bad/illegal format or missing URL for baseurl=https://johnfanv2.github.io/LenovoLegionLinux/package_repo/fedora/packages/repodata/repomd.xml [URL rejected: Port number was not a decimal number between 0 and 65535]

MrDuartePT, 2023-12-04:
well remove the repo and readded because on my system is working but let me check

MrDuartePT, 2023-12-04:
@Fenimoure for some reason is not working use this repo instead:
https://github.com/MrDuartePT/LLL-pkg-repo