johnfanv2/LenovoLegionLinux #138: Support for LOQ 16 (16IRH8) [2023]

Jalopy-Tech, 2023-11-09:
I can install using forcereloadmodule.

### Results:

- GUI works (legion_gui).
- Showing and changing power modes work (performance, balanced, quiet), but the GUI doesn't dynamically update the checkboxes if I use Fn Q.
- Enabling/disabling conservation mode works.
- Enabling/disabling touchpad works.
- Enabling/disabling Fn lock works.
- Enabling/disabling rapid charging works.
- Fan curve doesn't work, but in GUI, "Read from HW" provides Fan 1 Speed row values.
- Sensors don't work. They shows zero for fan speeds and temperatures.
- Note that the L5P-Keyboard-RGB application works for the 4-zone RGB lighting keyboard - the Lenovo LOQ is now in the list of supported models.

### Model and Debug Info:
Distribution: Kubuntu 23.10
Model name: LOQ 16 (16IRH8)
CPU model: Intel 13th Gen i7 (i7-13620H)
GPU model: NVIDIA GeForce RTX 4050 Laptop
Keyboard backlight: 4-zone RGB
Light in lid or logo: no
Light at IO-Ports at back: no

### Output of sudo dmidecode -t bios:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: LZCN30WW
        Release Date: 06/15/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.30
        Firmware Revision: 1.30

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 8
                en|US|iso8859-1,0
                fr|FR|iso8859-1,0
                zh|TW|unicode,0
                ja|JP|unicode,0
                it|IT|iso8859-1,0
                es|ES|iso8859-1,0
                de|DE|iso8859-1,0
                pt|PT|iso8859-1,0
        Currently Installed Language: en|US|iso8859-1,0
```
### Output of sudo cat /sys/kernel/debug/legion/fancurve:
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4245487636
temperature access method: 1
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 45
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 45
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 42
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 42
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1400
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1400
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1400
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
8000     0       0       0       0       0       0       0       0       255
9200     0       0       0       0       0       0       0       0       255
3500     0       0       0       0       0       0       0       0       255
8000     0       0       0       0       0       0       0       0       255
9200     0       0       0       0       0       0       0       0       255
3500     0       0       0       0       0       0       0       0       255
8200     0       0       0       0       0       0       0       0       255
9900     0       0       0       0       0       0       0       0       255
3500     0       0       0       0       0       0       0       0       255
9400     0       0       0       0       0       0       0       0       255
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```
### Inspecting WMI entries
[fwts_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/13311985/fwts_re.tar.gz)

### Disassembling ACPI tables
Running iasl -e SSDT* -d DSDT resulted in this error (created empty DSDT.dsl file):

Firmware Error (ACPI): Failure creating named object [\_SB.PC00.XHCI.RHUB.HS01._UPC], AE_ALREADY_EXISTS (20200925/dswload-387)
ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20200925/psobject-264)
Could not parse ACPI tables, AE_ALREADY_EXISTS

After removing SSDT17, no error occurred and the DSDT.dsl was created. This zip file contains all SSDT files, the DSDT file and the successful DSDT.dsl file:
[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/13312021/acpi_re.tar.gz)
### Gathering WMI info in Windows:
[windows_wmi.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/13312047/windows_wmi.txt)
### Thank you:
Please let me know if I can help in any way. Thank you for this excellent project.

Isurwars, 2023-11-30:
I have the exact same laptop and would like to help in adding support for it.

MrDuartePT, 2024-03-12:
It allready support since this PR: https://github.com/johnfanv2/LenovoLegionLinux/pull/151

And it seems everthing works since the EC is very identical to gen 6 and 7.