johnfanv2/LenovoLegionLinux #140: [BUG] Fancurve not showing and EC Chip ID not 8227

Bryntet, 2023-11-15:
**Problem Description**
EC Chip ID is 5507 (not 8227) and fancurve showing as 0 everywhere.

Has the bug occurred following steps described in the README?
Yes

What steps did you take?
Installed it using the AUR

What commands did you enter?
`sudo cat /sys/kernel/debug/legion/fancurve`

What was the (full) error you received and what command or action did lead to it?
```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 5
CPU temperature error: 0
CPU temperature: 75
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 75
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 75
GPU temperature error: 0
GPU temperature: 50
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 50
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 50
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 2500
1 fanspeed EC error: 0
1 fanspeed EC: 48
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2500
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 2500
2 fanspeed error: 0
2 fanspeed: 2700
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 2700
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 18
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
0	0	0	0	0	0	0	0	0	0
=====================```

**Model and Debug Info**

Distribution: Arch (including version, e.g. Ubuntu 24.04)
Model name: Lenovo Legion Pro 5 16ARX8
CPU model: AMD Ryzen 9 7945HX
GPU model: NVIDIA RTX 4070
Keyboard backlight: RGB
Light in lid or logo: No(yes/no)
Light at IO-Ports at back: Yes, at the power input (yes/no)

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WM
	Version: Legion Pro 5 16ARX8
	Serial Number: XXXXX
	UUID: XXXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WM_BU_idea_FM_Legion Pro 5 16ARX8
	Family: Legion Pro 5 16ARX8

Handle 0x0017, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0026, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: LPCN44WW
	Release Date: 06/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.44
	Firmware Revision: 1.44

Handle 0x0018, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
Already put this above :)

gunters63, 2023-11-17:
I have the same model, only with 4060 GPU and exactly the same debug output.
I am willing to help troubleshooting this :)

jgirardet, 2024-01-08:
Same model as OP, same thing

zedMar65, 2025-04-23:
Same problem as op

zedMar65, 2025-05-04:
@Bryntet any solutions/progress?
Also same problem on:
Arch, Legion 7 16IRX9

daniel-baf, 2026-09-11:
I have this exact machine (Legion Pro 5 16ARX8, 82WM; newer BIOS LPCN65WW, same EC 0x5507) and dug into what's happening — there are two separate things stacked in this report:

**1. The all-zeros table (partially fixed already):** your build's feature line (`fancurve powermode platformprofile platformprofilenotify minifancurve`) predates the WMI3 fan-table read/write support, so nothing was read at all. On current main the speed column reads correctly (fan curve points show real percentages) — worth retesting with a recent build.

**2. The remaining zeros are an artifact of the WMI method itself:** `wmi_read_fancurve_custom()` memset-zeros the curve and only fills per-point `speed1`; `wmi_write_fancurve_custom()` likewise only transmits per-point `speed1` (buffer 0x06-0x18). Temperature thresholds, fan2 speed and accel/decel have **no WMI backing** on this EC: writes are accepted but never persist, readback always shows 0. The EC's own thermal control is fine — it keeps using its firmware thresholds; only the curve *speeds* are actually adjustable.

Practical workaround today: write the speeds directly, e.g.

```bash
for i in 1 2 3 4 5 6 7 8 9 10; do echo <pwm 0-255> | sudo tee /sys/class/hwmon/hwmon*/pwm1_auto_point${i}_pwm; done
```

(the hwmon pwm value maps to a percentage of the EC max: stored% = pwm*100/255)

I've opened #563, which marks `model_lpcn` as `wmi_fancurve_speed_only` so the module stops advertising the attributes the firmware silently drops — after that, `legion_cli fancurve-write-file-to-hw` and the GUI complete without the `OSError: [Errno 95]` crash on accel/decel.

One more firmware quirk worth knowing on this BIOS: switching to the `max-power` platform profile **cuts power instantly** (hard power-off, no shutdown sequence) — stick to quiet/balanced/performance/custom.

st0nie, 2026-09-14:
LPCN is in the allowlist and its fan-curve config was corrected recently (fbca373, 1b9d275). Please retest with the current release; if the fancurve still doesn't show, reopen with `sudo cat /sys/kernel/debug/legion/fancurve` and `sudo dmesg | grep -i legion`.

closed 2026-09-14 by st0nie