johnfanv2/LenovoLegionLinux #145: [Question] Is `system76-power` conflict with LenovoLegion module ?

ikus060, 2023-12-03:
As I'm running PopOS, is it possible to run system76-power in addition to this LenovoLegion kernel module ? Quickly going trough the C code, I dont see any problem excepth the fact system76-power will probably not switch the power profile, but I could still toggle using Fn+Q.

Are are you tough.

MrDuartePT, 2023-12-13:
I don't think system76-power doesn't conflicts but yes the power profile change it will not work, Pls tested and report back

ikus060, 2023-12-18:
After testing. It seams `system76-power` does integrate with `/sys/firmware/acpi/platform_profile`. So when using the top-left menu in Gnome and updating the power profile, it also make call to acpi platform_profile.

closed 2023-12-18 by ikus060