johnfanv2/LenovoLegionLinux #146: Confused: Is there a COPR for LLL? (copr.fedorainfracloud.org)

msmafra, 2023-12-11:
Hello there

Is there a COPR available to install LLL?
I saw that is described as PPA instead of COPR, but did not find an URL nor found in copr.fedorainfracloud.org. 
Are those just to generate RPMS?

MrDuartePT, 2023-12-13:
Hello.

I dont use the Fedora COPR and I not planing on using since I don't use fedora as my main OS.
If you what to maintain the package on COPR pls tag me, also in the repo you have the workflow and the script on how to build the package.

Pls use the project fedora repo (`inside the package_repo folder`) to install the package!

MrDuartePT, 2023-12-13:
On a second thought I could try to use Copr since it only require to provide the spec file if work well I will just drop the repo here and use corp instead

msmafra, 2023-12-13:
I dabbled in COPR spec files once many years back for two packages I wanted to use on Fedora, but I gave up. I also read and article/post (don´t recall the url) about the quirks and documentation problems/inconsistency of RPMs sometime back. I don't know if that got improved.
Since I use CachyOS's kernel as my default on Fedora nowadays, LLL is already integrated by them. The COPR would be a huge help when using Fedora's  kernel.

MrDuartePT, 2023-12-15:
Done now the Fedora Copr is the provider of LenovoLegionLinux
Pls follow this instruction to migrate:https://github.com/johnfanv2/LenovoLegionLinux/blob/main/package_repo/README.md

closed 2023-12-15 by MrDuartePT