johnfanv2/LenovoLegionLinux #158: [BUG] dmesg is spammed with "ACPI result for" messages

awsms, 2024-01-30:
**Problem Description**
<Please describe your problem here>
Has the bug occurred following steps described in the README?
yes
What steps did you take?
The package was installed from the AUR (lenovolegionlinux-dkms-git), I only changed the fancurve preset to "quiet-battery" in legion_gui
What commands did you enter?
\
What was the (full) error you received and what command or action did lead to it?
I'm getting spammed in dmesg with "ACPI results" messages. Just as an example:
```
[  594.049371] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.049475] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.049573] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.049644] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.160328] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.160431] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.160542] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.160610] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.272185] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.272292] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.272392] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.272466] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.384224] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.384337] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.384446] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.384515] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.495212] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.495308] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.495412] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.495480] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.606404] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
[  594.606504] legion_laptop: ACPI result for aaaaaaa-aaaaa-aaaa-aaaa-aaaaaaaaaa:1: ACPI buffer length: 2
```

**Model and Debug Info**

Distribution: EndeavourOS, 6.7.2-zen
Model name: 15ACH6H
CPU model: AMD Ryzen 5 5600H
GPU model: NVIDIA RTX 3600
Keyboard backlight: single color with off/on
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82JU
	Version: Legion 5 15ACH6H
	Serial Number: XXXXX
	UUID: xxxxxxxxxx-xxx-xx-xxxx-xxxxxxxxxx
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82JU_BU_idea_FM_Legion 5 15ACH6H
	Family: Legion 5 15ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: GKCN54WW
	Release Date: 05/05/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.54
	Firmware Revision: 1.54

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 62
CPU temperature EC error: 0
CPU temperature EC: 62
CPU temperature ACPI error: 0
CPU temperature ACPI: 62
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 62
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 41
GPU temperature EC error: 0
GPU temperature EC: 41
GPU temperature ACPI error: 0
GPU temperature ACPI: 41
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 41
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2366
1 fanspeed EC error: 0
1 fanspeed EC: 2366
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2300
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 2300
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 2464
2 fanspeed EC error: 0
2 fanspeed EC: 2464
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2400
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 2400
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 4
EC fan curve points size: 9
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 48	 0	 60	 0	 41
1700	 1900	 2	 2	 45	 54	 56	 60	 39	 44
1900	 2000	 2	 2	 51	 58	 56	 60	 42	 50
2200	 2100	 2	 2	 55	 62	 56	 60	 46	 127
2300	 2400	 2	 2	 59	 71	 56	 60	 127	 127
2600	 2700	 2	 2	 68	 76	 56	 65	 127	 127
2900	 3000	 2	 2	 72	 81	 61	 69	 127	 127
3500	 3500	 2	 2	 78	 90	 66	 77	 127	 127
3800	 3800	 2	 2	 87	 127	 72	 127	 127	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

MrDuartePT, 2024-03-28:
It completely normal is part of the driver initialization not bother with it.

awsms, 2024-03-28:
> It completely normal is part of the driver initialization not bother with it.

I'm getting spammed in dmesg forever, this isn't about initialization here

MrDuartePT, 2024-03-28:
Try updating your bios, now that I notice, it very old at this point the latest one is `GKCN60WW` I will not reopen this issue since I have the same model and on the latest bios that not happen

awsms, 2024-03-29:
I've just spent hours installing Windows on another partition to upgrade the BIOS to GKCN64WW

And now that's dmesg:
```
[   12.515453] Bluetooth: RFCOMM TTY layer initialized
[   12.515464] Bluetooth: RFCOMM socket layer initialized
[   12.515469] Bluetooth: RFCOMM ver 1.11
[   12.809233] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   12.809563] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   12.922900] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   12.923040] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.034407] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.034552] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.145493] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.145644] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.256609] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.256745] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.367973] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.368110] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.478755] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.478891] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.589562] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.589698] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.700478] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.700618] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.811649] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.811787] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.922393] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   13.922533] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.033221] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.033358] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.143824] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.143974] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.254226] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.254360] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.364927] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.365057] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.475196] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.475326] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.585849] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.585982] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.696065] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.696195] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.806475] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.806610] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.916887] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   14.917019] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   15.026911] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   15.027044] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   15.137365] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
[   15.137510] legion_laptop: ACPI result for xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx:1: ACPI buffer length: 2
```

Nothing has changed.

MrDuartePT, 2024-03-29:
> I've just spent hours installing Windows on another partition to upgrade the BIOS to GKCN64WW
>
> And now that's dmesg:
>
> ```
> [   12.515453] Bluetooth: RFCOMM TTY layer initialized
> [   12.515464] [...]

You can have just use Windows PE ISO to update the bios you don't need to install.

I was a bit wrong about how that logs appear (since you basically delete the GUID without any reason, that is not sensible information is just say what WMI function is being used). 

But basically that logs appear ever time the driver is call to check the capabilities of the device:
1 - When using the gui/cli, since interact with the driver.
2 - When initialization the driver.
If you know C you can understand how that logs are created: https://github.com/johnfanv2/LenovoLegionLinux/blob/c460f141834c4166e6e3ed0b41ecabb3cfb61396/kernel_module/legion-laptop.c#L1311

I don't fell like reducing the verbosity since that is not really a BUG with the driver, it how it was design to work.
Also the ACPI Result can sometime (at least for older model) help debugging.

MrDuartePT, 2024-03-29:
Well I gonna reduce verbosity for now, but In the future I can revert this commit: 4c2f031

closed 2024-03-29 by MrDuartePT