johnfanv2/LenovoLegionLinux #159: support for Legion Pro 7 16IRX8H - 2023 model

realsnick, 2024-01-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Model and Debug Info**

Distribution: ??? (including version, e.g. Ubuntu 24.04)
Model name: ???
CPU model: Intel 13th Gen Intel i9-13900HX
GPU model: NVIDIA GeForce RTX 4080 Max-Q
Keyboard backlight: RGB 
Light in lid or logo: no
Light at IO-Ports at back: no 

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
	Serial Number: PF4FYD0E
	UUID: 0f10e908-6e6d-41e8-abff-745d22037539
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
	Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN36WW
	Release Date: 04/28/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.36
	Firmware Revision: 1.36

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 53
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 53
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 53
GPU temperature error: 0
GPU temperature: 43
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 43
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 43
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1900
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1900
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1900
2 fanspeed error: 0
2 fanspeed: 1800
2 fanspeed EC error: 0
2 fanspeed EC: 14083
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1800
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1800
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 0
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
```

arnfaldur, 2024-11-18:
The audio seems to be fixed for me, as well as disabling the touchpad. Running on Arch Linux, 6.11.8

Getting the fancurves to work would be amazing though.

i-am-logger, 2024-11-18:
can you share your config to disable the touchpad? i'm on nixos, might have not done it right. and yeah, audio seems to be working last few weeks for me.. but still have issues at times...seems something with pipewire

st0nie, 2026-09-14:
The Legion Pro 7 16IRX8H (BIOS KWCN) is in the allowlist and listed as supported in the README (sensors, fan curve, power profile, fan unlock). KWCN fan-curve handling was reworked in #567 (level indices). Please retest with the current release and reopen with dmesg output if something still fails.

closed 2026-09-14 by st0nie