johnfanv2/LenovoLegionLinux #161: Lenovo Ideapad Gaming 3 15IHU6 (8K21) support

ardashir77, 2024-02-01:
Dunno if possible all things considered, but I figured out I'd ask here for support for my laptop as I've been maining Linux for almost 8 years and having to be in Windows has broken my workflow. Figured out I should've bought another laptop but oh well :) 

I tried to use LLL on Ubuntu just as a test using the official PPAs but all features were greyed out, my laptop is considerably slower in Linux than on Windows (at the point where I'd get around 120-ish fps in games that would easily been a stable 360 fps for me on Windows) and I'd like to get back to Linux if possible. Main distro is Arch though NixOS is also pretty cool :D 

Model and Debug Info:

Distribution: Arch Linux
Model name: Lenovo Ideapad Gaming 3 15IHU6 (82K1)
CPU model: Intel Core i5-11320H
GPU model: Nvidia Geforce GTX 1650
Keyboard backlight: RGB with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```# dmidecode 3.5 
Getting SMBIOS data from sysfs. 
SMBIOS 3.3.0 present. 

Handle 0x0001, DMI type 1, 27 bytes 
System Information
 	Manufacturer: LENOVO 
	Product Name: 82K1
 	Version: IdeaPad Gaming 3 15IHU6
 	# removed Serial
        # removed UUID
 	Wake-up Type: Power Switch
 	SKU Number: LENOVO_MT_82K1_BU_idea_FM_IdeaPad Gaming 3 15IHU6
 	Family: IdeaPad Gaming 3 15IHU6

Handle 0x0020, DMI type 12, 5 bytes 
System Configuration Options
 	Option 1: ConfigOptions1
 	Option 2: ConfigOptions2 
	Option 3: ConfigOptions3
 Handle 0x0023, DMI type 15, 29 bytes
 System Event Log
 	Area Length: 0 bytes
 	Header Start Offset: 0x0000 
	Header Length: 8192 bytes
 	Data Start Offset: 0x2000
 	Access Method: General-purpose non-volatile data functions
 	Access Address: 0x0000
 	Status: Valid, Not Full
 	Change Token: 0x12345678
 	Header Format: OEM-specific
 	Supported Log Type Descriptors: 3
 	Descriptor 1: POST memory resize
 	Data Format 1: None
 	Descriptor 2: POST error
 	Data Format 2: POST results bitmap
 	Descriptor 3: Log area reset/cleared
 	Data Format 3: None
 Handle 0x002E, DMI type 32, 11 bytes
 System Boot Information
 	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
       Vendor: LENOVO
       Version: H4CN33WW(V2.02)
       Release Date: 09/07/2023
       Address: 0xE0000
       Runtime Size: 128 kB
[... 606 lines cut ...]
Name: Reserved2
WmiMethodId: 6
Class Name: LENOVO_SR_DATA
Class GUID: {77E614ED-F19E-46D6-A613-A8669FEE1FF0}
Description: Super Resolution Reserved function
Implemented: True

########################################
########################################
########################################
Name: LENOVO_GAMEZONE_CPU_OC_DATA
Class Name: LENOVO_GAMEZONE_CPU_OC_DATA
Class GUID: {B7F3CA0A-ACDC-42D2-9217-77C6C628FBD2}
Description: 0
Methods:

########################################
########################################
########################################
Name: LENOVO_GAMEZONE_GPU_OC_DATA
Class Name: LENOVO_GAMEZONE_GPU_OC_DATA
Class GUID: {887B54E2-DDDC-4B2C-8B88-68A26A8835D0}
Description: 0
Methods:
```

ardashir77, 2024-02-01:
About the missing two commands (fwts and acpica-tools), I could not make them run on my archiso live session, God knows why. I was wondering if running them on Ubuntu would be the same, if so I will provide the outputs as requested.

MrDuartePT, 2024-03-17:
What you post is good enough to test the supports also try `make forcereloadmodule` and try get the output: `sudo cat /sys/kernel/debug/legion/fancurve`

MrDuartePT, 2024-03-17:
@ardashir77 try it and report back

ardashir77, 2024-03-17:
This is the output of `sudo cat /sys/kernel/debug/legion/fancurve`: 

vscythe@ubuntu-vscythe:~/Downloads/LenovoLegionLinux/kernel_module$ sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4244963348
temperature access method: 1
CPU temperature error: 0
CPU temperature: 255
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: 0
CPU temperature WMI: 40
CPU temperature WMI2 error: -14
CPU temperature WMI2: 40
CPU temperature WMI3 error: -14
CPU temperature WMI3: 40
GPU temperature error: 0
GPU temperature: 255
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: -14
GPU temperature WMI2: 35
GPU temperature WMI3 error: -14
GPU temperature WMI3: 35
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 65535
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 65535
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light Y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

ardashir77, 2024-03-17:
This is the output of `cat /sys/kernel/debug/legion/fancurve` : 

```
vscythe@ubuntu-vscythe:~/Downloads/LenovoLegionLinux/kernel_module$ sudo cat /sys/kernel/debug/legion/fancurve
EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4244963348
temperature access method: 1
CPU temperature error: 0
CPU temperature: 255
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: 0
CPU temperature WMI: 40
CPU temperature WMI2 error: -14
CPU temperature WMI2: 40
CPU temperature WMI3 error: -14
CPU temperature WMI3: 40
GPU temperature error: 0
GPU temperature: 255
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: 0
GPU temperature WMI: 35
GPU temperature WMI2 error: -14
GPU temperature WMI2: 35
GPU temperature WMI3 error: -14
GPU temperature WMI3: 35
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 65535
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 65535
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light Y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
25500	 25500	 255	 255	 255	 255	 255	 255	 255	 255
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
``

antheas, 2024-03-17:
acpica is an arch package
https://archlinux.org/packages/extra/x86_64/acpica/

and it is also on ubuntu
https://packages.debian.org/sid/acpica-tools

MrDuartePT, 2024-03-17:
well I find a issue with ideapad 3 liking to rewrite the fancurve even after saving: https://github.com/johnfanv2/LenovoLegionLinux/issues/44

ardashir77, 2024-03-17:
tried, to no avail, this is how it looks like though: 

[image]

MrDuartePT, 2024-03-17:
> tried, to no avail, this is how it looks like though:
> 
> ![Screenshot from 2024-03-17 10-48-49](https://private-user-images.githubusercontent.com/109394750/313486314-8bde4b9c-f585-4bca-93d1-87791a315041.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA2OTA5NzcsIm5iZiI6MTcxMDY5MDY3NywicGF0aCI6Ii8xMDkzOTQ3NTAvMzEzNDg2MzE0LThiZGU0YjljLWY1ODUtNGJjYS05M2QxLTg3NzkxYTMxNTA0MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxN1QxNTUxMTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mZjUyZTI5Zjk0NjQ0ZDBmOTNlNmZkMjRiOTQ5MDlhYTk0ZGE0YTQwZTY5ZGI4ZmIwMTI1YjUyMDgwNjg5YWU5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.OvI4BbLcpbcaEs0gJ5RXoJPgTGc4T3NYNF3lFvMff60)

Explain better `Read from HW` work or not, changing values and `Apply to HW` works?????

ardashir77, 2024-03-17:
If I press `Read from HW` every value appears to be marked as 255, `Apply to HW` doesn't work at all; changing battery mode doesn't work either, every option looks like it gets unchecked/rechecked instantly

ardashir77, 2024-03-17:
> well I find a issue with ideapad 3 liking to rewrite the fancurve even after saving: #44 @ardashir77 try change fancurve values in the gui
> 
> To open the gui:`./python/legion_linux/legion_linux/legion_gui.py --use_legion_cli_to_write`

Speaking of that issue, I try to change the performance mode as per the GUI/GNOME options and `cat /sys/firmware/acpi/platform_profile` still displays as `quiet`, maybe that's why my performance is so dramatically low compared to Windows?

MrDuartePT, 2024-03-17:
> > well I find a issue with ideapad 3 liking to rewrite the fancurve even after saving: #44 @ardashir77 try change fancurve values in the gui
> > To open the gui:`./python/legion_linux/legion_linux/legion_gui.py --use_legion_cli_to_write`
> 
> Speaking of that issue, I try to change the performance mode as per the LLL GUI and GNOME options and `cat /sys/firmware/acpi/platform_profile` still displays as `quiet`, maybe that's why my performance is so dramatically low compared to Windows?

The power profile can be change using `fn+q`

ardashir77, 2024-03-17:
> > > well I find a issue with ideapad 3 liking to rewrite the fancurve even after saving: #44 @ardashir77 try change fancurve values in the gui
> > > To open the gui:`./python/legion_linux/legion_linux/legion_gui.py --use_legion_cli_to_write`
> > 
> > 
> > Speaking of that issue, I try to change the performance mode as per the LLL GUI and GNOME options and `cat /sys/firmware/acpi/platform_profile` still displays as `quiet`, maybe that's why my performance is so dramatically low compared to Windows?
> 
> The power profile can be change using `fn+q`. The powerprofile should work fine without this driver

I changed it to performance and so it appears as such, though performance doesn't seem to increase

closed 2026-09-12 by github-actions[bot]