johnfanv2/LenovoLegionLinux #162: Support for 82K2 IdeaPad Gaming 3 15ACH6

OverStyleFR, 2024-02-09:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like to use this software for my Lenovo laptop

**Describe alternatives you've considered**
I search for another tools but i don't find a good one.

**Model and Debug Info**

Distribution: Fedora Linux 39 (KDE Plasma) x86_64
Model name: 82K2 IdeaPad Gaming 3 15ACH6 
CPU model: AMD Ryzen 7 5800H with Radeon Graphics (16) @ 4.463GHz
GPU model:  NVIDIA GeForce RTX 3060 Mobile / Max-Q
Keyboard backlight: I don't understand what you meen
Light in lid or logo: I don't understand what you meen
Light at IO-Ports at back: I don't understand what you meen

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82K2
        Version: IdeaPad Gaming 3 15ACH6
        Serial Number: ???
        UUID: ???
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82K2_BU_idea_FM_IdeaPad Gaming 3 15ACH6
        Family: IdeaPad Gaming 3 15ACH6

Handle 0x0020, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0030, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: H3CN39WW(V2.09)
        Release Date: 05/31/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.39
        Firmware Revision: 1.32

Handle 0x0021, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: ffff
EC Chip Version: ffff
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228186128
temperature access method: 3
CPU temperature error: 0
CPU temperature: 55
CPU temperature EC error: 0
CPU temperature EC: 255
CPU temperature ACPI error: -5
CPU temperature ACPI: 255
CPU temperature WMI error: 0
CPU temperature WMI: 55
CPU temperature WMI2 error: -14
CPU temperature WMI2: 55
CPU temperature WMI3 error: -14
CPU temperature WMI3: 55
GPU temperature error: 0
GPU temperature: 44
GPU temperature EC error: 0
GPU temperature EC: 255
GPU temperature ACPI error: -5
GPU temperature ACPI: 255
GPU temperature WMI error: 0
GPU temperature WMI: 44
GPU temperature WMI2 error: -14
GPU temperature WMI2: 44
GPU temperature WMI3 error: -14
GPU temperature WMI3: 44
fan speed access method: 0
1 fanspeed error: -22
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 65535
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 65535
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 0
2 fanspeed error: -22
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 65535
2 fanspeed ACPI error: -5
2 fanspeed ACPI: 65535
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 255
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 0
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light Y logo/lid: -5
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: -1
EC lockfancontroller: error
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

ManuaL46, 2024-08-09:
I have the same model and would also love to see support for this. All things work as expected, but I was just wishing for something to be able to read and modify the performance modes built into the laptop like the vantage utility did on windows. 

Currently I have to resort pressing Fn + q and hearing the fan noise to guess the performance mode.

TTF-fog, 2025-01-04:
@ManuaL46 
the arch wiki details acpi calls for power modes etc.. (we love the arch wiki)
https://wiki.archlinux.org/title/Lenovo_IdeaPad_Gaming_3 
i also made a [tray widget](https://github.com/TTF-fog/ideapad-control) for controlling it using a GUI

st0nie, 2026-09-14:
The IdeaPad Gaming 3 15ACH6 (82K2, BIOS H3CN) is covered by the H3CN allowlist entry. Please retest with the current release and reopen with dmesg output if it still fails.

closed 2026-09-14 by st0nie