johnfanv2/LenovoLegionLinux #163: Support Legion Pro 5 16IRX9 (2024)

Drakanio, 2024-02-10:
Distribution: Arch
Model name: Legion Pro 5 16IRX9
CPU model: i9-14900HX
GPU model: RTX 4070
Keyboard backlight: 4-Zone RGB
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83DF
	Version: Legion Pro 5 16IRX9
	Serial Number: 
	UUID: 
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83DF_BU_idea_FM_Legion Pro 5 16IRX9
	Family: Legion Pro 5 16IRX9

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: N0CN15WW
	Release Date: 12/29/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.15
	Firmware Revision: 1.15

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: Datei oder Verzeichnis nicht gefunden (File / Folder not found)

```

[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/14228873/fwts_wmi.log)

[acpi_re.tar.gz](https://github.com/johnfanv2/LenovoLegionLinux/files/14228875/acpi_re.tar.gz)

Drakanio, 2024-02-16:
[wmi_windows.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/14314384/wmi_windows.txt)

Drakanio, 2024-03-11:
Added to legion-laptop.c (try and error :D)

```
// test Legion Pro 5 16IRX9

static const struct model_config model_n0cn = {
	.registers = &ec_register_offsets_v0, //v0 ok?
	.check_embedded_controller_id = false, //???
	.embedded_controller_id = 0x5507,
	.memoryio_physical_ec_start = 0xC400,
	.memoryio_size = 0x300,
	.has_minifancurve = false, //???
	.has_custom_powermode = true, //OK
	.access_method_powermode = ACCESS_METHOD_WMI, //OK 
	.access_method_keyboard = ACCESS_METHOD_WMI, // ???
	.access_method_fanspeed = ACCESS_METHOD_WMI3, //OK
	.access_method_temperature = ACCESS_METHOD_WMI3, //OK
	.access_method_fancurve = ACCESS_METHOD_WMI3, //???
	.access_method_fanfullspeed = ACCESS_METHOD_WMI, //???
	.acpi_check_dev = true, //???
	.ramio_physical_start = 0xFE0B0400,
	.ramio_size = 0x600
};
```
and

```
	{

		// e.g. Legion Pro 5 16IRX9
		.ident = "N0CN",
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "LENOVO"),
			DMI_MATCH(DMI_BIOS_VERSION, "N0CN"),
		},
		.driver_data = (void *)&model_n0cn
	},
	{}

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:

```EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 5
CPU temperature error: 0
CPU temperature: 52
CPU temperature EC error: 0
CPU temperature EC: 55
CPU temperature ACPI error: 0
CPU temperature ACPI: 52
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 52
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 87
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 5
1 fanspeed error: 0
1 fanspeed: 1800
1 fanspeed EC error: 0
1 fanspeed EC: 18039
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1800
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1800
2 fanspeed error: 0
2 fanspeed: 1700
2 fanspeed EC error: 0
2 fanspeed EC: 14177
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1700
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1700
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 2
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: false
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100      0       0       0       0       0       0       0       0       0
200      0       0       0       0       0       0       0       0       0
300      0       0       0       0       0       0       0       0       0
400      0       0       0       0       0       0       0       0       0
500      0       0       0       0       0       0       0       0       0
600      0       0       0       0       0       0       0       0       0
700      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
100      0       0       0       0       0       0       0       0       0
200      0       0       0       0       0       0       0       0       0
300      0       0       0       0       0       0       0       0       0
400      0       0       0       0       0       0       0       0       0
500      0       0       0       0       0       0       0       0       0
600      0       0       0       0       0       0       0       0       0
700      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
800      0       0       0       0       0       0       0       0       0
=====================
```

gou4shi1, 2024-08-18:
> [legion_gui_py_Log.log](https://github.com/johnfanv2/LenovoLegionLinux/files/14553304/legion_gui_py_Log.log)
> 
> Change Power Options work, Read from HW / Apply to HW doesn´t:
> 
> Read from HW LOG
> ```
> 2024-03-11 04:38:56,736 - INFO: update_view_from_feature: 0
> 2024-03-11 04:38:56,736 - INFO: Feature StrFileFeature reading
> 2024-03-11 04:38:56,737 - INFO: Feature StrFileFeature reading: quiet balanced balanced-performance performance
> 2024-03-11 04:38:56,737 - INFO: Feature PlatformProfileFeature reading
> 2024-03-11 04:38:56,737 - INFO: Feature PlatformProfileFeature reading: quiet
> 2024-03-11 04:38:56,738 - INFO: Feature CPULongtermPowerLimit reading
> 2024-03-11 04:38:56,738 - ERROR: Feature CPULongtermPowerLimit reading error [Errno 22] Das Argument ist ungültig
> 2024-03-11 04:38:56,761 - ERROR: [17960.177171] legion_laptop: get_simple_wmi_attributewith raw value: 1
> [17960.177470] legion_laptop: get_simple_wmi_attributewith raw value: 0
> [17960.178571] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: expected type 3 but got 1; expected length 16 but got 0;
> [17960.197657] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: expected type 3 but got 1; expected length 16 but got 0;
> [17960.216922] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [17960.236724] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:16
> [17960.255659] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:12
> [17960.275845] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:5
> [17960.292966] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:3
> [17960.313537] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [17960.346293] legion_laptop: get_simple_wmi_attributewith raw value: 1
> [18423.758131] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: expected type 3 but got 1; expected length 16 but got 0;
> [18423.783887] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: expected type 3 but got 1; expected length 16 but got 0;
> [18423.818187] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [18423.840960] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:16
> [18423.864316] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:12
> [18423.887329] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:5
> [18423.907871] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:3
> [18423.929693] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [18610.529454] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: expected type 3 but got 1; expected length 16 but got 0;
> 
> 2024-03-11 04:38:56,764 - INFO: Feature CPUShorttermPowerLimit reading
> 2024-03-11 04:38:56,764 - ERROR: Feature CPUShorttermPowerLimit reading error [Errno 22] Das Argument ist ungültig
> 2024-03-11 04:38:56,786 - ERROR: [17960.177470] legion_laptop: get_simple_wmi_attributewith raw value: 0
> [17960.178571] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: expected type 3 but got 1; expected length 16 but got 0;
> [17960.197657] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: expected type 3 but got 1; expected length 16 but got 0;
> [17960.216922] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [17960.236724] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:16
> [17960.255659] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:12
> [17960.275845] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:5
> [17960.292966] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:3
> [17960.313537] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [17960.346293] legion_laptop: get_simple_wmi_attributewith raw value: 1
> [18423.758131] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: expected type 3 but got 1; expected length 16 but got 0;
> [18423.783887] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: expected type 3 but got 1; expected length 16 but got 0;
> [18423.818187] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [18423.840960] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:16
> [18423.864316] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:12
> [18423.887329] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:5
> [18423.907871] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:3
> [18423.929693] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> [18610.529454] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:5: expected type 3 but got 1; expected length 16 but got 0;
> [18610.555520] legion_laptop: Unexpected ACPI result for 14afd777-106f-4c9b-b334-d388dc7809be:3: expected type 3 but got 1; expected length 16 but got 0;
> 
[... cut ...]
> [18610.624387] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:12
> [18610.643197] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:5
> [18610.663059] legion_laptop: ACPI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:3
> [18610.681478] legion_laptop: WMI evaluation error for: da7547f1-824d-405f-be79-d9903e29ced7:8
> ```
> 
> Options: Fn Lock: work Win Key Enable: work Touchpad Enable: work Camera Power: ??? Battery Conversation: work

same for me

st0nie, 2026-09-14:
The Pro 5 16IRX9 (BIOS NMCN) is covered by the NMCN allowlist entry. Please retest with the current release and reopen with `sudo dmesg | grep -i legion` if it still doesn't load.

closed 2026-09-14 by st0nie