johnfanv2/LenovoLegionLinux #169: Ability to select between hybrid and dGPU from a kernel parameter

MathisP75, 2024-02-28:
**Is your feature request related to a problem? Please describe.**
Hybrid GPUs on Wayland is problematic and I would like Linux to be able to choose from hybrid or dGPU only modes without changing the BIOS parameter. This way, I do not have to make compromises on my Windows installation concerning battery life. The issue I have with hybrid graphics is that my external monitor is laggy when both GPUs are in use.

**Describe the solution you'd like**
If it is not already possible, I would like to have a kernel parameter that allows switching between hybrid mode and dGPU only mode. This utility (https://github.com/null-dev/lenovo-legion-wmi-dkms) managed to achieve it, however it is outdated and does not compile currently.

**Describe alternatives you've considered**
I have tried using the utility mentioned above, but it fails to compile. I have also tried fixing the external display issue, but without success for now. It is apparently a widespread issue, but I have talked to someone who seemingly does not experience it.

**Model and Debug Info**

Distribution: Arch Linux
Model name: Legion Pro 7 16IRX8H
CPU model: Intel i9 13900HX
GPU model: Nvidia RTX 4090
Keyboard backlight: RGB
Light in lid or logo: No

johnfanv2, 2024-03-03:
Try using the gsync feature Hybrid Mode (sometimes also called GSync) feature. You need a reboot so that the changes apply.

MrDuartePT, 2024-03-29:
@MathisP75 the project you send does the same thing and the Hybrid/Gsync feature does on your cli/gui. Instead of manually changing the option on the bios. It change though wmi and apply after a reboot.

Is the same thing as select dGPU only on Vantage/LLT on Windows

closed 2024-03-29 by MrDuartePT