johnfanv2/LenovoLegionLinux #170: Touchpad not working for Legion 9 16IRX8

mashu, 2024-02-28:
**Is your feature request related to a problem? Please describe.**
- Touchpad not working (should be by intel-lpss-pci driver on Windows it is Serial IO I2C)
- Lots of ACPI errors when booting
- Battery status not working
- Builtin speaker not working

**Describe the solution you'd like**
Would like to primarily have **Touchpad issue**  resolved as first step to get this laptop functional under Linux.

**Describe alternatives you've considered**
I tried Arch, Debian, Ubuntu and different kernel version from 6.5 to 6.8rc none of which fixed the problem with touchpad not being responsive at all. I tried lock-unlock touchpad button it didnt solve the problem either.

**Model and Debug Info**

Distribution: GNU/Linux Debian (sid at the moment)
Model name: Legion 9 16IRX8 83AG
CPU model: 13th Gen Intel(R) Core(TM) i9-13980HX
GPU model: NVIDIA GeForce RTX 4090 Laptop GPU
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: yes

**Additional information potentially useful to resolve the problem also submitted by several people having this very laptop at https://bugzilla.kernel.org/show_bug.cgi?id=218444**

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83AG
	Version: Legion 9 16IRX8
	Serial Number: XXXXXXXX
	UUID: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83AG_BU_idea_FM_Legion 9 16IRX8
	Family: Legion 9 16IRX8

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: MHCN40WW
	Release Date: 12/15/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.40
	Firmware Revision: 1.33

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Any help on this appreciated.

Kind regards,

MrDuartePT, 2024-02-28:
Unfortunately that not the right place to report touchpad and battery status not working. 
That is handle by the kernel ideapad_driver.
Is better to report that to the kernel bugzilla.

MrDuartePT, 2024-02-28:
I also notice you are using Debian Sid try to see if with latest mainline and Linux-firmware the problem remains

closed 2024-02-28 by MrDuartePT