johnfanv2/LenovoLegionLinux #171: [BUG] Error: Missing .env and other files when toggle Fn+Q

Flashwalker, 2024-03-01:
**Problem Description**
I get error about the script can't find .env and other files when i toggle the performance profiles via **Fn+Q**.

```
...
/usr/bin/fancurve-set: line 21: /etc/legion_linux//.env: No such file or directory
...
FileNotFoundError: [Errno 2] No such file or directory: '/etc/legion_linux//balanced-ac.yaml'
...
FileNotFoundError: [Errno 2] No such file or directory: '/etc/legion_linux//performance-ac.yaml'
...
FileNotFoundError: [Errno 2] No such file or directory: '/etc/legion_linux//quiet-ac.yaml'
...
```

Here is the syslog traceback:
<details>

```
# toggle to ballanced

Mar  2 00:28:56 smarttop systemd[1]: Starting Apply Fan Curve to Lenovo Legion Laptop depending on Power State and Charger Mode (LenovoLegionLinux kernel Module)...
Mar  2 00:28:56 smarttop systemd[1]: Started Apply Fan Curve to Lenovo Legion Laptop depending on Power State and Charger Mode (LenovoLegionLinux kernel Module).
Mar  2 00:28:56 smarttop fancurve-set[280836]: /usr/bin/fancurve-set: line 21: /etc/legion_linux//.env: No such file or directory
Mar  2 00:28:56 smarttop fancurve-set[280836]: Applying Balance Mode Profile   -> charger...
Mar  2 00:28:56 smarttop fancurve-set[280845]: Traceback (most recent call last):
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/bin/legion_cli", line 33, in <module>
Mar  2 00:28:56 smarttop fancurve-set[280845]:     sys.exit(load_entry_point('legion-linux==0.0.9', 'console_scripts', 'legion_cli')())
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/lib/python3/dist-packages/legion_linux/legion_cli.py", line 471, in main
Mar  2 00:28:56 smarttop fancurve-set[280845]:     args.func(legion, **vars(args))
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/lib/python3/dist-packages/legion_linux/legion_cli.py", line 312, in fancurve_write_preset_to_hw
Mar  2 00:28:56 smarttop fancurve-set[280845]:     legion.fancurve_write_preset_to_hw(presetname, write_minifancurve=True)
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 1481, in fancurve_write_preset_to_hw
Mar  2 00:28:56 smarttop fancurve-set[280845]:     self.load_fancurve_from_preset(name)
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 1475, in load_fancurve_from_preset
Mar  2 00:28:56 smarttop fancurve-set[280845]:     self.fan_curve = self.fancurve_repo.load_by_name(name)
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 1093, in load_by_name
Mar  2 00:28:56 smarttop fancurve-set[280845]:     return FanCurve.load_from_file(self._name_to_filename(name))
Mar  2 00:28:56 smarttop fancurve-set[280845]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 81, in load_from_file
Mar  2 00:28:56 smarttop fancurve-set[280845]:     with open(filename, 'r', encoding=DEFAULT_ENCODING) as filepointer:
Mar  2 00:28:56 smarttop fancurve-set[280845]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/legion_linux//balanced-ac.yaml'
Mar  2 00:28:56 smarttop systemd[1]: legion-linux.service: Main process exited, code=exited, status=1/FAILURE
Mar  2 00:28:56 smarttop systemd[1]: legion-linux.service: Failed with result 'exit-code'.

# toggle to performance

Mar  2 00:29:36 smarttop kernel: [152546.665351] legion_laptop: Fan event: legion type: 2;  acpi type: 3 (1=integer)
Mar  2 00:29:37 smarttop systemd[1]: Starting Apply Fan Curve to Lenovo Legion Laptop depending on Power State and Charger Mode (LenovoLegionLinux kernel Module)...
Mar  2 00:29:37 smarttop systemd[1]: Started Apply Fan Curve to Lenovo Legion Laptop depending on Power State and Charger Mode (LenovoLegionLinux kernel Module).
Mar  2 00:29:37 smarttop fancurve-set[280959]: /usr/bin/fancurve-set: line 21: /etc/legion_linux//.env: No such file or directory
Mar  2 00:29:37 smarttop fancurve-set[280959]: Applying Performance Mode Profile 龍  -> charger...
Mar  2 00:29:37 smarttop fancurve-set[280968]: Traceback (most recent call last):
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/bin/legion_cli", line 33, in <module>
Mar  2 00:29:37 smarttop fancurve-set[280968]:     sys.exit(load_entry_point('legion-linux==0.0.9', 'console_scripts', 'legion_cli')())
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/lib/python3/dist-packages/legion_linux/legion_cli.py", line 471, in main
Mar  2 00:29:37 smarttop fancurve-set[280968]:     args.func(legion, **vars(args))
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/lib/python3/dist-packages/legion_linux/legion_cli.py", line 312, in fancurve_write_preset_to_hw
Mar  2 00:29:37 smarttop fancurve-set[280968]:     legion.fancurve_write_preset_to_hw(presetname, write_minifancurve=True)
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 1481, in fancurve_write_preset_to_hw
Mar  2 00:29:37 smarttop fancurve-set[280968]:     self.load_fancurve_from_preset(name)
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 1475, in load_fancurve_from_preset
Mar  2 00:29:37 smarttop fancurve-set[280968]:     self.fan_curve = self.fancurve_repo.load_by_name(name)
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 1093, in load_by_name
Mar  2 00:29:37 smarttop fancurve-set[280968]:     return FanCurve.load_from_file(self._name_to_filename(name))
Mar  2 00:29:37 smarttop fancurve-set[280968]:   File "/usr/lib/python3/dist-packages/legion_linux/legion.py", line 81, in load_from_file
Mar  2 00:29:37 smarttop fancurve-set[280968]:     with open(filename, 'r', encoding=DEFAULT_ENCODING) as filepointer:
Mar  2 00:29:37 smarttop fancurve-set[280968]: FileNotFoundError: [Errno 2] No such file or directory: '/etc/legion_linux//performance-ac.yaml'
Mar  2 00:29:37 smarttop systemd[1]: legion-linux.service: Main process exited, code=exited, status=1/FAILURE
Mar  2 00:29:37 smarttop systemd[1]: legion-linux.service: Failed with result 'exit-code'.

[... 213 lines cut ...]
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
0	 0	 0	 0	 0	 0	 0	 0	 0	 0
=====================

```
</details>

Flashwalker, 2024-03-02:
I manually downloaded presets files from here https://github.com/johnfanv2/LenovoLegionLinux/tree/main/extra/service/profiles and now error gone.
But looks like applying the preset to HW doesn't really works properly.

**Load from Preset** looks like this:
[image]

And after **Apply to HW** it looks different, like not all of the curve settings are applied:
[image]

Flashwalker, 2024-03-04:
Is there a way i can reset to defaults?

MrDuartePT, 2024-03-29:
If you don’t have the systemd service enable on the `Automation tab` just change to the same power profile again. 

The custom fancurve by the profiles only apply if the system service is enable

MrDuartePT, 2024-03-29:
> **Problem Description** I get error about the script can't find .env and other files when i toggle the performance profiles via **Fn+Q**.
>
> ```
> ...
> /usr/bin/fancurve-set: line 21: [...]
Also this error occur since your didn’t copy `/usr/share/legion_linux` to `/etc/legion_linux`

closed 2024-03-29 by MrDuartePT

Flashwalker, 2024-03-29:
> Also this error occur since your didn’t copy `/usr/share/legion_linux` to `/etc/legion_linux` and enable the systemd service

Thanks