johnfanv2/LenovoLegionLinux #173: Support for Lenovo 7 Pro 16IRX9H (2024)

blargywargy, 2024-03-04:
**Is your feature request related to a problem? Please describe.**
I'd like to request support for the Support for Lenovo 7 Pro 16IRX9H (2024).

**Describe the solution you'd like**
Using the data I have attached to this to help support this model.

 and Debug Info**

Distribution: Nobara 39
Model name: Lenovo 7 Pro 16IRX9H (2024)
CPU model: Intel i9-14900HX
GPU model: Nvidia RTX 4080
Keyboard backlight: RGB
Light in lid or logo: No
Light at IO-Ports at back: No

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83DE
	Version: Legion Pro 7 16IRX9H
	Serial Number:
	UUID:
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83DE_BU_idea_FM_Legion Pro 7 16IRX9H
	Family: Legion Pro 7 16IRX9H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: N2CN18WW
	Release Date: 01/24/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.18
	Firmware Revision: 1.18

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
the debug dir is empty
```

Here is the WMI script ouput run on Windows 11 Pro 23H2

[wmi_script_output_legion_pro_7_16irx9h.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/14480997/wmi_script_output_legion_pro_7_16irx9h.txt)

Here are the fwts and acpi outputs from Nobara 39

[fwts_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/14481008/fwts_re.zip)

[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/14481009/acpi_re.zip)

I'm happy to provide any additional information or test.

blargywargy, 2024-03-24:
I tried with CachyOS and compiled git clone of repo today.

Forcemodulereload to bypass whitelist

dmesg output

```
[Sun Mar 24 11:29:39 2024] legion_laptop: Loading legion_laptop
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: legion_laptop platform driver probing
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 83DE; DMI_BIOS_VERSION:N2CN18WW
[Sun Mar 24 11:29:39 2024] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 83DE; DMI_BIOS_VERSION:N2CN18WW
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 1
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: legion_laptop is forced to load.
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: legion_laptop is forced to load and would otherwise not be loaded
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Using configuration for system: GKCN
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Skipping ACPI _STA check
[Sun Mar 24 11:29:39 2024] legion_laptop: Creating RAM access to embedded controller
[Sun Mar 24 11:29:39 2024] legion_laptop: Successfully mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x00000000119731a5
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Read embedded controller ID 0x5507
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Skipped checking embedded controller id
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Creating debugfs interface
[Sun Mar 24 11:29:39 2024] legion_laptop: Creating sysfs interface
[Sun Mar 24 11:29:39 2024] legion_laptop: Creating hwmon interface
[Sun Mar 24 11:29:39 2024] legion_laptop: Creating platform profile support
[Sun Mar 24 11:29:39 2024] legion_laptop: Init WMI driver support
[Sun Mar 24 11:29:39 2024] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[Sun Mar 24 11:29:39 2024] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[Sun Mar 24 11:29:39 2024] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[Sun Mar 24 11:29:39 2024] legion_laptop: Init keyboard backlight LED driver
[Sun Mar 24 11:29:39 2024] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Mar 24 11:29:39 2024] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Mar 24 11:29:39 2024] legion_laptop: Init Y-Logo LED driver
[Sun Mar 24 11:29:39 2024] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Mar 24 11:29:39 2024] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Mar 24 11:29:39 2024] legion_laptop: Init IO-Port LED driver
[Sun Mar 24 11:29:39 2024] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
[Sun Mar 24 11:29:39 2024] legion_laptop: Error WMI call for reading brightness: expected a value between 1 and 2, but got 0
[Sun Mar 24 11:29:39 2024] legion_laptop: Error reading brightness for light: 5
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: Failed to init IO-Port LED driver. Skipping ...
[Sun Mar 24 11:29:39 2024] legion PNP0C09:00: legion_laptop loaded for this device
[Sun Mar 24 11:29:39 2024] legion_laptop: ACPI result for 8C5B9127-ECD4-4657-980F-851019F99CA5:1: ACPI buffer length: 2
bash -c "./issue-warning.sh"
```

Fancurve

```
cat /sys/kernel/debug/legion/fancurve

EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 80
CPU temperature EC error: 0
CPU temperature EC: 80
CPU temperature ACPI error: 0
CPU temperature ACPI: 55
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 55
GPU temperature error: 0
GPU temperature: 87
[... 115 lines cut ...]
Core 44:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 45:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 46:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 47:       +48.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-0600
Adapter: PCI adapter
Composite:    +47.9°C  (low  =  -0.1°C, high = +85.8°C)
                       (crit = +86.8°C)
Sensor 1:     +42.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +45.0°C

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.34 V
```

jbg7, 2024-05-01:
I'm in a similar state as @blargywargy with Lenovo Legion 7 PRO 16IRX9H (2024) version. I've tested the `forcemodulereload` and the module seems to be loaded fine but the addresses are most probably wrong as I'm getting all zeros from 

```
cat /sys/kernel/debug/legion/fancurve
```

I'd be happy to help tuning for my version and find the right data from the EC chip but I'd appreciate any advice from anyone experienced in that field.

I'd also like to understand if operating the module in this mode (force loaded) brings any good. Without it I don't even know what profile am I switched to using `Fn + Q` (I just suppose the profile switching still works "natively" on a BIOS level and I just don't get the feedback/state to my Linux without LLL). So even if the curve setting, temperatures and fan speeds were unavailable/wrong, would it still make sense to operate LLL in "read-only" mode at least to be able to read the current perf. profile of the EC chip?

Thanks to @johnfanv2 or anyone for any advice, in advance.

blargywargy, 2024-05-01:
@jbg7 I found the same things. Most functions seem to work Fn + Q etc. I could get batter conservation mode working, but from memory, I couldn't find where to disable Rapid Charge. Lenovo Vantage seems to disable Rapid Charge when you enable Battery Conservation.

With tweaking I also found I couldn't get the power consumption as low as with Windows. Fn + Q seems set Windows power settings at the same time. Perhaps there is a way they can be dumped from Windows? 

If there's any extra data or any tests/things I can do to help with this, I'm definitely interested in trying.

grantcarthew, 2024-07-24:
I've also installed using the `forcemodulereload` and then `make install`. I would love to have this device supported.

I had some crashing issues and opened this discussion:
https://forum.endeavouros.com/t/crashing-issues-on-lenovo-legion-pro-7i-16-gen-9/58375/8

Some of the people responding above had some helpful ideas including this package.

I'm documenting my setup and information [in my notes repo](https://github.com/grantcarthew/notes/blob/main/LenovoLegionPro7i.md).

Thanks for the hard work you've done @johnfanv2 and others, it's appreciated.

closed 2026-09-12 by github-actions[bot]