johnfanv2/LenovoLegionLinux #174: Legion Pro 7 16IRX8H -  ALC287 & NVIDIA HDA PRO 9 detected but no internal sound produced

SimonHeggie, 2024-03-05:
**Problem Description**

I have no internal sound from my Nvidia HDA/ALC287 sound card setup.  I cannot get the audiojack to work either it seems. Bluetooth and HDMI output are my only workable solutions right now. I assume since the HDMI is working it's not the HDA that's the issue but the ALC287 realtek card.

I found no information in the readme to address this issue but I notice others have had the problem.
I have run some diagnostics and found that the devices are detected but not communicating properly with pulse.
Bellow is some info and some commands I entered in the hopes of providing more insight.

I have had this problem since I first got the machine. This is a big problem for presenting videos to students in one on one VFX tutoring.

**Model and Debug Info**

Distribution: .................Nobara 39 KDE (6.7.6-201.fsync.fc39.x86_64)
Model name:.................Legion Pro 7 16IRX8H
CPU model:.................13th Gen Intel(R) Core(TM) i9-13900HX
GPU model: .................NVIDIA® GeForce RTX™ 4090 Laptop GPU 16GB
Keyboard backlight:.................Per-Key RGB Backlit, Black - English
Light in lid or logo: no
Light at IO-Ports at back: yes

<details>
  <summary>Output of sudo dmidecode -t system.</summary>
  
```# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
      Manufacturer: LENOVO
      Product Name: 82WQ
      Version: Legion Pro 7 16IRX8H
      Serial Number: #
      UUID: #
      Wake-up Type: Power Switch
      SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
      Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
      Option 1: ConfigOptions1
      Option 2: ConfigOptions2
      Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
      Area Length: 0 bytes
      Header Start Offset: 0x0000
      Header Length: 8192 bytes
      Data Start Offset: 0x2000
      Access Method: General-purpose non-volatile data functions
      Access Address: 0x0000
      Status: Valid, Not Full
      Change Token: 0x12345678
      Header Format: OEM-specific
      Supported Log Type Descriptors: 3
      Descriptor 1: POST memory resize
      Data Format 1: None
      Descriptor 2: POST error
      Data Format 2: POST results bitmap
      Descriptor 3: Log area reset/cleared
      Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
      Status: No errors detected
);
```

</details>
[... 1800 lines cut ...]
        "object.serial": 689,
        "pipewire.access": "unrestricted",
        "pipewire.protocol": "protocol-native",
        "pipewire.sec.gid": 1001,
        "pipewire.sec.pid": 9190,
        "pipewire.sec.socket": "pipewire-0-manager",
        "pipewire.sec.uid": 1000,
        "remote.name": "[pipewire-0-manager,pipewire-0]",
        "settings.check-quantum": false,
        "settings.check-rate": false,
        "window.x11.display": ":0"
      }
    }
  }
]
```

<details>

Any more info required? let me know!
Can't wait to one day listen to spotify as I work without donning bluetooth headphones..

Kind regards,

Simon.

pflanzenandi, 2024-04-04:
Just an idea: Have you tried updating the Linux Kernel?

st0nie, 2024-04-28:
Added to the list already
https://github.com/johnfanv2/LenovoLegionLinux/commit/4b35e0dbbc23c545996c320743211cd22950c539
Can U help test it?

darkpowerxo, 2024-05-26:
I have the Legion Pro 7 16IRX8H (7i with the i9 and 4090) running debian. I can see the audio moving in the audio setting section but can't hear anything. 
help :(

SimonHeggie, 2024-09-09:
Update, with Nobara 40, my ALC287 is working and sounds great... on boot... if I had it previously selected before. Sometimes when I don't use the sound and go to play something, I hear no sound, which tells me it's not longer connected somehow. Then if restart, it's back again. Another surefire way to disable the ALC287 to require a restart to fix this is to put the computer to sleep or into hibernation, particularly if there is another sound device like a monitor plugged in. 

So I guess the nature of the issue has changed for me; it works but only temporarily and I have to restart to get it back.

-S

closed 2026-09-12 by github-actions[bot]