johnfanv2/LenovoLegionLinux #177: [BUG/ERROR] the app has installed successfully but is not opening

deepanshpandey, 2024-03-10:
**Problem Description**
<Please describe your problem here>
Has the bug occurred following steps described in the README? -YES
What steps did you take? - the one that requires add the repo and installing via TAG Repo
What commands did you enter? - As described
What was the (full) error you received and what command or action did lead to it?
after checking logs this was the error message
Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x4200004

**Model and Debug Info**

Distribution: Ubuntu 23.10
Model name: Lenovo Lenovo Legion 5P 15IMH05H
CPU model: i7 10750H
GPU model: NVIDIA RTX 2060
Keyboard backlight: RGB
Light in lid or logo: yes
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82AW
	Version: Lenovo Legion 5P 15IMH05H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82AW_BU_idea_FM_Legion 5P 15IMH05H
	Family: Legion 5P 15IMH05H

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:

```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: EFCN58WW
	Release Date: 11/15/2022
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 11 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.58
	Firmware Revision: 1.58

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:

```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289494
temperature access method: 1
CPU temperature error: 0
CPU temperature: 54
CPU temperature EC error: 0
CPU temperature EC: 54
CPU temperature ACPI error: 0
CPU temperature ACPI: 54
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: -14
CPU temperature WMI2: 0
CPU temperature WMI3 error: -14
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 53
GPU temperature EC error: 0
GPU temperature EC: 53
GPU temperature ACPI error: 0
GPU temperature ACPI: 53
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: -14
GPU temperature WMI2: 0
GPU temperature WMI3 error: -14
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 2616
1 fanspeed EC error: 0
1 fanspeed EC: 2616
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 2600
1 fanspeed WMI error: 0
1 fanspeed WMI: 2600
1 fanspeed WMI2 error: -14
1 fanspeed WMI2: 2600
1 fanspeed WMI3 error: -14
1 fanspeed WMI3: 2600
2 fanspeed error: 0
2 fanspeed: 2604
2 fanspeed EC error: 0
2 fanspeed EC: 2604
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 2600
2 fanspeed WMI error: 0
2 fanspeed WMI: 2600
2 fanspeed WMI2 error: -14
2 fanspeed WMI2: 2600
2 fanspeed WMI3 error: -14
2 fanspeed WMI3: 2600
powermode access method: 3
powermode error: 0
powermode: 1
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 1
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -5
WMI light IO port: -5
WMI light Y logo/lid: -5
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 5
EC fan curve points size: 9
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 5	 7	 0	 127	 0	 52	 0	 38
1700	 1600	 5	 7	 127	 127	 48	 52	 35	 45
2000	 1900	 5	 7	 127	 127	 48	 52	 41	 50
2100	 2100	 5	 7	 127	 127	 48	 52	 47	 127
2300	 2300	 5	 7	 127	 127	 48	 56	 127	 127
2600	 2600	 5	 7	 127	 127	 53	 60	 127	 127
2900	 2900	 5	 7	 127	 127	 57	 64	 127	 127
3500	 3500	 5	 7	 127	 127	 61	 73	 127	 127
3800	 3800	 5	 7	 127	 127	 64	 127	 127	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

MrDuartePT, 2024-03-29:
I never see that type of error, can you send the full error when opening the gui from the latest git commit

MrDuartePT, 2024-05-05:
Closing since is inactive

closed 2024-05-05 by MrDuartePT