johnfanv2/LenovoLegionLinux #178: Support for Lenovo Legion 5   15IMH6

phy-man, 2024-03-10:
**Problem Description**

Has the bug occurred following steps described in the README? Yes

What steps did you take?
Trying to build and test as instructed.

What commands did you enter?

After:
sudo apt-get update
sudo apt-get install -y make gcc linux-headers-$(uname -r) build-essential git lm-sensors wget python3-pyqt5 python3-yaml python3-venv python3-pip python3-argcomplete python3-darkdetect
# Install the following for installation with DKMS
sudo apt-get install dkms openssl mokutil

And then 

git clone https://github.com/johnfanv2/LenovoLegionLinux.git
cd LenovoLegionLinux/kernel_module
make
sudo make reloadmodule

Result is "probe of PNP0C09:00 failed with error -12"

What was the (full) error you received and what command or action did lead to it?

[ned ožu 10 23:05:52 2024] legion_laptop: legion_laptop exit
[ned ožu 10 23:05:52 2024] legion_laptop: Loading legion_laptop
[ned ožu 10 23:05:52 2024] legion PNP0C09:00: legion_laptop platform driver probing
[ned ožu 10 23:05:52 2024] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82NL; DMI_BIOS_VERSION:G8CN25WW
[ned ožu 10 23:05:52 2024] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82NL; DMI_BIOS_VERSION:G8CN25WW
[ned ožu 10 23:05:52 2024] legion PNP0C09:00: is_denied: 0; is_allowed: 0; do_load_by_list: 0; do_load: 0
[ned ožu 10 23:05:52 2024] legion PNP0C09:00: Module not usable for this laptop because it is not in allowlist. Notify the maintainer if you want to add your device or force load with param force.
[ned ožu 10 23:05:52 2024] legion PNP0C09:00: legion_laptop not loaded for this device
[ned ožu 10 23:05:52 2024] legion: **probe of PNP0C09:00 failed with error -12**

This is end  of very long list attached here:
[output list.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/14552083/output.list.txt)

**Model and Debug Info**

Distribution: Linux Mint 21.3
Model name: Legion 5   15IMH6  (2023)
CPU model: Intel i5-10500H @ 2.50GHz
GPU model: NVIDIA GeForce RTX 3050 Laptop GPU
Keyboard backlight:  single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82NL
	Version: Legion 5 15IMH6
	Serial Number: xxxxx
	UUID: xxxxxxxxxxxxx
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82NL_BU_idea_FM_Legion 5 15IMH6
	Family: Legion 5 15IMH6

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: G8CN25WW
	Release Date: 09/20/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 11 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.25
	Firmware Revision: 1.25

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
[???](cat: /sys/kernel/debug/legion/fancurve: No such file or directory)
```

Here are WMI  entries and ACPI tables:
[fwts_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/14552136/fwts_re.zip)
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/14552138/acpi_re.zip)

I see that this error means my laptop is not on the list. I would appreciate if it can get on the list.

I am also running windows 10, so I will try to append its output here later.

Thank you and regards

phy-man, 2024-03-11:
This is output of my Windows 10 pro 22H2 WMI:  
[wmi output win10.txt](https://github.com/johnfanv2/LenovoLegionLinux/files/14552261/wmi.output.win10.txt)

st0nie, 2024-04-28:
Added to the list already
https://github.com/johnfanv2/LenovoLegionLinux/commit/4b35e0dbbc23c545996c320743211cd22950c539
Can U help test it?
need output of `sudo cat /sys/kernel/debug/legion/fancurve`

Dyakovvn, 2024-08-09:
Hi @st0nie 
I have the same laptop. Master and this commitId (4b35e0d)  have a problem. (Kubuntu 22.04)
 `
# sudo dmidecode -t system
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82NL
	Version: Legion 5 15IMH6
	Serial Number: xxxxx
	UUID: xxxxxxxxxxxxx
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82NL_BU_idea_FM_Legion 5 15IMH6
	Family: Legion 5 15IMH6

Handle 0x001F, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0023, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0030, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected
`

`
# sudo dmidecode -t bios
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: G8CN25WW
	Release Date: 09/20/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 11 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.25
	Firmware Revision: 1.25

Handle 0x0020, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0
`

`
[Fri Aug  9 23:07:29 2024] legion: probe of PNP0C09:00 failed with error -12
[Fri Aug  9 23:09:20 2024] legion_laptop: legion_laptop exit
[Fri Aug  9 23:09:20 2024] legion_laptop: Loading legion_laptop
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: legion_laptop platform driver probing
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82NL; DMI_BIOS_VERSION:G8CN25WW
[Fri Aug  9 23:09:20 2024] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82NL; DMI_BIOS_VERSION:G8CN25WW
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: Using configuration for system: G8CN
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: ACPI CFG: 2081289494
[Fri Aug  9 23:09:20 2024] legion_laptop: Creating RAM access to embedded controller
[Fri Aug  9 23:09:20 2024] legion_laptop: Successfully mapped embedded controller: 0xfe0b0400 (in RAM)/0x0 (in EC) to virtual 0x00000000d607586b
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: Read embedded controller ID 0x8227
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: Expected EC chip id 0x5507 but read 0x8227
[Fri Aug  9 23:09:20 2024] legion_laptop: Unloading legion ecram
[Fri Aug  9 23:09:20 2024] legion_laptop: Unloading legion ecram done
[Fri Aug  9 23:09:20 2024] legion_laptop: Unmapping embedded controller memory at 0xfe0b0400 (in RAM)/0x0 (in EC) at virtual 0x00000000d607586b
[Fri Aug  9 23:09:20 2024] legion_laptop: Unloading legion shared
[Fri Aug  9 23:09:20 2024] legion_laptop: Unloading legion shared done
[Fri Aug  9 23:09:20 2024] legion PNP0C09:00: legion_laptop not loaded for this device
[Fri Aug  9 23:09:20 2024] legion: probe of PNP0C09:00 failed with error -12
bash -c "./issue-warning.sh"
  If you have a 2022 or 2023 model, please help testing the new features 'here'
  If you have a light on the lid (Y-logo) or at the IO-ports (all Legion 7 models), please help test controlling it 'here'
  Please give a star on github to support. my goal is to merge the driver into the main Linux kernel, so no recompilation would be required after a Linux update: 'here'
`

`
sudo cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: No such file or directory
`

closed 2026-09-12 by github-actions[bot]