johnfanv2/LenovoLegionLinux #182: [Feature]: `LenovoLegionLinux.hook` should to handle old kernel image files during kernel/module upgrade

nukemiko, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
On Arch Linux, every kernel or module update triggers the hook `LenovoLegionLinux.hook`, thus rebuilding the kernel image via dracut. But each build results in a new kernel image and does not clean up old kernel images (at `<mountpoint of EFI part>/EFI/Linux/Linux-<kernel information>-<random string>-rolling.efi`), thus exhausting the EFI partition's free space after several updates.

**Describe the solution you'd like**
I can't think of a good solution, but it seems possible to tell which kernel image files are old and need to be deleted by the file creation date.

**Describe alternatives you've considered**
Improved LenovoLegionLinux.hook so that the files it produces have the constant filename, allowing the old kernel image file to be overwritten directly.

**Model and Debug Info**

Distribution: Arch Linux
Model name: 82JQ
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA GeForce RTX 3060 Laptop GPU

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82JQ
        Version: Lenovo Legion R9000P2021H
        Serial Number: [REDACTED for privacy]
        UUID: [REDACTED for privacy]
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82JQ_BU_idea_FM_Legion R9000P2021H
        Family: Legion R9000P2021H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: GKCN50WW
        Release Date: 11/24/2021
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.50
        Firmware Revision: 1.50

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
cat: /sys/kernel/debug/legion/fancurve: 没有那个文件或目录
```

MrDuartePT, 2024-03-17:
I not using Arch Linux anymore so make a PR to fix it.

st0nie, 2024-04-29:
https://archlinux.org/packages/extra/any/kernel-modules-hook/
You may need something like this

closed 2024-04-29 by st0nie