johnfanv2/LenovoLegionLinux #184: [BUG] Fan control not working on Lenovo Yoga Pro 7 14ARP8 (BIOS: LNCN21WW)  iwth AMD R7 7735HS.

gsingh704, 2024-03-19:
**Problem Description**
Fan control is not working 
Has the bug occurred following steps described in the README?
Tried building manually , i get error -12. From AUR the GUI loads but fan control is greyed out.
What steps did you take?
Tried installing manually wuth or without dkms
What commands did you enter?
What was the (full) error you received and what command or action did lead to it?
```2024-03-19 16:03:31,390 - INFO: 
2024-03-19 16:03:31,391 - INFO: Feature LockFanController with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/lockfancontroller and path None
2024-03-19 16:03:31,391 - WARNING: Feature LockFanController does not exist. exits: 0
2024-03-19 16:03:31,391 - INFO: Feature RapidChargingFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/rapidcharge and path None
2024-03-19 16:03:31,392 - WARNING: Feature RapidChargingFeature does not exist. exits: 0
2024-03-19 16:03:31,392 - INFO: Feature BatteryConservation with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode and path /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode
2024-03-19 16:03:31,392 - INFO: Feature MaximumFanSpeedFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/fan_fullspeed and path None
2024-03-19 16:03:31,392 - WARNING: Feature MaximumFanSpeedFeature does not exist. exits: 0
2024-03-19 16:03:31,392 - INFO: Feature FnLockFeature with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/fn_lock and path /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/fn_lock
2024-03-19 16:03:31,392 - INFO: Feature WinkeyFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/winkey and path None
2024-03-19 16:03:31,392 - WARNING: Feature WinkeyFeature does not exist. exits: 0
2024-03-19 16:03:31,392 - INFO: Feature TouchpadFeature with pattern ['/sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/touchpad', '/sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/touchpad'] and path None
2024-03-19 16:03:31,393 - WARNING: Feature TouchpadFeature does not exist. exits: 0
2024-03-19 16:03:31,393 - INFO: Feature CameraPowerFeature with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/camera_power and path /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/camera_power
2024-03-19 16:03:31,393 - INFO: Feature OverdriveFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/overdrive and path None
2024-03-19 16:03:31,393 - WARNING: Feature OverdriveFeature does not exist. exits: 0
2024-03-19 16:03:31,393 - INFO: Feature GsyncFeature with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gsync and path None
2024-03-19 16:03:31,393 - WARNING: Feature GsyncFeature does not exist. exits: 0
2024-03-19 16:03:31,393 - INFO: Feature PlatformProfileFeature with pattern /sys/firmware/acpi/platform_profile and path /sys/firmware/acpi/platform_profile
2024-03-19 16:03:31,393 - INFO: Feature StrFileFeature with pattern /sys/firmware/acpi/platform_profile_choices and path /sys/firmware/acpi/platform_profile_choices
2024-03-19 16:03:31,393 - INFO: Feature IsOnPowerSupplyFeature with pattern /sys/class/power_supply/ADP0/online and path None
2024-03-19 16:03:31,393 - WARNING: Feature IsOnPowerSupplyFeature does not exist. exits: 0
2024-03-19 16:03:31,393 - INFO: Feature AlwaysOnUSBChargingFeature with pattern /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/usb_charging and path /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/usb_charging
2024-03-19 16:03:31,394 - INFO: Feature BatteryCurrentCapacityPercentage with pattern /sys/class/power_supply/BAT0/capacity and path /sys/class/power_supply/BAT0/capacity
2024-03-19 16:03:31,394 - INFO: Feature CPUOverclock with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_oc and path None
2024-03-19 16:03:31,394 - WARNING: Feature CPUOverclock does not exist. exits: 0
2024-03-19 16:03:31,394 - INFO: Feature CPULongtermPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_longterm_powerlimit and path None
2024-03-19 16:03:31,394 - WARNING: Feature CPULongtermPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,394 - INFO: Feature CPUShorttermPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_shortterm_powerlimit and path None
2024-03-19 16:03:31,394 - WARNING: Feature CPUShorttermPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,394 - INFO: Feature CPUPeakPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_peak_powerlimit and path None
2024-03-19 16:03:31,394 - WARNING: Feature CPUPeakPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,394 - INFO: Feature CPUDefaultPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_default_powerlimit and path None
2024-03-19 16:03:31,394 - WARNING: Feature CPUDefaultPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,394 - INFO: Feature CPUCrossLoadingPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_cross_loading_powerlimit and path None
2024-03-19 16:03:31,395 - WARNING: Feature CPUCrossLoadingPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature CPUAPUSPPTPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/cpu_apu_sppt_powerlimit and path None
2024-03-19 16:03:31,395 - WARNING: Feature CPUAPUSPPTPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature GPUOverclock with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_oc and path None
2024-03-19 16:03:31,395 - WARNING: Feature GPUOverclock does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature GPUBoostClock with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_boost_clock and path None
2024-03-19 16:03:31,395 - WARNING: Feature GPUBoostClock does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature GPUCTGPPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_ctgp_powerlimit and path None
2024-03-19 16:03:31,395 - WARNING: Feature GPUCTGPPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature GPUPPABPowerLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_ppab_powerlimit and path None
2024-03-19 16:03:31,395 - WARNING: Feature GPUPPABPowerLimit does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature GPUTemperatureLimit with pattern /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/gpu_temperature_limit and path None
2024-03-19 16:03:31,395 - WARNING: Feature GPUTemperatureLimit does not exist. exits: 0
2024-03-19 16:03:31,395 - INFO: Feature YLogoLight with pattern /sys/class/leds/platform::ylogo/brightness and path None
2024-03-19 16:03:31,396 - WARNING: Feature YLogoLight does not exist. exits: 0
2024-03-19 16:03:31,396 - INFO: Feature IOPortLight with pattern /sys/class/leds/platform::ioport/brightness and path None
2024-03-19 16:03:31,396 - WARNING: Feature IOPortLight does not exist. exits: 0
2024-03-19 16:03:31,396 - INFO: CommandFeature PowerProfilesDeamonService: []
2024-03-19 16:03:31,396 - WARNING: Feature PowerProfilesDeamonService does not exist. exits: 0
2024-03-19 16:03:31,396 - INFO: CommandFeature PowerProfilesDeamonService execute "systemctl status power-profiles-daemon"
2024-03-19 16:03:31,437 - INFO: CommandFeature PowerProfilesDeamonService reading with code 0: ● power-profiles-daemon.service - Power Profiles daemon
     Loaded: loaded (/usr/lib/systemd/system/power-profiles-daemon.service; enabled; preset: disabled)
     Active: active (running) since Tue 2024-03-19 15:46:59 CET; 16min ago
   Main PID: 620 (power-profiles-)
      Tasks: 4 (limit: 16367)
     Memory: 1.6M (peak: 2.2M)
        CPU: 76ms
[... 144 lines cut ...]
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.21
	Firmware Revision: 1.21

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
sudo cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: No such file or directory

```

will update with files

gsingh704, 2024-03-19:
[fwts_wmi.log](https://github.com/johnfanv2/LenovoLegionLinux/files/14652451/fwts_wmi.log)
[acpi_re.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/14652453/acpi_re.zip)

gsingh704, 2024-03-19:
Also with sensors there is no fan rpm listed or with inxi 
```
System:
  Host: j-83au Kernel: 6.8.1-arch1-1 arch: x86_64 bits: 64 compiler: gcc
    v: 13.2.1 clocksource: tsc
  Desktop: GNOME v: 45.5 tk: GTK v: 3.24.41 wm: gnome-shell
    tools: gsd-screensaver-proxy dm: GDM v: 45.0.1 Distro: EndeavourOS
    base: Arch Linux
Machine:
  Type: Laptop System: LENOVO product: 83AU v: Yoga Pro 7 14ARP8
    serial: <superuser required> Chassis: type: 10 v: Yoga Pro 7 14ARP8
    serial: <superuser required>
  Mobo: LENOVO model: LNVNB161216 v: SDK0T76463 WIN
    serial: <superuser required> part-nu: LENOVO_MT_83AU_BU_idea_FM_Yoga Pro 7
    14ARP8 uuid: <superuser required> UEFI: LENOVO v: LNCN21WW
    date: 12/28/2023
Battery:
  ID-1: BAT0 charge: 41.6 Wh (55.8%) condition: 74.6/73.0 Wh (102.2%)
    power: 6.4 W volts: 15.8 min: 15.5 model: Sunwoda L22D4PF4 type: Li-poly
    serial: 1798 status: discharging cycles: 2
CPU:
  Info: 8-core model: AMD Ryzen 7 7735HS with Radeon Graphics bits: 64
    type: MT MCP smt: enabled arch: Zen 3+ rev: 1 cache: L1: 512 KiB L2: 4 MiB
    L3: 16 MiB
  Speed (MHz): avg: 462 high: 1397 min/max: 400/4829 cores: 1: 400 2: 400
    3: 400 4: 400 5: 400 6: 400 7: 400 8: 400 9: 400 10: 400 11: 400 12: 400
    13: 400 14: 400 15: 400 16: 1397 bogomips: 102247
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: AMD Rembrandt [Radeon 680M] vendor: Lenovo driver: amdgpu
    v: kernel arch: RDNA-2 pcie: speed: 16 GT/s lanes: 16 ports: active: eDP-1
    empty: DP-1, DP-2, DP-3, DP-4, DP-5, DP-6, HDMI-A-1, Writeback-1
    bus-ID: 72:00.0 chip-ID: 1002:1681 class-ID: 0300 temp: 36.0 C
  Device-2: Bison Integrated RGB Camera driver: uvcvideo type: USB rev: 2.0
    speed: 480 Mb/s lanes: 1 bus-ID: 5-1:2 chip-ID: 5986:215d class-ID: fe01
    serial: 01.00.00
  Display: wayland server: X.org v: 1.21.1.11 with: Xwayland v: 23.2.4
    compositor: gnome-shell driver: gpu: amdgpu display-ID: 0
  Monitor-1: eDP-1 model: AU Optronics 0x7aa7 res: 2560x1600 dpi: 208
    size: 312x195mm (12.28x7.68") diag: 368mm (14.5") modes: max: 2560x1600
    min: 640x480
  API: EGL v: 1.5 hw: drv: amd radeonsi platforms: device: 0 drv: radeonsi
    device: 1 drv: swrast surfaceless: drv: radeonsi wayland: drv: radeonsi x11:
    drv: radeonsi inactive: gbm
  API: OpenGL v: 4.6 compat-v: 4.5 vendor: amd mesa v: 24.0.3-arch1.1
    glx-v: 1.4 direct-render: yes renderer: AMD Radeon Graphics (radeonsi
    rembrandt LLVM 17.0.6 DRM 3.57 6.8.1-arch1-1) device-ID: 1002:1681
    display-ID: :0.0
Audio:
  Device-1: AMD Rembrandt Radeon High Definition Audio vendor: Lenovo
    driver: snd_hda_intel v: kernel pcie: speed: 16 GT/s lanes: 16
    bus-ID: 72:00.1 chip-ID: 1002:1640 class-ID: 0403
  Device-2: AMD ACP/ACP3X/ACP6x Audio Coprocessor vendor: Lenovo
    driver: snd_pci_acp6x v: kernel pcie: speed: 16 GT/s lanes: 16
    bus-ID: 72:00.5 chip-ID: 1022:15e2 class-ID: 0480
  Device-3: AMD Family 17h/19h HD Audio vendor: Lenovo driver: snd_hda_intel
    v: kernel pcie: speed: 16 GT/s lanes: 16 bus-ID: 72:00.6 chip-ID: 1022:15e3
    class-ID: 0403
  API: ALSA v: k6.8.1-arch1-1 status: kernel-api
  Server-1: PipeWire v: 1.0.4 status: active with: 1: pipewire-pulse
    status: active 2: wireplumber status: active 3: pipewire-alsa type: plugin
    4: pw-jack type: plugin
Network:
  Device-1: MEDIATEK MT7922 802.11ax PCI Express Wireless Network Adapter
    vendor: Lenovo driver: mt7921e v: kernel pcie: speed: 5 GT/s lanes: 1
    bus-ID: 01:00.0 chip-ID: 14c3:0616 class-ID: 0280
  IF: wlan0 state: up mac: a8:3b:76:29:88:55
Bluetooth:
  Device-1: Foxconn / Hon Hai Bluetooth Adapter driver: btusb v: 0.8 type: USB
    rev: 2.1 speed: 480 Mb/s lanes: 1 bus-ID: 3-3:2 chip-ID: 0489:e0d8
    class-ID: e001 serial: 000000000
  Report: btmgmt ID: hci0 rfk-id: 2 state: down bt-service: disabled
    rfk-block: hardware: no software: no address: N/A
Drives:
  Local Storage: total: 476.94 GiB used: 13.76 GiB (2.9%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WD PC SN740
    SDDPMQD-512G-1101 size: 476.94 GiB speed: 63.2 Gb/s lanes: 4 tech: SSD
    serial: 2338EX403960 fw-rev: 73116001 temp: 29.9 C scheme: GPT
Partition:
  ID-1: / size: 232.57 GiB used: 13.69 GiB (5.9%) fs: ext4 dev: /dev/nvme0n1p5
  ID-2: /boot/efi size: 256 MiB used: 62.2 MiB (24.3%) fs: vfat
    dev: /dev/nvme0n1p1
Swap:
  Alert: No swap data was found.
Sensors:
  System Temperatures: cpu: 42.0 C mobo: 30.0 C gpu: amdgpu temp: 36.0 C
  Fan Speeds (rpm): N/A
Info:
  Memory: total: 16 GiB note: est. available: 13.34 GiB used: 2.25 GiB (16.9%)
  Processes: 338 Power: uptime: 23m states: freeze,mem,disk suspend: s2idle
    wakeups: 0 hibernate: platform Init: systemd v: 255 default: graphical
  Packages: pm: pacman pkgs: 851 Compilers: gcc: 13.2.1 Shell: Zsh v: 5.9
    running-in: kgx inxi: 3.3.33
```

st0nie, 2024-04-06:
For other ideapad series, this project may not work, it is only for legion series

closed 2026-09-12 by github-actions[bot]