johnfanv2/LenovoLegionLinux #188: Add an option to disconnect the power mode from the fan setting

WojciechSzade, 2024-03-25:
**Is your feature request related to a problem? Please describe.**
I'm using GNOME and I'd like to use the quiet fan setting without using the power saving GNOME's setting.  
I know I can disable the power profiles daemon but that's not what I want to achieve - I want to be able to utilize the GNOME power settings and the fan settings separably - or at least have control over if they're connected or not - how it was working before using your app.

**Describe the solution you'd like**
A setting in the app - "Make the power profiles control the fan profiles" - when turned on turning on the quiet mode enables power saver and the other way around. When off it doesn't. Or even a switch - if I turn on the power saver then the quiet mode is on, but not the other way around - you could use fn+q to change fan modes but it doesn't control the power settings.

**Describe alternatives you've considered**
As above.

**Model and Debug Info**

Distribution: Arch Linux x86_64
Model name: 82RG (Legion 5 Pro 16ARH7H)
CPU model: AMD Ryzen 7 6800H (16) @ 4.79 GHz
GPU model: NVIDIA Geforce RTX 3070 Ti Lapt
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: yes?

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82RG
	Version: Legion 5 Pro 16ARH7H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82RG_BU_idea_FM_Legion 5 Pro 16ARH7H
	Family: Legion 5 Pro 16ARH7H

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0029, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected

```

Output of `sudo dmidecode -t bios`:
```
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: JUCN63WW
	Release Date: 09/18/2023
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.63
	Firmware Revision: 1.63

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 43
CPU temperature EC error: 0
CPU temperature EC: 43
CPU temperature ACPI error: 0
CPU temperature ACPI: 43
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 43
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: 0
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1871
1 fanspeed EC error: 0
1 fanspeed EC: 1871
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1800
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1800
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 1876
2 fanspeed EC error: 0
2 fanspeed EC: 1876
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1800
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1800
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 0
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 1
WMI backlight 2 state: 0
WMI backlight 3 state: 1
WMI light IO port: 0
WMI light y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 2
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 5	 5	 0	 58	 0	 63	 0	 34
1600	 1600	 5	 5	 54	 58	 60	 63	 31	 36
1800	 1800	 5	 5	 54	 58	 60	 63	 33	 43
2000	 2000	 5	 5	 54	 58	 60	 63	 41	 127
2200	 2200	 2	 2	 54	 64	 60	 63	 127	 127
2500	 2600	 2	 2	 62	 66	 60	 63	 127	 127
2800	 2800	 2	 2	 65	 70	 60	 65	 127	 127
3000	 3100	 2	 2	 66	 73	 62	 69	 127	 127
3400	 3500	 2	 2	 70	 87	 66	 81	 127	 127
3700	 3700	 2	 2	 74	 127	 75	 127	 127	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================

```

antheas, 2024-03-29:
The fan mode essentially controls the performance of the legion laptops and is intrinsically linked to the platform profile, so the driver binds them together and that can not change.

However, the platform profile is not inherently linked to the power saving mode of your device. This is done through the power manager, e.g., ppd, tuned etc, that implement the interface which makes the power saving button appear.

At least TuneD has the option to make it ignore the platform profile. The manager you use will probably have the same option. If you make it ignore it, entering power saving mode will no longer mess with the fan curve.

st0nie, 2024-04-06:
When you change the power mode, the platform_profile actually changes, and when the platform_profile changes, the corresponding fan curve is automatically applied, so if you want to keep the fan curve unchanged, one possible way is to softlink all /etc/legion_linux/*.yaml to the same file, so that when switching power mode, the fan curve will be overridden automatically.

closed 2024-04-28 by st0nie