johnfanv2/LenovoLegionLinux #196: [BUG] Error when trying the GUI

growlingman, 2024-04-04:
Hello I also have the same problem with [#160] but I dont understand how is it solved
trying to run `sudo python/legion_linux/legion_linux/legion_gui.py`
produces:
Traceback (most recent call last):
  File "/home/xjken/LenovoLegionLinux/python/legion_linux/legion_linux/legion_gui.py", line 12, in <module>
    from PyQt6 import QtGui, QtCore
ModuleNotFoundError: No module named 'PyQt6'

However I have installed PyQt6 on my own and importing it on python3 terminal return no error.
Thanks

st0nie, 2024-04-06:
How did you install PyQt6 and what is your Linux distribution?
We need more detailed information

st0nie, 2024-04-08:
```shell
pip uninstall PyQt6
sudo apt install python3-pyqt6
```

growlingman, 2024-04-14:
Sorry for the late response. 
`sudo apt install python3-pyqt6` return E: Unable to locate package python3-pyqt6. 
Btw I am actually on Ubuntu 22.04.4 LTS not 20.04, sorry for the mistake.

st0nie, 2024-04-15:
https://packages.ubuntu.com/search?keywords=python3-pyqt6

bloodhori, 2024-04-18:
I have the exact problem on Pop_OS. PyQt6 is installed, i reinstalled it with Pip because apt can't find it. The link provided above has packages for Ubuntu and Firefox deems it a security risk when i try to download it. Am i supposed to add it to the repo list manually?

st0nie, 2024-04-25:
CC  @MrDuartePT
The deb package now seems to rely on python-pyqt5, which should now be changed to python-pyqt6

st0nie, 2024-04-25:
> Sorry for the late response. `sudo apt install python3-pyqt6` return E: Unable to locate package python3-pyqt6. Btw I am actually on Ubuntu 22.04.4 LTS not 20.04, sorry for the mistake.

Ubuntu 24.04 LTS was released, update to it so you can install python-pyqt6

MrDuartePT, 2024-04-25:
> CC @MrDuartePT The deb package now seems to rely on python-pyqt5, which should now be changed to python-pyqt6

This is a problem with github CI being on jammy, I need to update the build process when I have time to use a docker container instead. Also when I do that change I will drop support to Ubuntu 22.04 LTS favor and lower.

MrDuartePT, 2024-05-04:
Well I drop the deb packages since, a debian packages are available, but pyqt6 need to be install manually, forget to change to pyqt6 in the install requirements of the setup.cfg (opps, I gonna do new tag)

MrDuartePT, 2024-05-04:
@thomasgoirand when you can update the debian package to latest TAG to fix the PyQT6 dependencies

closed 2024-05-04 by MrDuartePT

MercifulWind922, 2024-06-29:
On fedora 40, dnf works as well, solved the issue for me.

Flashwalker, 2024-06-29:
I have the same issue after upgrading LenovoLegionLinux GUI.
Ubuntu 22.04

MrDuartePT, 2024-06-29:
> I have the same issue after upgrading LenovoLegionLinux GUI from repo. Ubuntu 22.04. There is no pyqt6 in repos.

The problems is 22.04 upgrade to something newer like 24.04.