johnfanv2/LenovoLegionLinux #198: Could this project support windows?

rango886, 2024-04-06:
Could this project support windows?

st0nie, 2024-04-07:
You can use the official Lenovo legion center application on windows

closed 2024-04-07 by st0nie

rango886, 2024-04-07:
my laptop is r9000k 2021, official Lenovo legion center not support fan control

rango886, 2024-04-07:
I use this project on ubuntu r9000k 2021, this project can control fan speed

st0nie, 2024-04-08:
Use legionfancontrol

rango886, 2024-04-08:
I have tried legionfancontrol, but its fan control logic is different from LenovoLegionLinux. While legionfancontrol has set up fan curves, when the CPU load suddenly increases, the fan speeds up abruptly and then drops back to the set value. On the other hand, LenovoLegionLinux, tested on Ubuntu, maintains a stable fan curve control regardless of load fluctuations.

legionfancontrol struggles to effectively control the fan; the fan tends to intermittently make sudden noise, which is quite annoying.

rango886, 2024-04-08:
legionfancontrol is closed-source, and I cannot make changes to it. I have tried decompiling legionfancontrol, but I found that I couldn't understand the decompiled code. I looked at the code of this project, but I don't have much knowledge of Linux programming, so I also find it difficult to understand.

rango886, 2024-04-15:
I've written a command-line tool that supports controlling fan speed on Windows !!! https://github.com/rango886/LegionFanEmbeddedController