johnfanv2/LenovoLegionLinux #203: Fedora 40 issue

Legion2016, 2024-04-16:
Fedora 40 updated to kernel 6.8.6 
So when I ran this it downgraded kernel to 6.8.5 the test failed. After reboot pc graphics were a mess.  Had to reboot into grub and use newer kernel to get my desktop back to normal.  Haven't seen this before so I figured I would share. I've been using this awesome tool on my legion slim 7 for quite some time and love it a lot.  Is it just I have to wait a few days before running this?  This is the first time I've had issues.

st0nie, 2024-04-27:
nvidia issue

st0nie, 2024-04-28:
I don't think this issue is related to this project, it might just be that your nvidia kernel module doesn't support your old kernel, as far as I know fedora uses dkms or similar technology. legion_laptop's kernel module supports at least version 6.8.7 of the kernel.

closed 2024-04-28 by st0nie