johnfanv2/LenovoLegionLinux #211: [BUG] Can't run `maximumfanspeed` command on 82AX with BIOS EFCN54WW

StarOnTheSky, 2024-04-27:
**Problem Description**
```
❯ legion_cli maximumfanspeed-status
ERROR:legion_linux.legion:Feature MaximumFanSpeedFeature reading error [Errno 22] Invalid argument
ERROR:legion_linux.legion:
<UFW logs removed>
[413237.300391] legion_laptop: WMI evaluation error for: 92549549-4bde-4f06-ac04-ce8bf898dbaa:1

Traceback (most recent call last):
  File "/usr/bin/legion_cli", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.11/site-packages/legion_linux/legion_cli.py", line 471, in main
    args.func(legion, **vars(args))
  File "/usr/lib/python3.11/site-packages/legion_linux/legion_cli.py", line 28, in <lambda>
    func=lambda l, *args, **kwargs: self.command_status_cli(**kwargs))
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/legion_linux/legion_cli.py", line 56, in command_status_cli
    return self.command_status()
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/legion_linux/legion_cli.py", line 137, in command_status
    print(self.model.maximum_fanspeed.get())
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/legion_linux/legion.py", line 335, in get
    invalue = self._read_file_int(self.filename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/legion_linux/legion.py", line 269, in _read_file_int
    return int(self._read_file_str(file_path))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/legion_linux/legion.py", line 266, in _read_file_str
    raise err
  File "/usr/lib/python3.11/site-packages/legion_linux/legion.py", line 260, in _read_file_str
    out = str(filepointer.read()).strip()
              ^^^^^^^^^^^^^^^^^^
OSError: [Errno 22] Invalid argument
```
Full dmesg:
```
❯ dmesg|grep legion
[412691.654963] legion_laptop: Loading legion_laptop
[412691.655026] legion PNP0C09:00: legion_laptop platform driver probing
[412691.655028] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82AX; DMI_BIOS_VERSION:EFCN54WW
[412691.655030] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82AX; DMI_BIOS_VERSION:EFCN54WW
[412691.655032] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[412691.655033] legion PNP0C09:00: Using configuration for system: EFCN
[412691.655062] legion PNP0C09:00: ACPI CFG: 2081289494
[412691.655063] legion_laptop: Creating RAM access to embedded controller
[412691.655080] legion_laptop: Successfully mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x000000008441e128
[412691.655433] legion PNP0C09:00: Read embedded controller ID 0x8227
[412691.655434] legion PNP0C09:00: Creating debugfs interface
[412691.655440] legion_laptop: Creating sysfs interface
[412691.655455] legion_laptop: Creating hwmon interface
[412691.655510] legion_laptop: Creating platform profile support
[412691.655512] legion_laptop: Init WMI driver support
[412691.655529] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[412691.655545] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[412691.655555] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[412691.655565] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[412691.655573] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[412691.655581] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[412691.655589] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[412691.655616] legion_laptop: Init keyboard backlight LED driver
[412691.655619] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[412691.655620] legion_laptop: Error for WMI method call to get brightness
[412691.655620] legion_laptop: Error reading keyboard brightness
[412691.655621] legion PNP0C09:00: Failed to init keyboard backlight LED driver. Skipping ...
[412691.655621] legion_laptop: Init Y-Logo LED driver
[412691.655623] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[412691.655623] legion_laptop: Error for WMI method call to get brightness
[412691.655624] legion_laptop: Error reading brightness for light: 3
[... 196 lines cut ...]
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: -22
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 9
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0        0       2       2       0       56      0       60      0       38
1700     1600    2       2       51      56      57      60      35      45
2000     1900    2       2       51      56      57      60      41      127
2100     2100    2       2       51      61      57      60      127     127
2300     2300    2       2       57      70      57      60      127     127
2600     2600    2       2       67      75      57      60      127     127
2900     2900    2       2       71      83      57      64      127     127
3500     3500    2       2       77      87      61      74      127     127
3800     3800    2       2       80      127     71      77      127     127
4500     4500    2       2       127     127     74      127     127     127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

st0nie, 2024-04-28:
https://github.com/johnfanv2/LenovoLegionLinux/blob/4b35e0dbbc23c545996c320743211cd22950c539/kernel_module/legion-laptop.c#L411

I think it might be due to your model using the wrong access_method_fanfullspeed, you can try changing its value to ACCESS_METHOD_WMI3 and report back if it works

StarOnTheSky, 2024-04-28:
> https://github.com/johnfanv2/LenovoLegionLinux/blob/4b35e0dbbc23c545996c320743211cd22950c539/kernel_module/legion-laptop.c#L411
> 
> I think it might be due to your model using the wrong access_method_fanfullspeed, you can try changing its value to `ACCESS_METHOD_WMI3` and report back if it works

```
❯ dmesg|grep legion
[530734.066590] legion_laptop: Loading legion_laptop
[530734.066652] legion PNP0C09:00: legion_laptop platform driver probing
[530734.066655] legion PNP0C09:00: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82AX; DMI_BIOS_VERSION:EFCN54WW
[530734.066656] legion_laptop: Read identifying information: DMI_SYS_VENDOR: LENOVO; DMI_PRODUCT_NAME: 82AX; DMI_BIOS_VERSION:EFCN54WW
[530734.066658] legion PNP0C09:00: is_denied: 0; is_allowed: 1; do_load_by_list: 1; do_load: 1
[530734.066661] legion PNP0C09:00: Using configuration for system: EFCN
[530734.066696] legion PNP0C09:00: ACPI CFG: 2081289494
[530734.066697] legion_laptop: Creating RAM access to embedded controller
[530734.066713] legion_laptop: Successfully mapped embedded controller: 0xfe00d400 (in RAM)/0x0 (in EC) to virtual 0x000000008441e128
[530734.067051] legion PNP0C09:00: Read embedded controller ID 0x8227
[530734.067052] legion PNP0C09:00: Creating debugfs interface
[530734.067059] legion_laptop: Creating sysfs interface
[530734.067070] legion_laptop: Creating hwmon interface
[530734.067120] legion_laptop: Creating platform profile support
[530734.067122] legion_laptop: Init WMI driver support
[530734.067138] legion_wmi 887B54E3-DDDC-4B2C-8B88-68A26A8835D0: Register after probing for WMI.
[530734.067153] legion_wmi BFD42481-AEE3-4501-A107-AFB68425C5F8: Register after probing for WMI.
[530734.067161] legion_wmi D062906B-12D4-4510-999D-4831EE80E985: Register after probing for WMI.
[530734.067168] legion_wmi BFD42481-AEE3-4502-A107-AFB68425C5F8: Register after probing for WMI.
[530734.067178] legion_wmi BC72A435-E8C1-4275-B3E2-D8B8074ABA59: Register after probing for WMI.
[530734.067188] legion_wmi 10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1: Register after probing for WMI.
[530734.067199] legion_wmi D320289E-8FEA-41E0-86F9-611D83151B5F: Register after probing for WMI.
[530734.067232] legion_laptop: Init keyboard backlight LED driver
[530734.067234] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[530734.067235] legion_laptop: Error for WMI method call to get brightness
[530734.067236] legion_laptop: Error reading keyboard brightness
[530734.067236] legion PNP0C09:00: Failed to init keyboard backlight LED driver. Skipping ...
[530734.067237] legion_laptop: Init Y-Logo LED driver
[530734.067238] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[530734.067238] legion_laptop: Error for WMI method call to get brightness
[530734.067239] legion_laptop: Error reading brightness for light: 3
[530734.067239] legion PNP0C09:00: Failed to init Y-Logo LED driver. Skipping ...
[530734.067240] legion_laptop: Init IO-Port LED driver
[530734.067241] legion_laptop: ACPI evaluation error for: 8C5B9127-ECD4-4657-980F-851019F99CA5:1
[530734.067242] legion_laptop: Error for WMI method call to get brightness
[530734.067242] legion_laptop: Error reading brightness for light: 5
[530734.067243] legion PNP0C09:00: Failed to init IO-Port LED driver. Skipping ...
[530734.067243] legion PNP0C09:00: legion_laptop loaded for this device
[530761.954579] legion_laptop: No access method for fan full speed: 5
```

st0nie, 2024-04-28:
try ACCESS_METHOD_EC
ACCESS_METHOD_EC2
ACCESS_METHOD_EC3

StarOnTheSky, 2024-04-29:
> try ACCESS_METHOD_EC ACCESS_METHOD_EC2 ACCESS_METHOD_EC3

None works.
EC: [566645.048700] legion_laptop: Unexpected value in maximumfanspeed register: 26
EC2: [566676.410413] legion_laptop: No access method for fan full speed: 10
EC3: [566707.515474] legion_laptop: No access method for fan full speed: 11

st0nie, 2024-04-29:
Maybe your model doesn't have ExtremeCooling support, have you tested it on Windows?

StarOnTheSky, 2024-04-29:
> Maybe your model doesn't have ExtremeCooling support, have you tested it on Windows?

How to enable "extremecooling" on Windows?

st0nie, 2024-04-29:
https://www.legionfancontrol.com/
Maybe this will work?

st0nie, 2024-04-29:
> https://www.legionfancontrol.com/ Maybe this will work? But it's a closed source program, so we don't know exactly how it works. Don't feel too bad though, this feature will make your laptop as noisy as a jet. If your laptop's temperature is high, probably setting the fan curve is a better solution. I never use this feature.
>
> It can make the fans run up to 4400 RPM, but I didn't see anything about "extreme cooling".

I think they call this Turbo Mode

StarOnTheSky, 2024-04-29:
Okay, thanks for replying. Maybe 4500 RPM is fast enough for me to train models.

closed 2024-04-29 by StarOnTheSky