johnfanv2/LenovoLegionLinux #213: [BUG] After the latest update in Garuda LenovoLegionLinux doesnt work anymore.

themadrigal, 2024-04-30:
Please be easy on me as I am a bit of a nood and had a friend help me install LenovoLegionLinux and it was working awesome but the latest update on Garuda seems to have broken it and my friend says he doesnt work with Dracut and can't help me. 

This is the error I get in the consol when trying to run LenovoLegionLinux

╭─legion@LEGION in ~
╰─λ LenovoLegionLinux
dracut[I]: Executing: /usr/bin/dracut --kver=6.8.7-zen1-2-zen -f --uefi
dracut[F]: No permission to write to /boot/efi/EFI/Linux.
dracut[I]: Executing: /usr/bin/dracut --kver=6.8.8-zen1-1-zen -f --uefi
dracut[F]: No permission to write to /boot/efi/EFI/Linux.

It was working fine until the latest updates, how can I fix this?

themadrigal, 2024-04-30:
╭─legion@LEGION in ~ took 1s
[🔴] × sudo LenovoLegionLinux
[sudo] password for legion:
dracut[I]: Executing: /usr/bin/dracut --kver=6.8.8-arch1-1 -f --uefi
dracut[F]: Can't find a kernel image '/boot/vmlinuz-6.8.8-arch1-1' to create a UEFI executable
dracut[I]: Executing: /usr/bin/dracut --kver=6.8.8-zen1-1-zen -f --uefi
dracut[W]: If you need to use bluetooth, please include it explicitly.
dracut[I]: memstrack is not available
dracut[I]: If you need to use rd.memdebug>=4, please install memstrack and procps-ng
dracut[W]: If you need to use bluetooth, please include it explicitly.
dracut[I]: memstrack is not available
dracut[I]: If you need to use rd.memdebug>=4, please install memstrack and procps-ng
dracut[I]: *** Including module: systemd ***
dracut[I]: *** Including module: systemd-initrd ***
dracut[I]: *** Including module: i18n ***
dracut[I]: *** Including module: drm ***
dracut[I]: *** Including module: plymouth ***
dracut[I]: *** Including module: btrfs ***
dracut[I]: *** Including module: kernel-modules ***
dracut[I]: *** Including module: kernel-modules-extra ***
dracut[I]: *** Including module: btrfs-snapshot-overlay ***
dracut[I]: *** Including module: rootfs-block ***
dracut[I]: *** Including module: terminfo ***
dracut[I]: *** Including module: udev-rules ***
dracut[I]: *** Including module: dracut-systemd ***
dracut[I]: *** Including module: usrmount ***
dracut[I]: *** Including module: base ***
dracut[I]: *** Including module: fs-lib ***
dracut[I]: *** Including module: shutdown ***
dracut[I]: *** Including modules done ***
dracut[I]: *** Installing kernel module dependencies ***
dracut[I]: *** Installing kernel module dependencies done ***
dracut[I]: *** Resolving executable dependencies ***
dracut[I]: *** Resolving executable dependencies done ***
dracut[I]: *** Hardlinking files ***
dracut[I]: *** Hardlinking files done ***
dracut[I]: *** Generating early-microcode cpio image ***
dracut[I]: *** Constructing AuthenticAMD.bin ***
dracut[I]: *** Store current command line parameters ***
dracut[I]: *** Stripping files ***
dracut[I]: *** Stripping files done ***
dracut[I]: *** Creating image file '/boot/efi/EFI/Linux/linux-6.8.8-zen1-1-zen-0f6f91570d904824bba5f4a98ebc72fa-rolling.efi' ***
cp: error writing '/boot/efi/EFI/Linux/linux-6.8.8-zen1-1-zen-0f6f91570d904824bba5f4a98ebc72fa-rolling.efi': No space left on device
dracut[F]: Creation of /boot/efi/EFI/Linux/linux-6.8.8-zen1-1-zen-0f6f91570d904824bba5f4a98ebc72fa-rolling.efifailed

st0nie, 2024-05-05:
cp: error writing '/boot/efi/EFI/Linux/linux-6.8.8-zen1-1-zen-0f6f91570d904824bba5f4a98ebc72fa-rolling.efi': No space left on device

I think U should delete some old kernel to free your esp

st0nie, 2024-05-05:
Not LLL issue, anyway

closed 2024-05-05 by st0nie

st0nie, 2024-05-05:
maybe dupe of #214