johnfanv2/LenovoLegionLinux #218: Support for Lenovo Legion Slim 5 16AHP9

mato157, 2024-05-11:
Is your feature request related to a problem? Please describe.
I'd like to request support for the Support for Lenovo Legion Slim 5 16AHP9 (2024).

Describe the solution you'd like
I have attached data and debug to help support this model.

Distribution: Ubuntu 24.04
Model name: Lenovo Legion Slim 16AHP9
CPU model: AMD Ryzen 7 8845HS
GPU model: Nvidia RTX 4070
Keyboard backlight: RGB
Light in lid or logo: No
Light at IO-Ports at back: No

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:

````
 dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.6.0 present.
 SMBIOS implementations newer than version 3.5.0 are not
 fully supported by this version of dmidecode.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 83DH
	Version: Legion Slim 5 16AHP9
	Serial Number:
	UUID:
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_83DH_BU_idea_FM_Legion Slim 5 16AHP9
	Family: Legion Slim 5 16AHP9

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002E, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
````

Output of `sudo dmidecode -t bios`:

```` dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.6.0 present.
 SMBIOS implementations newer than version 3.5.0 are not
 fully supported by this version of dmidecode.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: NRCN16WW
	Release Date: 03/26/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 32 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.16
	Firmware Revision: 1.16

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
````

Output of `sudo cat /sys/kernel/debug/legion/fancurve` after using `sudo make forcereloadmodule`:

````
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 4228186128
temperature access method: 1
CPU temperature error: 0
CPU temperature: 0
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: -5
CPU temperature ACPI: 0
CPU temperature WMI error: 0
CPU temperature WMI: 42
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
CPU temperature WMI3 error: 0
CPU temperature WMI3: 42
GPU temperature error: 0
GPU temperature: 0
GPU temperature EC error: 0
GPU temperature EC: 0
GPU temperature ACPI error: -5
GPU temperature ACPI: 0
GPU temperature WMI error: 0
GPU temperature WMI: 45
GPU temperature WMI2 error: 0
GPU temperature WMI2: 0
GPU temperature WMI3 error: 0
GPU temperature WMI3: 45
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 0
1 fanspeed EC error: 0
1 fanspeed EC: 0
1 fanspeed ACPI error: -5
1 fanspeed ACPI: 0
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 0
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 1900
2 fanspeed error: 0
2 fanspeed: 0
2 fanspeed EC error: 0
2 fanspeed EC: 0
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1900
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 0
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 1900
powermode access method: 3
powermode error: 0
powermode: 2
powermode EC error: 0
powermode EC: 90
powermode ACPI error: -5
powermode ACPI: 0
powermode WMI error: 0
powermode WMI: 2
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 10
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
8000	 0	 0	 0	 0	 0	 0	 0	 0	 255
9000	 0	 0	 0	 0	 0	 0	 0	 0	 255
2900	 0	 0	 0	 0	 0	 0	 0	 0	 255
8000	 0	 0	 0	 0	 0	 0	 0	 0	 255
9400	 0	 0	 0	 0	 0	 0	 0	 0	 255
3500	 0	 0	 0	 0	 0	 0	 0	 0	 255
8200	 0	 0	 0	 0	 0	 0	 0	 0	 255
10500	 0	 0	 0	 0	 0	 0	 0	 0	 255
3500	 0	 0	 0	 0	 0	 0	 0	 0	 255
9400	 0	 0	 0	 0	 0	 0	 0	 0	 255
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
````

Here are WMI :
[fwts_log.zip](https://github.com/johnfanv2/LenovoLegionLinux/files/15284458/fwts_log.zip)

I see that  laptop is not on the list of supported devices.

Thank you

RWL-69, 2024-08-03:
I will second this request.  
Same laptop, will try to get the data to add to see if there is any difference as it is not my laptop.

One issue that occurs using Fedora 40.  Laptop powers down if the lid is closed when the laptop has been put to sleep/suspend.  Power light on back of laptop that indicates that the power block is plugged in and supplying power turns off.  Doesn't do this in Windows.

RWL-69, 2024-08-03:
Asked the person to generate the necessary files.  Running Fedora 40 and using the COPR repository for installation of the GUI.  Not in a condition to compile anything if it is needed at this time.

Installed from fedora copr repo

Model Name: Legion Slim 5 16AHP9
CPU model: AMD Ryzen 7 8845HS w/ Radeon 780M
GPU model: NVIDIA RTX 4060
Keyboard Backlight: 4-zone RGB

`TEST`

`output of dmidecode -t system:`

`# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.6.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 83DH
        Version: Legion Slim 5 16AHP9
        Serial Number:
        UUID:
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_83DH_BU_idea_FM_Legion Slim 5 16AHP9
        Family: Legion Slim 5 16AHP9

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002E, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected

RWL-69, 2024-08-03:
Cannot use 7z files which is stupid IMHO.

Installed requested files from fedora copr repo and multiple reboots.  Cannot compile any programs at this time or run make.
-------------------

Model Name: Legion Slim 5 16AHP9
CPU model: AMD Ryzen 7 8845HS w/ Radeon 780M
GPU model: NVIDIA RTX 4060
Keyboard Backlight: 4-zone RGB

output of dmidecode -t system:

```# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.6.0 present.

`Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 83DH
        Version: Legion Slim 5 16AHP9
        Serial Number:
        UUID:
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_83DH_BU_idea_FM_Legion Slim 5 16AHP9
        Family: Legion Slim 5 16AHP9

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options`
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002E, DMI type 32, 20 bytes
System Boot Information
       Status: No errors detected

```

Output of dmidecode -t bios

```# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.6.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: NRCN17WW
        Release Date: 04/11/2024
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.17
        Firmware Revision: 1.17

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
        Language Description Format: Long
        Installable Languages: 4
                en|US|iso8859-1
                fr|FR|iso8859-1
                ja|JP|unicode
                zh|TW|unicode
        Currently Installed Language: en|US|iso8859-1

```

_do not have file /sys/kernel/debug/legion/fancurve_ 

Was not created by anything from the COPR repository unless there is a different command that will create it without compiling.

Output of iasl -e SSDT* -d DSDT
Attempted to generate DSDT.dsl, output file was empty. Copied terminal output to this file instead:

[ACPI_Table_Data.zip](https://github.com/user-attachments/files/16478005/ACPI_Table_Data.zip)

Linux WMI log file

[FWTS_WMI.zip](https://github.com/user-attachments/files/16478007/FWTS_WMI.zip)

Windows WMI log file

[Windows_WMI.zip](https://github.com/user-attachments/files/16478044/Windows_WMI.zip)

**If there is more information needed, please comment and I will strive to get it ASAP.**

ryanabx, 2024-08-22:
I have a `Legion Slim 5 16APH8`, so I'm similarly interested in support for 2024 devices.

petyy, 2024-08-27:
Hello. On on the same boat as @ryanabx, @RWL-69 and @mato157. I builded the module on Ubuntu 24.04 and I'm getting the same results/output as others, sadly :-(   If I could help debugging somehow, tag me, pls.

HakaishinShwet, 2024-09-09:
if possible please add support for 16APH8 too

awsms, 2024-09-17:
Using this issue to ask if you guys have the same issue as I do, the DisableWhileTyping Xorg conf doesn't work for the touchpad. While xinput list-props id shows that it's indeed disabled, it definitely isn't.
Another thing, were any of you able to toggle the keyboard backlight (I've the default white one) in any other way than Fn+Space Bar?
and of course +1 for any patch that could make the driver work on 16AHP9.

Evthestrike, 2024-09-28:
This would be great!

Evthestrike, 2024-11-21:
I'm willing to help with getting this supported

ryanabx, 2024-11-21:
There's not a lot of new activity in the repo, I wonder if the project is dead or in maintenance mode

lesmic11, 2024-12-02:
Hello, 
I also have a Legion Slim 5 16AHP9 (AMD Ryzen 7 8845HS w/ Radeon 780M Graphics + NVIDIA GeForce RTX 4060) and also humbly request support for this amazing piece of hardware.

Mulukulum, 2024-12-10:
Thanks for posting the High Idle power fix. I was using supergfxctl to switch it to integrated to hopefully save power, but seems like the battery life is better in Hybrid mode. This laptop is actually usable on battery now. 

I also have had issues with suspending the laptop and I posted about it [here](https://forums.lenovo.com/t5/Fedora/Legion-Slim-5-Gen-9-16AHP9-Ryzen-7-8845HS-4060-turns-OFF-when-trying-to-Suspend-on-Linux/m-p/5331953?random=hp4UgLina7bvPB_9CKG_xLZJC3QocKM8KPWRxNJTNIw_1be6351d613157f9fe39949dc7b9b7c8&page=1#6435539)
Sometimes I can shut the lid and it works just fine, other times it completely shuts off. It does however work 100% consistently when I choose the suspend option in Gnome, so I think its probably something to do with the lid.

I don't think you get fine grained fan control for this laptop, since you can't do it from windows either, and I'm not sure how you can change the keyboard RGB values like you can from windows. 

My only major thing left now is to find a way to change keyboard lighting from linux.

Mulukulum, 2024-12-11:
Hey folks, I've figured out that the unexpected shutdowns when trying to suspend does not occur if the `ideapad_laptop` module is not loaded. I also found [this project](https://github.com/4JX/L5P-Keyboard-RGB/) to control the lighting. I'm not sure if its possible to get the keyboard supported via OpenRGB, but I plan to try during winter break.

As for whats going on with `ideapad_laptop` and suspend, I have no clue what to do. I'm hoping there's some setting I can change to magically fix this but I'm not sure where to even start, since I'm not too familiar with who exactly maintains these modules. 

How to reproduce suspend issue:
- Plug out laptop from A/C power. Ensure `ideapad_laptop` module is loaded.
- Close the lid and wait for suspend (this works fine for me)
- Plug in A/C power, and you should get the charging light come on for a few seconds, before turning off and then turning back on again. Your keyboard lighting should now be reset.
- Try this again with `modprobe.blacklist=ideapad_laptop` added to kernel boot params and your laptop shouldn't shutdown when suspending.

ryanabx, 2024-12-11:
> Hey folks, I've figured out that the unexpected shutdowns when trying to suspend does not occur if the `ideapad_laptop` module is not loaded. I also found [this [...]

We need to find out where to report the issue. This definitely shouldn't be happening. I can also reproduce

codyit, 2024-12-11:
> unexpected shutdowns when trying to suspend does not occur if the ideapad_laptop module is not loaded

Nice find. 

> Try this again with `modprobe.blacklist=ideapad_laptop` added to kernel boot params and your laptop shouldn't shutdown when suspending.

When I looked into the source briefly, the code does intentionally associate by detecting Legion 5 fn keys WMI GUID and the seemingly Lenovo universal ACPI platform device ID, so the driver name is a bit of a misnomer at its currently state. There's a case to be made for a clean solution dealing with these family of similar but different devices, that's a pending discussion between maints here and the in tree kernel driver, there are more nuances regarding new models ACPI-WMI interface they have to talk about.

But before LLL support is done here, for now it's all we've got meaning disabling that would disable the ACPI platform driver thus all fn keys (top row, backlight, touchpad state, fn+Q performance profile change, fn+space keyboard backlight toggle among others) and interface exposing things like conservation mode and usb charging. Personally I'd just live with the suspend problem for now.

Mulukulum, 2024-12-12:
Yeah, still would be nice to have this fixed though. ideapad_laptop seems to work great. Would it be possible to submit this as a bug to the kernel developers and request some sort of workaround for this issue for this specific model? I had a look at the source and I don't really understand most of whats there. I'm using `acpi_listen` to listen for events and so far this is what I've got:

Plug in A/C
```
wmi PNP0C14:03 000000d7 00000000
wmi PNP0C14:03 000000d9 00000000
ac_adapter ACPI0003:00 00000080 00000001
battery PNP0C0A:00 00000080 00000001
wmi PNP0C14:03 000000d7 00000000
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000db 00000000
```

Plug out A/C
```
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000d7 00000000
wmi PNP0C14:03 000000d9 00000000
ac_adapter ACPI0003:00 00000080 00000000
battery PNP0C0A:00 00000080 00000001
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000d9 00000000
```
Lid Close/Open
```
wmi PNP0C14:03 000000d4 00000000
button/lid LID close
button/lid LID open
wmi PNP0C14:03 000000d4 00000000
```

And this is what happened when the laptop shut off. 
```
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000d7 00000000
wmi PNP0C14:03 000000d9 00000000
ac_adapter ACPI0003:00 00000080 00000000
battery PNP0C0A:00 00000080 00000001
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000d9 00000000
wmi PNP0C14:03 000000d4 00000000
button/lid LID close
wmi PNP0C14:03 000000d9 00000000
```

codyit, 2024-12-16:
Might be useful, so I'd just put it here. The 17 fwts failures with their corresponding WMI event from Windows perspective if found.
```
#####
Class Name: LENOVO_UTILITY_EVENT
Class GUID: {8fc0de0c-b4e4-43fd-b0f3-8871711c1294}
#####
Class Name: LENOVO_GAMEZONE_TEMP_EVENT
Class GUID: {BFD42481-AEE3-4501-A107-AFB68425C5F8}
#####
Class Name: LENOVO_GAMEZONE_OC_EVENT
Class GUID: {D062906B-12D4-4510-999D-4831EE80E985}
#####
Class Name: LENOVO_GAMEZONE_GPU_TEMP_EVENT
Class GUID: {BFD42481-AEE3-4502-A107-AFB68425C5F8}
#####
Class Name: LENOVO_GAMEZONE_FAN_COOLING_EVENT
Class GUID: {BC72A435-E8C1-4275-B3E2-D8B8074ABA59}
#####
Class Name: LENOVO_GAMEZONE_KEYLOCK_STATUS_EVENT
Class GUID: {10AFC6D9-EA8B-4590-A2E7-1CD3C84BB4B1}
#####
Class Name: LENOVO_GAMEZONE_SMART_FAN_MODE_EVENT
Class GUID: {D320289E-8FEA-41E0-86F9-611D83151B5F}
#####
Class Name: LENOVO_GAMEZONE_SMART_FAN_SETTING_EVENT
Class GUID: {D320289E-8FEA-41E1-86F9-611D83151B5F}
#####
Class Name: LENOVO_GAMEZONE_POWER_CHARGE_MODE_EVENT
Class GUID: {D320289E-8FEA-41E0-86F9-711D83151B5F}
#####
Class Name: LENOVO_GAMEZONE_LIGHT_PROFILE_CHANGE_EVENT
Class GUID: {D320289E-8FEA-41E0-86F9-811D83151B5F}
#####
Class Name: LENOVO_GAMEZONE_THERMAL_MODE_EVENT
Class GUID: {D320289E-8FEA-41E0-86F9-911D83151B5F}
#####
Class Name: LENOVO_LIGHTING_EVENT
Class GUID: {1e3391a1-2c89-464d-95d9-3028b72e7a33}
#####
Class Name: LENOVO_AC_PD_EVENT
Class GUID: {be1346e5-3a32-4a31-b8b0-2cd195e1f572}
#####
9D746A02-1536-B44D-A3B3-79D9D53F7227
(not found)
#####
Class Name: LENOVO_DISPATCHER_EVENT
Class GUID: {0F7B30EC-AC90-4B34-B62F-87F85115505D}
#####
CCC6A667-0A38-254D-AA35-A1397D1FBDFB
(not found)
#####
2C3FC11C-2CEA-9741-95A3-060E2285A645
(not found)
```

codyit, 2024-12-16:
@ryanabx
For the suspend issue, a similar problem is already tracked by the module maintainer's bug tracker https://bugs.launchpad.net/ideapad-laptop/+bug/2068785 since 6 months ago

Apparently it's quite widespread across newer Lenovo laptops when I look around, small differences in terms of reported symptom but revolving around resume after suspend. Unfortunately there's not much traction.

@Mulukulum 
So if you would like to coordinate a contribution that would be the right place.

Evthestrike, 2025-02-12:
My laptop model is "Legion Slim 5 16ARP9"

Evthestrike, 2025-02-12:
I have nvidia-powerd and power-profiles-daemon and it's not working :/  It does not change the power limit either.

lesmic11, 2025-02-13:
> Also, has anyone gotten "Fn + Q" to actually toggle the power profiles? My light will switch between blue and white but never to red for some reason, and when I boot into the BIOS I can see that the power profile is still stuck on "Balanced" regardless of the color of the light.
>
> #218 (comment)<https://github.com/johnfanv2/LenovoLegionLinux/issues/218#issuecomment-2531317758>

I’m getting the red light under the latest Linux Mint district.
The Lenovo SLIM’s BIOS won’t allow the power setting on RED on battery power. It must be connected to AC power.

…
M

lesmic11, 2025-02-13:
BTW my laptop is a LENOVO LEGION SLIM5 (16AHP9).

It's configured for dual-boot with Win11_24H2 & Linux Mint 22.1 (kernel 
6.8.0-52-generic)

System:
   Kernel: 6.8.0-52-generic arch: x86_64 bits: 64 compiler: gcc v: 
13.3.0 clocksource: tsc
   Desktop: Cinnamon v: 6.4.6 tk: GTK v: 3.24.41 wm: Muffin v: 6.4.1 vt: 
7 dm: LightDM v: 1.30.0
     Distro: Linux Mint 22.1 Xia base: Ubuntu 24.04 noble

Machine:
   Type: Laptop System: LENOVO product: 83DH v: Legion Slim 5 16AHP9 
serial: <superuser required>
     Chassis: type: 10 v: Legion Slim 5 16AHP9 serial: <superuser required>
   Mobo: LENOVO model: LNVNB161216 v: SDK0T76463 WIN serial: <superuser 
required>
     part-nu: LENOVO_MT_83DH_BU_idea_FM_Legion Slim 5 16AHP9 uuid: 
<superuser required> UEFI: LENOVO
     v: NRCN21WW date: 11/12/2024

CPU:
   Info: 8-core model: AMD Ryzen 7 8845HS w/ Radeon 780M Graphics bits: 
64 type: MT MCP smt: enabled
     arch: Zen 4 rev: 2 cache: L1: 512 KiB L2: 8 MiB L3: 16 MiB
   Speed (MHz): avg: 1405 high: 3240 min/max: 400/5137 cores: 1: 3240 2: 
400 3: 1730 4: 1741
     5: 1314 6: 400 7: 1297 8: 400 9: 1767 10: 400 11: 2250 12: 3130 13: 
2162 14: 400 15: 1458 16: 400
     bogomips: 121362
   Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm

Graphics:
   Device-1: NVIDIA AD107M [GeForce RTX 4060 Max-Q / Mobile] vendor: 
Lenovo driver: nvidia
     v: 550.120 arch: Lovelace pcie: speed: 2.5 GT/s lanes: 8 ports: 
active: none
     empty: DP-9,HDMI-A-1,eDP-2 bus-ID: 01:00.0 chip-ID: 10de:28e0 
class-ID: 0300
   Device-2: AMD Phoenix3 vendor: Lenovo driver: amdgpu v: kernel arch: 
RDNA-3 pcie:
     speed: 16 GT/s lanes: 16 ports: active: eDP-1 empty: DP-1, DP-2, 
DP-3, DP-4, DP-5, DP-6, DP-7,
     DP-8, Writeback-1 bus-ID: 06:00.0 chip-ID: 1002:1900 class-ID: 0300 
temp: 45.0 C
   Device-3: Chicony Integrated Camera driver: uvcvideo type: USB rev: 
2.0 speed: 480 Mb/s
     lanes: 1 bus-ID: 1-3:3 chip-ID: 04f2:b7b6 class-ID: fe01 serial: 
<filter>
   Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 
driver: X: loaded: amdgpu,nvidia
     unloaded: fbdev,modesetting,nouveau,vesa dri: radeonsi gpu: amdgpu 
display-ID: :0 screens: 1
   Screen-1: 0 s-res: 4096x2560 s-dpi: 120 s-size: 867x542mm (34.13x21.34")
     s-diag: 1022mm (40.25")
   Monitor-1: eDP-1 mapped: eDP model: BOE Display 0x0b38 serial: 
<filter> res: 4096x2560 dpi: 302
     size: 344x215mm (13.54x8.46") diag: 406mm (16") modes: max: 
2560x1600 min: 640x480
   API: EGL v: 1.5 hw: drv: nvidia drv: amd radeonsi platforms: device: 
0 drv: nvidia device: 2
     drv: radeonsi device: 3 drv: swrast gbm: drv: nvidia surfaceless: 
drv: nvidia x11: drv: radeonsi
     inactive: wayland,device-1
   API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: amd mesa v: 
24.2.8-1ubuntu1~24.04.1 glx-v: 1.4
     direct-render: yes renderer: AMD Radeon Graphics (radeonsi 
gfx1103_r1 LLVM 19.1.1 DRM 3.57
     6.8.0-52-generic) device-ID: 1002:1900
[... 97 lines cut ...]
            <br>
          </blockquote>
        </div>
        <blockquote type="cite">
          <div dir="ltr">
            <blockquote>
              <p dir="auto">Also, has anyone gotten &quot;Fn + Q&quot; to actually
                toggle the power profiles? My light will switch between
                blue and white but never to red for some reason, and
                when I boot into the BIOS I can see that the power
                profile is still stuck on &quot;Balanced&quot; regardless of the
                color of the light.</p>
            </blockquote>
            <p dir="auto"><a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="2291082365" data-permission-text="Title is private" data-url="https://github.com/johnfanv2/LenovoLegionLinux/issues/218" data-hovercard-type="issue" data-hovercard-url="/johnfanv2/LenovoLegionLinux/issues/218/hovercard?comment_id=2531317758&amp;comment_type=issue_comment" href="https://github.com/johnfanv2/LenovoLegionLinux/issues/218#issuecomment-2531317758" moz-do-not-send="true">#218 (comment)</a></p>
            <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
                unsubscribe</a>.<br>
                ID: <span>&lt;johnfanv2/LenovoLegionLinux/issues/218/2655048693</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
          </div>
        </blockquote>
      </div>
    </blockquote>
  </body>
</html>

--------------0dFiIo8Vij0lecd5x9KMo2cD--

Evthestrike, 2025-02-13:
> I’m getting the red light under the latest Linux Mint district.
> The Lenovo SLIM’s BIOS won’t allow the power setting on RED on battery power. It must be connected to AC power.

I thought I had tested that but I guess I did not. The light will now switch to red once I'm connected to AC power. After switching to red, I rebooted to the BIOS and the power mode was still balanced, so it seems like Fn + Q is not toggling the power modes. That information may have already been known.

demonindenim, 2025-05-05:
Hiya all :3
I've also got a Legion Slim 5 16AHP9. I would love to use this program, but I'm not sure what works so far and what doesn't.
I'm happy to help with any debugging. I'm on nvidia-driver-570, Linux Mint.
I just read through the chain; there seems to be a lot of technical stuff I'm not too familiar with, so I'm happy to read up any docs if anyone can point me in the right direction to help contributing!

codyit, 2025-08-17:
> Just wanted to update here that a lot of the sudden shutdown on suspend/charger plugin issues have been resolved by commit 5808c34216954cd832bd4b8bc52dfa287049122b on the Kernel, and this hasn't been an issue for a long time.
> 
> The power states seem to be toggling fine for me and I think this laptop is completely usable on Linux with only two minor issue. Gnome and GTK seem to have issues with suddenly waking up the GPU on Open/Exit of programs, but thats being discussed at https://gitlab.gnome.org/GNOME/gtk/-/issues/6689
> 
> The second issue which I could use some help reproducing is:
> 
> For some reason, the Nvidia graphics card is constantly woken up by SOME program, causing it to stay in a D0 state, instead of staying in D3Cold. `lsof /dev/nvidia0` and `lsof /dev/dri/card0` all pointed toward gnome and more specifically, XWayland using the GPU.
> 
> This issue occured across both KDE and Gnome. I then booted into recovery mode without a GUI and just a terminal, and surprisingly the issue still occured. So I have narrowed it down to something involving the Nvidia Driver.
> 
> `watch -n 1 cat /sys/class/drm/card0/device/power_state` should report the power state.
> 
> What's incredibly weird is that this issue seems to COMPLETELY go away once you put the laptop into suspend and wake it back up. It doesn't EVER come back for that particular boot. Could anyone else try to reproduce this? Thanks.

Yep, suspend issue was an annoyance but power consumption is the deal breaker for me right now. I suspect it's a regression because I used to be able to get ~7W idle with nvidia propitiatory and open drivers, with the config i have [above](https://github.com/johnfanv2/LenovoLegionLinux/issues/218#issuecomment-2529965361).  Now it's about 17W. Too late to isolate when, but I did a couple Fedora distro upgrade in between. `nvtop` shows it's `gnome-shell` using it, but it seems it's only using memory power with 0W on compute? Haven't done a structural test yet. I was about to look into the whole chain of gdm, gnome-shell, wayland and the driver since it's recurring problem for me over the years on different models so want a clear mental model of what's actually going, haven't found the bandwidth yet.

Mulukulum, 2025-08-17:
Maybe related to https://github.com/NVIDIA/open-gpu-kernel-modules/issues/730 and https://github.com/NVIDIA/open-gpu-kernel-modules/issues/905

imaGuru, 2025-08-28:
@codyit Is your dGPU able to powerdown straight from boot or only after system suspend? For me it was necessary to also add `NVreg_DynamicPowerManagement=0x02` and udev rules to enable power managment on driver bind (race condition?). With the default value of `0x03` after boot it entered a loop D0->D3cold(GC6)->D0->etc. Could you post the output of `cat /proc/driver/nvidia/params` or `systemd-analyze cat-config modprobe.d | grep nvidia`? Do you have ideapad_latop module loaded - `lsmod | grep ideapad`?

> I'll spend some time to confirm whether `NVreg_DynamicPowerManagementVideoMemoryThreshold` is necessary, and whether a 0 or high value is better for `NVreg_S0ixPowerManagementVideoMemoryThreshold` since a high value makes more sense, however when I set it to a high value I saw high power consumption that won't subside with nvidia chip staying in D3Hot, I have to try to reproduce this. If it's true it indicates there's some sort of bug in the memory management code path?

`NVreg_DynamicPowerManagementVideoMemoryThreshold=0` for me is definitely necessary because there seems to be some bug relating to turning off memory - basically with values higher than 0 the GPU can power down and turn off memory correctly only once and after it cycles D0->D3cold->D0->etc. ([issue](https://github.com/NVIDIA/open-gpu-kernel-modules/issues/905))
`NVreg_S0ixPowerManagementVideoMemoryThreshold` higher values in this parameter should only affect turning off memory during system suspend. 0 means memory is on in low power (self refresh state), higher values should turn off memory on suspend if memory consumption is below threshold

codyit, 2025-08-28:
> [@codyit](https://github.com/codyit) Is your dGPU able to powerdown straight from boot or only after system suspend? For me it was necessary to also add `NVreg_DynamicPowerManagement=0x02` and udev rules to enable power managment on driver bind (race condition?). Reddit link mentions this setting, but you do not? With the default value of `0x03` after boot it entered a loop D0->D3cold(GC6)->D0->etc. Could you post the output of `cat /proc/driver/nvidia/params` or `systemd-analyze cat-config modprobe.d | grep nvidia`? Do you have ideapad_laptop module loaded - `lsmod | grep ideapad`?
> 

I'm not with my laptop right now but both the config in the reddit post and mine work straight out from boot, and also post suspend. I'm not sure whether it's because of my gl_vendor workaround mentioned above. I can confirm I have ideapad loaded, I looked into it quite deeply and commented on the kernel mailing list PR for the new legion wmi drivers.

According to [nvidia's doc for 575.64](https://download.nvidia.com/XFree86/Linux-x86_64/575.64/README/dynamicpowermanagement.html) the default of 0x03 should be equivalent to 0x02 for this laptop which is post-ampere architecture. And no, I do not have to specify it.

I know you came from the Nvidia driver issue, i suppose toggling this memoery threshold should help isolate the bug.

> `NVreg_DynamicPowerManagementVideoMemoryThreshold=0` for me is definitely necessary because there seems to be some bug relating to turning off memory - basically with values higher than 0 the GPU [...]

So IIUC there are two suspend states here one being the card itself and another one for the laptop's ACPI? Then I suppose BIOS S3 config shouldnt matter then?

codyit, 2025-08-29:
> > According to [nvidia's doc for 575.64](https://download.nvidia.com/XFree86/Linux-x86_64/575.64/README/dynamicpowermanagement.html) the default of 0x03 should be equivalent to 0x02 for this laptop which is post-ampere architecture. And no, I do not have to specify it.
> 
> That is what I thought also, but in fact it is not equivalent. One observable difference is that NVreg_DynamicPowerManagement=0x02 starts driver with `/sys/bus/pci/devices/0000\:01\:00.0/power/control` set to 'on', where with 0x03 (default) it starts with 'auto', and thus requires udev rules to change it to 'auto'. 0x02 for my system gives working rtd3 but without the ability to turn off video memory (NVreg_DynamicPowerManagementVideoMemoryThreshold=0 disables turning off video memory)

Yeah I'm not sure why, I don't need any udev rule changes. I've detailed my setup above so you can replicate it, if it's missing anything let me know. What I notice is that when I start an application e.g. Firefox, About, calculator, with the watch gpu power script I do see the D3Cold > D0 > D3Cold transition, which completes in a few seconds. Let me know if you need more info, here's my nvidia params for completeness, don't think there's new info not inferred from the conifg:

```
ResmanDebugLevel: 4294967295
RmLogonRC: 1
ModifyDeviceFiles: 1
DeviceFileUID: 0
DeviceFileGID: 0
DeviceFileMode: 438
InitializeSystemMemoryAllocations: 1
UsePageAttributeTable: 4294967295
EnableMSI: 1
EnablePCIeGen3: 0
MemoryPoolSize: 0
KMallocHeapMaxSize: 0
VMallocHeapMaxSize: 0
IgnoreMMIOCheck: 0
EnableStreamMemOPs: 0
EnableUserNUMAManagement: 1
NvLinkDisable: 0
RmProfilingAdminOnly: 1
PreserveVideoMemoryAllocations: 1
EnableS0ixPowerManagement: 1
S0ixPowerManagementVideoMemoryThreshold: 0
DynamicPowerManagement: 3
DynamicPowerManagementVideoMemoryThreshold: 0
RegisterPCIDriver: 1
EnablePCIERelaxedOrderingMode: 0
EnableResizableBar: 0
EnableGpuFirmware: 18
EnableGpuFirmwareLogs: 2
RmNvlinkBandwidthLinkCount: 0
EnableDbgBreakpoint: 0
OpenRmEnableUnsupportedGpus: 1
DmaRemapPeerMmio: 1
ImexChannelCount: 2048
CreateImexChannel0: 0
GrdmaPciTopoCheckOverride: 0
RegistryDwords: ""
RegistryDwordsPerDevice: ""
RmMsg: ""
GpuBlacklist: ""
TemporaryFilePath: "/var/tmp"
ExcludedGpus: ""
```  

Here's my nvidia-smi output
```
ri Aug 29 09:55:19 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 575.64.05              Driver Version: 575.64.05      CUDA Version: 12.9     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070 ...    Off |   00000000:01:00.0 Off |                  N/A |
| N/A   32C    P8              2W /   50W |       2MiB /   8188MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```

awsms, 2025-12-12:
> Forgot to mention, regardless, it most likely won't be compatible with LLL frontend in its current form.

that's what I assumed too back then, so I forgot about it. feedback from the contributors of this project would be welcome, we still haven't heard from them on the issue :/

TheCodeFighter, 2025-12-18:
+1 for this support, I am interested on keyboard back light setting. (default is annoying).

mrbumpy409, 2026-02-14:
Just adding solutions for a couple of issues brought up in these comments:

### "Disable Touchpad While Typing" Not Working (@awsms)

By default, "disable touchpad while typing" does not work correctly as the Legion's keyboard is not properly detected as "internal". To fix this, create `/etc/libinput/local-overrides.quirks` containing the following:
```
[Lenovo Legion Slim 5 Keyboard Module]
MatchName=ITE Tech. Inc. ITE Device(8176) Keyboard
MatchUdevType=keyboard
AttrKeyboardIntegration=internal
```
### Control Keyboard Backlight Colors (@TheCodeFighter)

[LegionAura](https://github.com/nivedck/LegionAura) supports setting keyboard backlight color and brightness on the Legion Slim 5.