johnfanv2/LenovoLegionLinux #219: [BUG] Running `systemd_install.sh` failed

hacker-hackman, 2024-05-13:
**Problem Description**
Has the bug occurred following steps described in the README?
Yes.

What steps did you take?
I did every test from the README successfully and then installed LLL permanently.

What was the (full) error you received and what command or action did lead to it?
After I entered:
```
me@mint:~/LenovoLegionLinux/extra$ sudo ./systemd_install.sh
```
I got the message:
```
This script will Install the systemd service files
```
Then I hit enter and received the following output:
```
cp: cannot stat 'service/legion-linux.service': No such file or directory
cp: cannot stat 'service/legion-linux.path': No such file or directory
Invalid unit name "legion-linux.service,legion-linux.path" escaped as "legion-linux.service\x2clegion-linux.path" (maybe you should use systemd-escape?).
Failed to disable unit: Unit file legion-linux.service\x2clegion-linux.path does not exist.
Done
```

**Model and Debug Info**

Distribution: Linux Mint 21.3
Model name: Legion 5 15ACH6H
CPU model: AMD Ryzen 7 5800H
GPU model: NVIDIA RTX 3060
Keyboard backlight: single color with off/medium/bright
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82JU
	Version: Legion 5 15ACH6H
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_82JU_BU_idea_FM_Legion 5 15ACH6H
	Family: Legion 5 15ACH6H

Handle 0x0021, DMI type 12, 5 bytes
System Configuration Options
	Option 1: String1 for Type12 Equipment Manufacturer
	Option 2: String2 for Type12 Equipment Manufacturer
	Option 3: String3 for Type12 Equipment Manufacturer
	Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x0031, DMI type 32, 20 bytes
System Boot Information
	Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: GKCN50WW
	Release Date: 11/24/2021
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.50
	Firmware Revision: 1.50

Handle 0x0022, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 4
		en|US|iso8859-1
		fr|FR|iso8859-1
		ja|JP|unicode
		zh|TW|unicode
	Currently Installed Language: en|US|iso8859-1
```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```text
EC Chip ID: 8227
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 1
CPU temperature error: 0
CPU temperature: 40
CPU temperature EC error: 0
CPU temperature EC: 40
CPU temperature ACPI error: 0
CPU temperature ACPI: 40
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 40
CPU temperature WMI3 error: 0
CPU temperature WMI3: 0
GPU temperature error: 0
GPU temperature: 35
GPU temperature EC error: 0
GPU temperature EC: 35
GPU temperature ACPI error: 0
GPU temperature ACPI: 35
GPU temperature WMI error: 0
GPU temperature WMI: 0
GPU temperature WMI2 error: 0
GPU temperature WMI2: 35
GPU temperature WMI3 error: 0
GPU temperature WMI3: 0
fan speed access method: 1
1 fanspeed error: 0
1 fanspeed: 1730
1 fanspeed EC error: 0
1 fanspeed EC: 1730
1 fanspeed ACPI error: 0
1 fanspeed ACPI: 1700
1 fanspeed WMI error: 0
1 fanspeed WMI: 0
1 fanspeed WMI2 error: 0
1 fanspeed WMI2: 1700
1 fanspeed WMI3 error: 0
1 fanspeed WMI3: 0
2 fanspeed error: 0
2 fanspeed: 1916
2 fanspeed EC error: 0
2 fanspeed EC: 1916
2 fanspeed ACPI error: 0
2 fanspeed ACPI: 1900
2 fanspeed WMI error: 0
2 fanspeed WMI: 0
2 fanspeed WMI2 error: 0
2 fanspeed WMI2: 1900
2 fanspeed WMI3 error: 0
2 fanspeed WMI3: 0
powermode access method: 3
powermode error: 0
powermode: 3
powermode EC error: 0
powermode EC: 1
powermode ACPI error: -5
powermode ACPI: 70500864
powermode WMI error: 0
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: 2
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 1
EC minifancurve on cool: true
EC lockfancontroller error: 0
EC lockfancontroller: false
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: 0
fanfullspeed EC: 0
EC fan curve current point id: 1
EC fan curve points size: 10
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
0	 0	 2	 2	 0	 48	 0	 60	 0	 42
1700	 1900	 2	 2	 45	 54	 56	 60	 39	 44
1900	 2000	 2	 2	 51	 58	 56	 60	 42	 50
2200	 2100	 2	 2	 55	 62	 56	 60	 46	 127
2300	 2400	 2	 2	 59	 71	 56	 60	 127	 127
2600	 2700	 2	 2	 68	 76	 56	 65	 127	 127
2900	 3000	 2	 2	 72	 81	 61	 69	 127	 127
3500	 3500	 2	 2	 78	 90	 66	 77	 127	 127
3800	 3800	 2	 2	 87	 95	 72	 79	 127	 127
4500	 4500	 2	 2	 91	 127	 78	 127	 127	 127
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

Additional info:
At every boot I get these errors before and after installation of LLL:
```
[    0.359309] ACPI BIOS Error (bug): Could not resolve symbol [\ _SB.PCI0.PB2], AE_NOT_FOUND (20210730/dswload2-162)
[    0.359328] ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20210730/psobject-220)
[    1.001684] integrity: Problem loading X.509 certificate -65
```
These errors pop up right after I installed Linux Mint. Moreover they popped out even when the Linux Mint live USB drive was booting for the first time.

Legion2016, 2024-06-07:
I have the same issue using nobara
Legion slim 7 16ARHA7 82ug

I want to bring gpu to 90 watts as it does in windows.  But on Linux I'm stuck at 65 watts, as if I was running on battery.  I get same benchmark score on ac and battery same power consumption.

tmuird, 2024-11-17:
Interestingly it looks like the files referenced in the command
`sudo cp service/{legion-linux.service,legion-linux.path} /etc/systemd/system/`
Do not actually exist?

Instead these files reside in the referenced folder

`-rw-r--r--. 1 tom tom 216 Nov 17 12:49 legiond-cpuset.initd
-rw-r--r--. 1 tom tom  96 Nov 17 12:49 legiond-cpuset.service
-rw-r--r--. 1 tom tom  77 Nov 17 12:49 legiond-cpuset.sh
-rw-r--r--. 1 tom tom 147 Nov 17 12:49 legiond-cpuset.timer
-rw-r--r--. 1 tom tom 554 Nov 17 12:49 legiond.ini
-rw-r--r--. 1 tom tom 366 Nov 17 12:49 legiond.initd
-rw-r--r--. 1 tom tom 235 Nov 17 12:49 legiond-onresume.service
-rw-r--r--. 1 tom tom  86 Nov 17 12:49 legiond-onresume.sh
-rw-r--r--. 1 tom tom 192 Nov 17 12:49 legiond.service
drwxr-xr-x. 1 tom tom 344 Nov 17 12:49 profiles
`

In commit e07c6919c9ab029e06e28dbb76960e2b34923a86 legion-linux.path and legion-linux.service are both removed for a migration to legiond.service, but it appears the installscript has not been updated since this change in march

closed 2026-09-12 by github-actions[bot]