johnfanv2/LenovoLegionLinux #221: Legion Pro 7 16IRX8H 82WQ audio not working on debian

darkpowerxo, 2024-05-26:
**Is your feature request related to a problem? Legion Pro 7 16IRX8H 82WQ audio not working on debian. 
should I use pipewire or pulse audio.  

**Describe the solution you'd like**
hear audio 

**Describe alternatives you've considered**
have tried putting firmware-TAS.zip from here in /usr/lib/firmware and rebooting
https://forums.lenovo.com/t5/Ubuntu/Ubuntu-and-legion-pro-7-16IRX8H-audio-issues/m-p/5210709?page=17#6119839

also have switched between pulse audio and pipewire and not not sure which is the right one
i see the microphone and speaker being detected, and when i play audio, in the system settings i see the volume moving but i hear nothing out of the speakers

**Model and Debug Info**

Distribution: Debian  kernel 6.6.15-amd64
KDE Plasma Version: 5.27.10
Model name: lenovo (laptop)
CPU model: i9 13900HX
GPU model: RTX 4090
Keyboard backlight: Yes (RGB)
Light in lid or logo: lid
Light at IO-Ports at back: yes

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:

SMBIOS 3.4 present.
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: LENOVO
	Product Name: 82WQ
	Version: Legion Pro 7 16IRX8H
Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
	Option 1: ConfigOptions1
	Option 2: ConfigOptions2
	Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
	Area Length: 0 bytes
	Header Start Offset: 0x0000
	Header Length: 8192 bytes
	Data Start Offset: 0x2000
	Access Method: General-purpose non-volatile data functions
	Access Address: 0x0000
	Status: Valid, Not Full
	Change Token: 0x12345678
	Header Format: OEM-specific
	Supported Log Type Descriptors: 3
	Descriptor 1: POST memory resize
	Data Format 1: None
	Descriptor 2: POST error
	Data Format 2: POST results bitmap
	Descriptor 3: Log area reset/cleared
	Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Output of `sudo dmidecode -t bios`:
Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
	Vendor: LENOVO
	Version: KWCN44WW
	Release Date: 01/15/2024
	Address: 0xE0000
	Runtime Size: 128 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
		5.25"/360 kB floppy services are supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		8042 keyboard services are supported (int 9h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 1.44
	Firmware Revision: 1.44

Handle 0x0025, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 8
		en|US|iso8859-1,0
		fr|FR|iso8859-1,0
		zh|TW|unicode,0
		ja|JP|unicode,0
		it|IT|iso8859-1,0
		es|ES|iso8859-1,0
		de|DE|iso8859-1,0
		pt|PT|iso8859-1,0
	Currently Installed Language: en|US|iso8859-1,0

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
file not found

darkpowerxo, 2024-06-17:
Still no solution. 
plugging a headphone works but not the speakers.

tomasvalek, 2024-06-17:
@darkpowerxo I had the same problem. Switch to Kernel 6.7.9 helped me (Ubuntu 24.04). Both speakers are working now. For info: kernel 6.9.3 did not work too.

prattrs, 2025-07-11:
I had an issue with audio on this model, and I was able to solve it by setting a flag to force load audio firmware in alsa.state.

Here are my notes:

The Legion (and prob lots of other laptops) have a thing where the speaker firmware needs to be force pushed frequently, otherwise the speakers are silent.

After much toil, my answer seems to be this:

https://bbs.archlinux.org/viewtopic.php?id=172940

make sure the alsa-state.service is enabled and running and the /etc/alsa/state-daemon.conf file exists -- https://wiki.archlinux.org/title/Advanced_Linux_Sound_Architecture#ALSA_and_systemd
edit the saved state in /var/lib/alsa/asound.state (with vim or whatever) -- find the part for the card (prob snd_hda_intel or sth) and find the line -- name 'Speaker Force Firmware Load' -- toggle the flag to true
alsactl restore

---

tldr this is making sure that the proper alsa-state service is running (there are two alsa state services, so you need to pick), find the saved state, edit a false to true with a text editor in some file under /var/lib/alsa

closed 2026-09-12 by github-actions[bot]