johnfanv2/LenovoLegionLinux #225: [FEATURE REQUEST] Support `rpm-ostree`

Malix-off, 2024-06-04:
## Description

Support `rpm-ostree`

`rpm-ostree` is the `ostree` version of `dnf` (it's the package manager used on Fedora Atomic)
Which technically also accepts COPR, but has no instructions as of now.
Also, after having installed the COPR repo, it's unclear which package to install

```log
malix@malix-pc ~> rpm-ostree search LenovoLegionLinux

===== Summary & Name Matched =====
dkms-LenovoLegionLinux : LenovoLegionLinux Kernel Module Package
python-LenovoLegionLinux-debuginfo : Debug information for package python-LenovoLegionLinux
python-LenovoLegionLinux-debugsource : Debug sources for package python-LenovoLegionLinux

===== Name Matched =====
dkms-LenovoLegionLinux : LenovoLegionLinux Kernel Module Package
python-LenovoLegionLinux : Control Lenovo Legion laptop
python-LenovoLegionLinux-debuginfo : Debug information for package python-LenovoLegionLinux
python-LenovoLegionLinux-debugsource : Debug sources for package python-LenovoLegionLinux

===== Summary Matched =====
dkms-LenovoLegionLinux : LenovoLegionLinux Kernel Module Package
python-LenovoLegionLinux-debuginfo : Debug information for package python-LenovoLegionLinux
python-LenovoLegionLinux-debugsource : Debug sources for package python-LenovoLegionLinux
```

elirael, 2024-06-07:
@Malix-off I'd tried installing LLL on silverblue some time ago myself. Unfortunately, I never managed to get it to work. Dkms simply isn't compatible with immutable distros, and I'm too much of a noob to try getting akmod running. Would be nice though.

Malix-off, 2024-06-07:
Thanks for your experience report @elirael

See https://github.com/ublue-os/hwe/issues/253

RodrigoPrestes, 2024-11-08:
Just tried installing on a Silverblue 40 VM before erasing the laptop itself, it installed no problem but app doesn't do anything (which is expected inside the VM).

I installed by:
- Downloading the repo file on https://copr.fedorainfracloud.org/coprs/mrduarte/LenovoLegionLinux/, in my case was the `Fedora 40` button;
- Moving the file to `/etc/yum.repos.d/LenovoLegionLinux.repo`;
- Installing with `rpm-ostree install dkms-LenovoLegionLinux python-LenovoLegionLinux`;
- Reboot.

Looking at the logs it doesn't show errors, but again, don't know if it actually works.
Seems like [rpm-ostree still doesn't have support for dkms](https://github.com/coreos/rpm-ostree/issues/1091) so probably it doesn't work.

Malix-Labs, 2024-11-08:
Fedora 41 switched to `dnf5` for layering, `rpm-ostree` will probably be deprecated in a few

closed 2024-11-08 by Malix-Labs