johnfanv2/LenovoLegionLinux #234: [BUG] Can't read or set the fancurve for Lenovo Legion 7 Pro 16ARX8H

Syderitic, 2024-06-24:
**Problem Description**
Hello, I can't seem to set or read the fan curve of my laptop. Can you help?
I also posted here #133 
[image]
All the hardware works on my laptop (sound,graphics,etc) 

Distribution: Gentoo
Kernel: 6.6.33-gentoo-dist-hardened AuthenticAMD GNU/Linux (uname -roi)
Model name: Lenovo Legion 7 Pro 16ARX8H
CPU model: AMD 7945HX
GPU model: NVIDIA 4090
Keyboard backlight: RGB
Light in lid or logo: no
Light at IO-Ports at back: yes
```
System Information
        Manufacturer: LENOVO
        Product Name: 82WS
        Version: Legion Pro 7 16ARX8H
        Serial Number: 
        UUID: 
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WS_BU_idea_FM_Legion Pro 7 16ARX8H
        Family: Legion Pro 7 16ARX8H
```

```
BIOS Information
        Vendor: LENOVO
        Version: LPCN45WW
        Release Date: 08/02/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Function key-initiated network boot is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.45
        Firmware Revision: 1.45
```

```
EC Chip ID: 5507
EC Chip Version: 2a4
legion_laptop features: fancurve powermode platformprofile platformprofilenotify minifancurve
legion_laptop ec_readonly: 0
ACPI CFG error: 0
ACPI CFG: 2081289482
temperature access method: 5
CPU temperature error: 0
CPU temperature: 60
CPU temperature EC error: 0
CPU temperature EC: 0
CPU temperature ACPI error: 0
CPU temperature ACPI: 60
CPU temperature WMI error: 0
CPU temperature WMI: 0
CPU temperature WMI2 error: 0
CPU temperature WMI2: 0
[... 138 lines cut ...]
000003e0  ff 00 ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000003f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000400  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000480  40 eb bf 4c 1e 9b 11 b2  a8 5c 99 9c bb 3f ff dc  |@..L.....\...?..|
00000490  ff fb ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
000004a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000500  00 01 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000510  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000530  b0 00 b0 00 00 00 00 00  00 00 00 00 00 00 01 00  |................|
00000540  00 00 00 00 00 01 00 01  01 00 00 01 00 00 00 00  |................|
00000550  00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00  |................|
00000560  00 00 00 00 00 00 00 00  00 00 00 00 00 40 00 00  |.............@..|
00000570  80 00 1e 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000580  00 01 01 00 01 01 00 00  80 00 00 00 00 00 ff 00  |................|
00000590  00 00 00 00 00 00 00 00  00 00 00 00 28 00 00 00  |............(...|
000005a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000005b0  00 00 00 00 00 00 00 00  00 00 00 00 05 00 00 00  |................|
000005c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000005f0  00 00 00 00 00 00 00 00  00 00 00 10 00 03 00 00  |................|
00000600
```

ComputerCrack, 2025-02-26:
Request to change the title.
The issue is also present on the Lenovo Legion Pro 7 16IRX8H model.

All hardware seems to be working. I can see the fan speed with the GNOME Vitals plugin, but not the fan curve. The error messages and outputs are similar to those provided above. I installed both the Git and the DKMS versions via the AUR. Based on my research, most, if not all, Pro 16 models are affected. Changing fan speed with CoolerControl (or any other tool) won't work **Fan 1** and **Fan 2** are marked as "not supported". 
_"The currently installed driver does not support control of this channel"_

Please let me know if further information is needed.

```
Distribution: I use Arch Linux by the way
Kernel: Linux 6.13.4-arch1-1
Model name: Lenovo Legion 7 Pro 16IRX8H
CPU model: 13th Gen Intel® Core™ i9-13900HX × 32
GPU model: NVIDIA GeForce RTX 4090
Keyboard backlight: RGB
Light in lid or logo: no
Light at IO-Ports at back: yes
```

**sudo dmidecode -t system**
```

# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82WQ
        Version: Legion Pro 7 16IRX8H
        Serial Number: PF4K5ANR
        UUID: 124ff651-da6f-4201-853d-745d2265d932
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82WQ_BU_idea_FM_Legion Pro 7 16IRX8H
        Family: Legion Pro 7 16IRX8H

Handle 0x0024, DMI type 12, 5 bytes
System Configuration Options
        Option 1: ConfigOptions1
        Option 2: ConfigOptions2
        Option 3: ConfigOptions3

Handle 0x0027, DMI type 15, 29 bytes
System Event Log
        Area Length: 0 bytes
        Header Start Offset: 0x0000
        Header Length: 8192 bytes
        Data Start Offset: 0x2000
        Access Method: General-purpose non-volatile data functions
        Access Address: 0x0000
        Status: Valid, Not Full
        Change Token: 0x12345678
        Header Format: OEM-specific
        Supported Log Type Descriptors: 3
        Descriptor 1: POST memory resize
        Data Format 1: None
        Descriptor 2: POST error
        Data Format 2: POST results bitmap
        Descriptor 3: Log area reset/cleared
        Data Format 3: None

Handle 0x0034, DMI type 32, 11 bytes
System Boot Information
        Status: No errors detected
```

**sudo dmidecode -t bios**
```

[... 1630 lines cut ...]
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
=====================
Current fan curve in hardware (WMI; might be empty)
Fan curve current point id: 0
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
1        0       0       0       0       0       0       0       0       0       0       0       0
=====================
````

Gert-dev, 2025-06-11:
I ran FWTS on this device, the results can be found below.

In the source code I see that `WMI_GUID_LENOVO_FAN_METHOD` equals `92549549-4bde-4f06-ac04-ce8bf898dbaa`, which is **WMB2** instead on this device, and indeed:

```
╰─ echo '\_SB_.GZFD.WMB2 0 0x05 0x0002' | sudo tee /proc/acpi/call; sudo cat /proc/acpi/call
\_SB_.GZFD.WMB2 0 0x05 0x0002
{0x0a, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
```

This looks like the fan curve is all zeroes. I also tried querying some of the other WMI methods using the same 0x05 value, but unfortunately none of these values appear to change with setting the power mode either.

```
╰─ echo '\_SB_.WMI4.WQBD 0 0x05 0x0002' | sudo tee /proc/acpi/call; sudo cat /proc/acpi/call
\_SB_.WMI4.WQBD 0 0x05 0x0002
{0x46, 0x4f, 0x4d, 0x42, 0x01, 0x00, 0x00, 0x00, 0x65, 0x02, 0x00, 0x00, 0xf8, 0x05, 0x00, 0x00, 0x44, 0x53, 0x00, 0x01, 0x1a, 0x7d, 0xda, 0x54, 0x18, 0xd1, 0x82, 0x00, 0x01, 0x06, 0x18, 0x42, 0x10, 0x05, 0x10, 0x8a, 0x0d, 0x21, 0x02, 0x0b, 0x83, 0x50,⏎  

╰─ echo '\_SB_.WMI4.WQAD 0 0x05 0x0002' | sudo tee /proc/acpi/call; sudo cat /proc/acpi/call
\_SB_.WMI4.WQAD 0 0x05 0x0002
"BAT0 BatMaker,0x5"⏎  

╰─ echo '\_SB_.GZFD.WMB3 0 0x05 0x0000' | sudo tee /proc/acpi/call; sudo cat /proc/acpi/call
\_SB_.GZFD.WMB3 0 0x05 0x0000
0x0⏎     
```

## FWTS Output

```
Results generated by fwts: Version V25.05.00 (2025-05-27 02:18:07).

Some of this work - Copyright (c) 1999 - 2025, Intel Corp. All rights reserved.
Some of this work - Copyright (c) 2010 - 2025, Canonical.
Some of this work - Copyright (c) 2016 - 2025, IBM.
Some of this work - Copyright (c) 2017 - 2025, ARM Ltd.

This test run on 10/06/25 at 21:48:49 on host Linux hephaestus
6.15.1-zen1-2.1-zen #1 ZEN SMP PREEMPT_DYNAMIC Tue, 10 Jun 2025 09:05:23 +0000
x86_64.

Command: "fwts wmi -".
Running tests: wmi.

wmi: Extract and analyse Windows Management Instrumentation (WMI).
--------------------------------------------------------------------------------
ACPICA Exception AE_NOT_FOUND during execution of method \___
Test 1 of 1: Windows Management Instrumentation test.

\_SB_.WMI4._WDG (1 of 2)
  GUID: C3A03776-51AC-49AA-AD0F-F2F7D62C3F3C
  WMI Block:
    Flags          : 0x05 (Expensive | String)
    Object ID      : AD
    Instance       : 0x03
PASSED: Test 1, C3A03776-51AC-49AA-AD0F-F2F7D62C3F3C has associated query method
\_SB_.WMI4.WQAD
PASSED: Test 1, C3A03776-51AC-49AA-AD0F-F2F7D62C3F3C has more than zero
instances

\_SB_.WMI4._WDG (2 of 2)
  GUID: 05901221-D566-11D1-B2F0-00A0C9062910
  WMI Block:
    Flags          : 0x00 (None)
    Object ID      : BD
    Instance       : 0x01
    Driver         : wmi-bmof (Generic)
PASSED: Test 1, 05901221-D566-11D1-B2F0-00A0C9062910 has associated query method
\_SB_.WMI4.WQBD
PASSED: Test 1, 05901221-D566-11D1-B2F0-00A0C9062910 has more than zero
instances
[... 469 lines cut ...]
wmi: GUID D320289E-8FEA-41E0-86F9-611D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID D320289E-8FEA-41E1-86F9-611D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID D320289E-8FEA-41E0-86F9-711D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID D320289E-8FEA-41E0-86F9-811D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID D320289E-8FEA-41E0-86F9-911D83151B5F is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID 1E3391A1-2C89-464D-95D9-3028B72E7A33 is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID BE1346E5-3A32-4A31-B8B0-2CD195E1F572 is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID 67A6C6CC-380A-4D25-AA35-A1397D1FBDFB is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID 0F7B30EC-AC90-4B34-B62F-87F85115505D is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID 8FC0DE0C-B4E4-43FD-B0F3-8871711C1294 is unknown to the kernel, a driver may need to be implemented for this GUID.
wmi: GUID ABBC0F6A-8EA1-11D1-00A0-C90629100000 has likely been copied from the Windows driver samples, this is a firmware bug that prevents WMI drivers from reliably detecting such WMI devices since their GUID is not unique.

Low failures: 2
wmi: GUID B7F3CA0A-ACDC-42D2-9217-77C6C628FBD2 should have an associated query method WQA0 defined, however this does not seem to exist.
wmi: GUID ECAF3CE6-DCB2-473D-94C3-7BF4D96E0D9A should have an associated query method WQA5 defined, however this does not seem to exist.

Other failures: NONE

Test           |Pass |Fail |Abort|Warn |Skip |Info |
---------------+-----+-----+-----+-----+-----+-----+
wmi            |   61|   18|     |     |     |     |
---------------+-----+-----+-----+-----+-----+-----+
Total:         |   61|   18|    0|    0|    0|    0|
---------------+-----+-----+-----+-----+-----+-----+
```

Gert-dev, 2025-06-11:
> The LenovoLegionToolkit README suggests that some Lenovo models [have a fourth cooling mode](https://github.com/BartoszCichecki/LenovoLegionToolkit?tab=readme-ov-file#custom-mode) that allows adjusting the fans. Hence maybe why they are all zeroes; maybe this just needs to be set to the appropriate values and only takes effect when this fourth mode is activated (i.e. it doesn't contain the active fan curve, but only the custom fan curve)?

Looks indeed like this is the case. This activates the fourth cooling mode and makes the LED go purple:

```
echo '255' | sudo tee /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/powermode
```

Doing so makes the fancurve get populated from all zeroes to:

```
Current fan curve in hardware:
Fan curve current point id: 0
Fan curve points size: 10
u(speed_of_unit)|speed1[u]|speed2[u]|speed1[pwm]|speed2[pwm]|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|i
c_min_temp|ic_max_temp
1        1       0       2       2       0       0       0       0       0       0       0       0
1        2       0       5       5       0       0       0       0       0       0       0       0
1        3       0       7       7       0       0       0       0       0       0       0       0
1        4       0       10      10      0       0       0       0       0       0       0       0
1        5       0       12      12      0       0       0       0       0       0       0       0
1        6       0       15      15      0       0       0       0       0       0       0       0
1        7       0       17      17      0       0       0       0       0       0       0       0
1        8       0       20      20      0       0       0       0       0       0       0       0
1        8       0       20      20      0       0       0       0       0       0       0       0
1        8       0       20      20      0       0       0       0       0       0       0       0
```

However, when the purple LED is active, everything starts lagging for me. Doesn't appear to be thermal throttling but at some point the laptop shut itself down, perhaps a value that is incorrectly being read or written somewhere.

Not quite a custom fancurve yet, but progress I guess.

Gert-dev, 2025-06-11:
I _seem_ to be able to set fancurves now on this model, as test to set all fans to max. If you activate custom cooling mode as per the above, the fancurve populates, I can then update the curve:

```
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point2_pwm
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point3_pwm
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point4_pwm
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point5_pwm
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point6_pwm
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point7_pwm
```

I have points 8 and 9 as well, but they don't appear to set anything in the curve - perhaps this model only has 8 points instead of 10. 

Setting point 1 also doesn't change anything, perhaps some sort of safety measurement or also something that's different.

Setting PWM1 also changes both speed1 and speed2 in the curve, so perhaps these fans can only be set in tandem.

prattrs, 2025-07-11:
I found this earlier today (thanks!) and adapted it to this model (Legion Pro 7 16IRX8H). The changes were minimal. After running the script, seeing the light turn pink, hearing the fans rev up to about 3500 after a few seconds delay, I ran some games to test (Overwatch2, Stellaris). No crashes so far.

However, it eventually set the fan curve back to default, and setting it repeatedly using the script in this manner does nothing. Possibly after reboot it will start changing the fan speeds again.

```
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point2_pwm  # >=  0º
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point3_pwm  # >=  60º/50º, < 50º/40º
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point4_pwm  # >=  62/52 º
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point5_pwm  # >=  66/58 º
echo 22 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point6_pwm  # >=  68/60º
echo 24 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point7_pwm  # >= ~75/??º
echo 24 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point8_pwm  # >= ~80/??º
echo 26 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point9_pwm  # >= ? º
echo 26 > /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/hwmon/hwmon*/pwm1_auto_point10_pwm # >= ? º

echo '255' | sudo tee /sys/module/legion_laptop/drivers/platform:legion/PNP0C09:00/powermode
```

Maybe in the future I can figure out how to keep it going longer.

closed 2026-09-12 by github-actions[bot]