johnfanv2/LenovoLegionLinux #240: [BUG] Fan Curves Not Reset After Uninstalling AND Restarting

DillonHHH, 2024-07-12:
**Problem Description**
I think this service has permanently changed the fan curves on my system and I can't figure out how to reset them to default. I have fully uninstalled the service with make uninstall, and restarted multiple times, but my fans are stuck at a low speed and my laptop is thermal throttling.

**Model and Debug Info**

Distribution: Fedora 40
Model name:Legion 7 16ITHg6
CPU model: i9-11980HK
GPU model: RTX 3080

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```        
Manufacturer: 
Product Name: 
Version: Legion 7 
Wake-up Type: Power 
SKU Number: LENOVO_MT_82K6_BU_idea_FM_Legion 7 
Family: Legion 7 16ITHg6

```

Output of `sudo dmidecode -t bios`:
```
Vendor: LENOVO
        Version: H1CN53WW
        Release Date: 03/16/2023
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                8042 keyboard services are supported (int 9h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.53
        Firmware Revision: 1.53

```

Output of `sudo cat /sys/kernel/debug/legion/fancurve`:
```
No such file or directory
```

johnfanv2, 2024-07-13:
You might have locked the curve and you could unlock it (see FAQ of README). Alternatively, just reset the embedded controller by long pressing the power button (see FAQ of README). 

Please report back if you could fix the problem. This should never happen. Could you please give the output of `sudo cat /sys/kernel/debug/legion/fancurve` when the kernel module is installed (or temporarily loaded) and you have not changed/set anything?

DillonHHH, 2024-07-13:
@johnfanv2 Thank you! The long power button reset worked great. After restarting, here is the output:
`sudo cat /sys/kernel/debug/legion/fancurve
cat: /sys/kernel/debug/legion/fancurve: No such file or directory`

closed 2024-07-13 by DillonHHH