johnfanv2/LenovoLegionLinux #248: [BUG] 0.0.18 copr build failing

ryanabx, 2024-09-02:
**Problem Description**

The official COPR repository hasn't had a successful build of `dkms-LenovoLegionLinux` since `0.0.16` 3 months ago:

[image]

**Hardware info not applicable**

ralmachado, 2024-10-22:
Bumping this, as the last F40 build was on 22nd Aug, and commit 61f05e082afe2af9fec3d038e8c800192b2de07f patches the dkms module to be compatible with kernel 6.11.*.
I'm having an issue with resuming from sleep using 6.11, and would like to eliminate this project as the culprit.
@MrDuartePT can a new build be submitted to the COPR please?

MrDuartePT, 2024-10-28:
> Bumping this, as the last F40 build was on 22nd Aug, and commit [61f05e0](https://github.com/johnfanv2/LenovoLegionLinux/commit/61f05e082afe2af9fec3d038e8c800192b2de07f) patches the dkms module to be compatible with kernel 6.11.*. I'm having an issue with resuming from sleep using 6.11, and would like to eliminate this project as the culprit. @MrDuartePT can a new build be submitted to the COPR please?

@ryanabx should be fix now, also I don't think that our driver is causing you problems, since the only places the build fails is on fedora 41 and rawhide (in case of rawhide I don't really know how to fix it).

ryanabx, 2024-10-29:
@MrDuartePT Installing now gives me an issue:

```
RUN dnf copr enable -y mrduarte/LenovoLegionLinux && dnf install -y dkms-LenovoLegionLinux python-LenovoLegionLinux python-darkdetect
 https://copr.fedorainfracloud.org/api_ 100% |   1.9 KiB/s | 524.0   B |  00m00sEnabling a Copr repository. Please note that this repository is not part
of the main distribution, and quality may vary.

The Fedora Project does not exercise any power over the contents of
this repository beyond the rules outlined in the Copr FAQ at
<https://docs.pagure.org/copr.copr/user_documentation.html#what-i-can-build-in-copr>,
and packages are not held to any quality or security level.

Please do not file bug reports about these packages in Fedora
Bugzilla. In case of problems, contact the owner of this repository.
Updating and loading repositories:
 Copr repo for LenovoLegionLinux owned  100% |  17.7 KiB/s |   5.5 KiB |  00m00s
 Fedora 41 - x86_64 - Updates Archive   100% |   6.7 KiB/s |   4.3 KiB |  00m01s
>>> Status code: 403 for https://fedoraproject-updates-archive.fedoraproject.org/
>>> Status code: 403 for https://fedoraproject-updates-archive.fedoraproject.org/
>>> Status code: 403 for https://fedoraproject-updates-archive.fedoraproject.org/
>>> Status code: 403 for https://fedoraproject-updates-archive.fedoraproject.org/
>>> Librepo error: Yum repo downloading error: Downloading error(s): repodata/4a
Repositories loaded.
Failed to resolve the transaction:
Problem 1: conflicting requests
  - nothing provides python3.13dist(darkdetect) needed by python-LenovoLegionLinux-0.0.18-0.x86_64 from copr:copr.fedorainfracloud.org:mrduarte:LenovoLegionLinux
 Problem 2: conflicting requests
  - nothing provides python(abi) = 3.12 needed by python-darkdetect-0.8.0-1.noarch from copr:copr.fedorainfracloud.org:mrduarte:LenovoLegionLinux
```

MrDuartePT, 2024-10-30:
> @MrDuartePT Installing now gives me an issue:
>
>
>
> ```
>
> RUN dnf copr enable -y mrduarte/LenovoLegionLinux && dnf install -y dkms-LenovoLegionLinux python-LenovoLegionLinux python-darkdetect [...]

The error is because you probably not using Python 3.12 try to update your system and then install try to install darkdetect. I try on a Fedora 41 livecd and was working

Allephh, 2024-11-01:
I have a similar problem to this one, in this case, it is related to 'python(abi)=3.12'.
I use Lenovo IdeaPad Gaming 3 15ARH7, Fedora 41 and Bios JNCN51WW(V2.11).
I started using Linux in Fedora 39 and the LenovoLegionLinux driver always worked. However, in 41, it is not working anymore and i suggest that it must be a problem with Python (which is now 3.13 in f41) or DNF5 (at the end of the installation, it says commands no longer exist or something like that).
In previous installations, i was having the same two problems (problem 1 and 2), but now only one.
Every time i installed this driver it ended up not working, it even gave some errors on the screen when turning it off and on (when pressing alt + tab), however, it did not prevent the system overall for working. Unfortunately, i didn't take a photo and i don't want to install it again because the system is already prepared (only missing the lenovo and nvidia drivers).

`sudo dnf install dkms-LenovoLegionLinux python-darkdetect python-LenovoLegionLinux
[sudo] senha para alexandre: 
Updating and loading repositories:
Repositories loaded.
Problema: não é possível instalar o melhor candidato para a tarefa
  - nothing provides python(abi) = 3.12 needed by python-darkdetect-0.8.0-1.noarch from copr:copr.fedorainfracloud.org:mrduarte:LenovoLegionLinux

Package                                        Arch   Version                                        Repository                                           Size
Installing:
 dkms-LenovoLegionLinux                        x86_64 0.0.18-0                                       copr:copr.fedorainfracloud.org:mrduarte:LenovoL 312.7 KiB
 python-LenovoLegionLinux                      x86_64 0.0.18-0                                       copr:copr.fedorainfracloud.org:mrduarte:LenovoL 499.6 KiB
 python-darkdetect                             noarch 0.8.0-1                                        copr:copr.fedorainfracloud.org:mrduarte:LenovoL  36.4 KiB
Installing dependencies:
 SDL2                                          x86_64 2.30.3-2.fc41                                  fedora                                            1.8 MiB
 bison                                         x86_64 3.8.2-9.fc41                                   fedora                                            3.5 MiB
 cpp                                           x86_64 14.2.1-3.fc41                                  fedora                                           35.0 MiB
 dkms                                          noarch 3.1.1-1.fc41                                   updates                                         181.7 KiB
 elfutils-libelf-devel                         x86_64 0.192-5.fc41                                   updates                                          49.5 KiB
 flex                                          x86_64 2.6.4-18.fc41                                  fedora                                          805.4 KiB
 gcc                                           x86_64 14.2.1-3.fc41                                  fedora                                          104.3 MiB
 glibc-devel                                   x86_64 2.40-3.fc41                                    fedora                                           35.0 KiB
 glibc-headers-x86                             noarch 2.40-3.fc41                                    fedora                                            2.2 MiB
 kernel-devel                                  x86_64 6.11.5-300.fc41                                updates                                          75.1 MiB
 kernel-devel-matched                          x86_64 6.11.5-300.fc41                                updates                                           0.0   B
 kernel-headers                                x86_64 6.11.3-300.fc41                                fedora                                            6.4 MiB
 libmpc                                        x86_64 1.3.1-6.fc41                                   fedora                                          164.7 KiB
 libxcrypt-devel                               x86_64 4.4.36-7.fc41                                  fedora                                           30.3 KiB
 libzstd-devel                                 x86_64 1.5.6-2.fc41                                   fedora                                          202.4 KiB
 m4                                            x86_64 1.4.19-10.fc41                                 fedora                                          592.9 KiB
 make                                          x86_64 1:4.4.1-8.fc41                                 fedora                                            1.8 MiB
 openssl-devel                                 x86_64 1:3.2.2-9.fc41                                 fedora                                            4.3 MiB
 python-pyqt6-rpm-macros                       noarch 6.8.0-0.1.fc41                                 updates                                         144.0   B
 python3-pyqt6                                 x86_64 6.8.0-0.1.fc41                                 updates                                           5.7 MiB
 python3-pyqt6-base                            x86_64 6.8.0-0.1.fc41                                 updates                                          15.4 MiB
 python3-pyqt6-sip                             x86_64 13.8.0-1.fc41                                  fedora                                          146.1 KiB
[... cut ...]
 qt6-qtspeech-speechd                          x86_64 6.8.0-1.fc41                                   updates                                          67.6 KiB
 qt6-qttranslations                            noarch 6.8.0-1.fc41                                   updates                                          13.7 MiB

Transaction Summary:
 Installing:        56 packages

Total size of inbound packages is 127 MiB. Need to download 127 MiB.
After this operation, 406 MiB extra will be used (install 406 MiB, remove 0 B).
Is this ok [y/N]: n
Operação abortada pelo usuário.
`

johnfanv2, 2024-11-01:
> I have a similar problem to this one, in this case, it is related to 'python(abi)=3.12'. I use Lenovo IdeaPad Gaming 3 15ARH7, Fedora 41 and Bios JNCN51WW(V2.11). I started using Linux in Fedora 39 [...]

Do I get this correctly that the system crashes and you cannot use it anymore? Does that mean the screen does not work anymore ore anything else? Or is this just a problem with the Python GUI?

If the whole system crashes due to the kernel model or does not work anymore then please  describe your problem in more detail and maybe take a photo. This should never happen and I will try to fix this immediately.

ryanabx, 2024-11-01:
For me, it's not crashing my system, it's just a dnf conflict where it can't find something python related. It probably just needs a small fix in the spec file, but I don't know enough about this project to know exactly what the fix is.

Allephh, 2024-11-01:
Hi!
No, the system does not crash.
As i said before, it shows problems related to Python and, possibly, DNF5 (possibly commands not existing anymore).
The errors mentioned were at boot (turn on, restart and turn off), where they appeared when pressing ALT + TAB. There were few, in red and, speaking of them, i think there was one related to dkms.
Apart from that, the system worked as if it did not have the driver installed. The Python GUI was there, i could open it, but nothing changed. FN + Q, for example, did not change anything except the LED light.

Allephh, 2024-11-01:
> Do I get this correctly that the system crashes and you cannot use it anymore? Does that mean the screen does not work anymore ore anything else? Or is this just a problem with the Python GUI?
>
> If [...]

Hi! Thank your for the response.
No, the system does not crash.
As i said before, it shows problems related to Python and, possibly, DNF5 (possibly commands not existing anymore).
The errors mentioned were at boot (turn on, restart and turn off), where they appeared when pressing ALT + TAB. There were few, in red and, speaking of them, i think there was one related to dkms.
Apart from that, the system worked as if it did not have the driver installed. The Python GUI was there, i could open it, but nothing changed. FN + Q, for example, did not change anything except the LED light.

MrDuartePT, 2024-11-01:
I not understand why @Allephh @ryanabx I still having problems.

I rebuild the packages for Fedora 41 and now they were build with Python 3.13. I also did try to install them with the latest live cd and they install without any error.

I recommend you both to remove the repo and add it again to see if get solve.

ryanabx, 2024-11-01:
> I recommend you both to remove the repo and add it again to see if get solve.

I'll try it out after work

Allephh, 2024-11-01:
> I not understand why @Allephh @ryanabx I still having problems.
>
> I rebuild the packages for Fedora 41 and now they were build with Python 3.13. I also did try to install them with the latest live cd [...]

[image]

Still accuses python(abi).

Allephh, 2024-11-01:
> > I not understand why @Allephh @ryanabx I still having problems.
> > I rebuild the packages for Fedora 41 and now they were build with Python 3.13. I also did try to install them with the latest live cd and they install without any error.
> > I recommend you both to remove the repo and add it again to see if get solve.
> 
> ![Captura de tela de 2024-11-01 18-07-04](https://private-user-images.githubusercontent.com/48419496/382406515-8c885f02-24f4-4276-9e5f-64bfdccb2bf1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0OTU1OTUsIm5iZiI6MTczMDQ5NTI5NSwicGF0aCI6Ii80ODQxOTQ5Ni8zODI0MDY1MTUtOGM4ODVmMDItMjRmNC00Mjc2LTllNWYtNjRiZmRjY2IyYmYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTAxVDIxMDgxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ2MDEzOWY5NjljOWJjZjIzYWVlZjQ2ODc0ODI5ODhlZWU5MTMxNWFiMTIzMzlkZjQ5Yjc2YTI5ZDc0YWJiYjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.k9A8Okx8siofs0b8vB2DXiZPcf_kw8PD8ZmZS4oscX4)
> 
> Still accuses python(abi).

If i install it, maybe it will give the same problem. I don't know if i want to do this now, maybe later, because i will have to format the computer again to make it clean.

Allephh, 2024-11-01:
I remember installing python 3.12 to try to solve it. I also remember installing a "pip" package and then a "darkdetect" package, but it didn't solve it either and the error continued.

Allephh, 2024-11-02:
Hello and good morning, good afternoon or good night!
This time it didn't report the "non-existent commands" of DNF5, however, it didn't work either. 
Here is the log of installation:
[sudo dnf install dkms.txt](https://github.com/user-attachments/files/17606364/sudo.dnf.install.dkms.txt)
Here some videos, sorry for the quality. I hope it helps:

https://github.com/user-attachments/assets/8dd9f343-50ac-4eb1-a52e-1b8d0872e734

https://github.com/user-attachments/assets/ce4bea23-d0a9-4239-984f-1e6aa737cd51

https://github.com/user-attachments/assets/55b7f94b-7d71-46ab-850a-a71fbed98980

https://github.com/user-attachments/assets/0748e1ef-01b3-4dfc-bfe9-2ccbf803e19d

Allephh, 2024-11-02:
So, the from system shutdown video probably don't relate to this driver. It's some weird thing that appear since Fedora 40 (USB hub or something). Therefore, the errors that appear when i turn off or at the stage of turning off the machine during a restart have nothing to do with it.
So, the real driver errors is when i turn on the system. Due to the poor quality of the video, unfortunately it is not possible to see them. However, one of them is DKMS. The other, i will check. Wait a bit.

Allephh, 2024-11-02:
Nah, i think there one when i shutdown. 2 errors when i start the system.

https://github.com/user-attachments/assets/a73fbaac-deda-45c0-ab6b-ead939fd96f7

https://github.com/user-attachments/assets/208e1402-e91f-473d-b9f3-f75c2d8be86d

https://github.com/user-attachments/assets/a16158d4-4824-4da5-9edf-4900846ac00d

Allephh, 2024-11-02:
alexandre@Bolha-Desktop:~$ systemctl status dkms.service
× dkms.service - Builds and install new kernel modules through DKMS
     Loaded: loaded (/usr/lib/systemd/system/dkms.service; enabled; preset: ena>
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Sat 2024-11-02 07:04:16 -03; 17mi>
 Invocation: 12aa7c8c303b4e44a0d6148978822d8b
       Docs: man:dkms(8)
    Process: 1000 ExecStart=/usr/sbin/dkms autoinstall --verbose --kernelver 6.>
   Main PID: 1000 (code=exited, status=11)
   Mem peak: 136.8M
        CPU: 1.072s

nov 02 07:04:16 Bolha-Desktop dkms[1240]: (bad exit status: 2)
nov 02 07:04:16 Bolha-Desktop dkms[1610]: Error! Bad return status for module b>
nov 02 07:04:16 Bolha-Desktop dkms[1610]: Consult /var/lib/dkms/LenovoLegionLin>
nov 02 07:04:16 Bolha-Desktop dkms[1000]: Autoinstall on 6.11.5-300.fc41.x86_64>
nov 02 07:04:16 Bolha-Desktop dkms[1611]: Error! One or more modules failed to >
nov 02 07:04:16 Bolha-Desktop dkms[1611]: Refer to previous errors for more inf>
nov 02 07:04:16 Bolha-Desktop systemd[1]: dkms.service: Main process exited, co>
nov 02 07:04:16 Bolha-Desktop systemd[1]: dkms.service: Failed with result 'exi>
nov 02 07:04:16 Bolha-Desktop systemd[1]: Failed to start dkms.service - Builds>
nov 02 07:04:16 Bolha-Desktop systemd[1]: dkms.service: Consumed 1.072s CPU tim>
lines 1-22/22 (END)...skipping...
× dkms.service - Builds and install new kernel modules through DKMS
     Loaded: loaded (/usr/lib/systemd/system/dkms.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Sat 2024-11-02 07:04:16 -03; 17min ago
 Invocation: 12aa7c8c303b4e44a0d6148978822d8b
       Docs: man:dkms(8)
    Process: 1000 ExecStart=/usr/sbin/dkms autoinstall --verbose --kernelver 6.11.5-300.fc41.x86_64 (code=exited, status=11)
   Main PID: 1000 (code=exited, status=11)
   Mem peak: 136.8M
        CPU: 1.072s

nov 02 07:04:16 Bolha-Desktop dkms[1240]: (bad exit status: 2)
nov 02 07:04:16 Bolha-Desktop dkms[1610]: Error! Bad return status for module build on kernel: 6.11.5-300.fc41.x86_64 (x86_64)
nov 02 07:04:16 Bolha-Desktop dkms[1610]: Consult /var/lib/dkms/LenovoLegionLinux/0.0.18/build/make.log for more information.
nov 02 07:04:16 Bolha-Desktop dkms[1000]: Autoinstall on 6.11.5-300.fc41.x86_64 failed for module(s) LenovoLegionLinux(10).
nov 02 07:04:16 Bolha-Desktop dkms[1611]: Error! One or more modules failed to install during autoinstall.
nov 02 07:04:16 Bolha-Desktop dkms[1611]: Refer to previous errors for more information.
nov 02 07:04:16 Bolha-Desktop systemd[1]: dkms.service: Main process exited, code=exited, status=11/n/a
nov 02 07:04:16 Bolha-Desktop systemd[1]: dkms.service: Failed with result 'exit-code'.
nov 02 07:04:16 Bolha-Desktop systemd[1]: Failed to start dkms.service - Builds and install new kernel modules through DKMS.
nov 02 07:04:16 Bolha-Desktop systemd[1]: dkms.service: Consumed 1.072s CPU time, 136.8M memory peak.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~

Allephh, 2024-11-02:
cat /var/lib/dkms/LenovoLegionLinux/0.0.18/build/make.log
DKMS make.log for LenovoLegionLinux-0.0.18 for kernel 6.11.5-300.fc41.x86_64 (x86_64)
sáb 02 nov 2024 07:04:15 -03
make -C /lib/modules/6.11.5-300.fc41.x86_64/build M=/var/lib/dkms/LenovoLegionLinux/0.0.18/build modules
make[1]: Entrando no diretório '/usr/src/kernels/6.11.5-300.fc41.x86_64'

  CC [M]  /var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.o
/var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.c:6147:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 6147 |         .remove = legion_remove,
      |                   ^~~~~~~~~~~~~
/var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.c:6147:19: note: (near initialization for ‘legion_driver.<anonymous>.remove’)
/var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.c:1816:16: warning: ‘ecram_memoryio_write’ defined but not used [-Wunused-function]
 1816 | static ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
      |                ^~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/LenovoLegionLinux/0.0.18/build/legion-laptop.o] Error 1
make[2]: *** [/usr/src/kernels/6.11.5-300.fc41.x86_64/Makefile:1966: /var/lib/dkms/LenovoLegionLinux/0.0.18/build] Error 2
make[1]: *** [Makefile:236: __sub-make] Error 2
make[1]: Saindo do diretório '/usr/src/kernels/6.11.5-300.fc41.x86_64'

make: *** [Makefile:13: all] Error 2

Allephh, 2024-11-02:
[image]
[image]

closed 2025-05-11 by st0nie