johnfanv2/LenovoLegionLinux #250: [BUG] function signature, failing to compile on kernel-headers-6.11.0-0.rc5.43.fc41.x86_64

imalone, 2024-09-04:
On Fedora F41 I find this when building:
```
make -C /lib/modules/6.11.0-0.rc5.43.fc41.x86_64/build M=/home/ian/git/LenovoLegionLinux/kernel_module modules
make[1]: Entering directory '/usr/src/kernels/6.11.0-0.rc5.43.fc41.x86_64'
  CC [M]  /home/ian/git/LenovoLegionLinux/kernel_module/legion-laptop.o
/home/ian/git/LenovoLegionLinux/kernel_module/legion-laptop.c:6147:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 6147 |         .remove = legion_remove,
      |                   ^~~~~~~~~~~~~
/home/ian/git/LenovoLegionLinux/kernel_module/legion-laptop.c:6147:19: note: (near initialization for ‘legion_driver.<anonymous>.remove’)
/home/ian/git/LenovoLegionLinux/kernel_module/legion-laptop.c:1816:16: warning: ‘ecram_memoryio_write’ defined but not used [-Wunused-function]
 1816 | static ssize_t ecram_memoryio_write(const struct ecram_memoryio *ec_memoryio,
      |                ^~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /home/ian/git/LenovoLegionLinux/kernel_module/legion-laptop.o] Error 1
make[2]: *** [/usr/src/kernels/6.11.0-0.rc5.43.fc41.x86_64/Makefile:1943: /home/ian/git/LenovoLegionLinux/kernel_module] Error 2
make[1]: *** [Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.11.0-0.rc5.43.fc41.x86_64'
make: *** [Makefile:13: all] Error 2
```

Compilation and make reloadmodule tests go okay if I patch the return value of legion_remove:
```
diff --git a/kernel_module/legion-laptop.c b/kernel_module/legion-laptop.c
index 405b198..b510458 100644
--- a/kernel_module/legion-laptop.c
+++ b/kernel_module/legion-laptop.c
@@ -6084,7 +6084,7 @@ err_model_mismtach:
        return err;
 }
 
-static int legion_remove(struct platform_device *pdev)
+static void legion_remove(struct platform_device *pdev)
 {
        struct legion_private *priv = dev_get_drvdata(&pdev->dev);
 
@@ -6112,7 +6112,7 @@ static int legion_remove(struct platform_device *pdev)
        legion_shared_exit(priv);
 
        pr_info("Legion platform unloaded\n");
-       return 0;
+       return;
 }
```

johnfanv2, 2024-09-07:
Thanks a lot for the report and the solution. Please check if it also works with your kernel with my fix in #251, where I tried to make it compatible with old and new kernels.

imalone, 2024-09-07:
Thanks, yes, 61f05e082afe2af9fec3d038e8c800192b2de07f in #251 compiles and loads okay with make reloadmodule.

johnfanv2, 2024-09-08:
Great. Thank you for your help!

johnfanv2, 2024-09-08:
Great! I really appreciate your help.

closed 2024-09-08 by johnfanv2